ros2/ros2 #1331: why topics discontinuously disappear with enter the command [ ros2 topic list ]

chenyongmin1, 2022-10-11:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu20.04
- Installation type:
  -  ros2 foxy  from source  
- DDS implementation:
  -  rmw_implementation used is default 

#### Steps to reproduce issue

1. open terminal and run  one or more pkgs of mapping and nav2.
2. open another terminal to enter  [ ros2 topic list ] 
3. check the topic list 

#### Expected behavior

**list will show the full(normal) topic**

like:
cym@cym:~/sim_diff_demo$ ros2 topic list
/clicked_point
/diagnostics
/diffbot_base_controller/cmd_vel_unstamped
/dynamic_joint_states
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/move_base_simple/goal
/odom
/odom_diff
/odom_raw
/openzen/is_autocalibration_active
/openzen/mag
/openzen/raw_imu
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf

#### Actual behavior
:
cym@cym:~/sim_diff_demo$ ros2 topic list
/clicked_point
/diagnostics
/diffbot_base_controller/cmd_vel_unstamped
/dynamic_joint_states
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/move_base_simple/goal
/odom
/odom_diff
/odom_raw
/openzen/is_autocalibration_active
/openzen/mag
/openzen/raw_imu
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
cym@cym:~/sim_diff_demo$ ros2 topic list
/clicked_point
/diagnostics
/diffbot_base_controller/cmd_vel_unstamped
/dynamic_joint_states
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/move_base_simple/goal
/odom
/odom_diff
/odom_raw
/openzen/is_autocalibration_active
/openzen/mag
/openzen/raw_imu
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static
cym@cym:~/sim_diff_demo$ ros2 topic list
/parameter_events
/rosout
cym@cym:~/sim_diff_demo$ ros2 topic list
/diagnostics
/diffbot_base_controller/cmd_vel_unstamped
/dynamic_joint_states
/joint_states
/odom
/odom_diff
/odom_raw
/openzen/is_autocalibration_active
/openzen/mag
/openzen/raw_imu
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/tf
/tf_static
cym@cym:~/sim_diff_demo$ ros2 topic list
/parameter_events
/rosout
cym@cym:~/sim_diff_demo$ ros2 topic list
/parameter_events
/rosout
cym@cym:~/sim_diff_demo$ ros2 topic list
/dynamic_joint_states
/odom_raw
/parameter_events
/rosout
/scan
/tf
/tf_static
cym@cym:~/sim_diff_demo$ ros2 topic list
/clicked_point
/diagnostics
/diffbot_base_controller/cmd_vel_unstamped
/dynamic_joint_states
/initialpose
/joint_states
/map
/map_metadata
/map_updates
/move_base_simple/goal
/odom
/odom_diff
/odom_raw
/openzen/is_autocalibration_active
/openzen/mag
/openzen/raw_imu
/parameter_events
/robot_description
/rosout
/scan
/set_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static

#### Additional information

but the function of running nodes (mapping and navigation) seems is OK :

like:
[image]

chenyongmin1, 2022-10-11:
pls help me check this issue.thanks.

fujitatomoya, 2022-10-11:
@chenyongmin1 i am not sure what the problem here, could you explain a little bit more, like you can see the topics which are not supposed to exist? or redundant topic names? or topic name path is wrong?

and it would be helpful if you add,

- What rmw implementation is used? rmw_fastrtps_cpp?
- Foxy source repo hash that you can reproduce this issue?

chenyongmin1, 2022-10-12:
@fujitatomoya  thank you for your reply first.
  the issue is the topic list will change (sometimes it is full ,then ,when i enter [ros2 topic list] again, the topic list isn't full and lose some topics.),but the function of mapping and navigation seems running ok,can map and nav.

like this:

1. enter [ros2 topic list]
2. output :/diffbot_base_controller/cmd_vel_unstamped
/dynamic_joint_states
/joint_states
/map
/map_metadata
/openzen/is_autocalibration_active
/openzen/mag
/openzen/raw_imu
/parameter_events
/robot_description
/rosout
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static

3. go on,enter [ros2 topic list]
4. output:/parameter_events
/rosout

5. go on,enter [ros2 topic list]
6. output:/parameter_events
/rosout

7. go on,enter [ros2 topic list]
8. output:/map
/map_metadata
/parameter_events
/rosout
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf

like this photo:
[image]

add information for you:

- What rmw implementation is used? rmw_fastrtps_cpp?  >>yes ,use ros2 foxy default settings

- Foxy source repo hash that you can reproduce this issue? >>yes, the issue reproduces