ros2/ros2 #1332: Issue with ROS 2 and Python's subprocess library - "Failed to get logging directory..."

matthew-ay, 2022-10-13:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Binaries and from source 
- Version or commit hash: Galactic
- DDS implementation: Not sure what this is. 
- Client library (if applicable): rclpy 

#### Steps to reproduce issue

- Create a standard ROS 2 package from ros2/demos. (Package 1).  
- Create another standard ROS 2 package (Package 2), that contains one file that acts as the driver and entry point. 
- Within the driver code of Package 2, write the code necessary for starting a child process using Python's subprocess library. 
- Have the driver code open a child process with the goal of executing "ros2 run package2 package2_entry_point" command. 
    - Ensure that the arguments string is sourcing your ros2 install and you workspace overlay. 
    - `source ~/ros2_galactic/install/local_setup.bash\n` for binary and `source /opt/ros/galactic/setup.bash\n` for source. 
    - `. install/setup.bash\n` for workspace overlay (within the correct directory). 
- The child process will fail to open due to a "Failed to get logging directory..." 

(If y'all would prefer for me to put together a ROS 2 workspace that contains the files in the "Steps to reproduce issue" section, I can. My hope is that someone knows the solution to this issue already.) 

#### Expected behavior
The child process (ROS 2 node) should start. 

#### Actual behavior
I get the following errors on my "installed from binaries" setup of Galactic: 

```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'rcutils_expand_user failed, at /home/dasl/ros2_galactic/src/ros2/rcl_logging/rcl_logging_interface/src/logging_dir.c:81'

with this new error message:

  'Failed to get logging directory, at /home/dasl/ros2_galactic/src/ros2/rcl_logging/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp:84'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1665613676.346204470] [rclcpp]: rcl context unexpectedly not shutdown during cleanup
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to configure logging: Failed to get logging directory, at /home/dasl/ros2_galactic/src/ros2/rcl_logging/rcl_logging_spdlog/src/rcl_logging_spdlog.cpp:84
```

I get the following errors on my "installed from source" setup of Galactic: 

```
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'rcutils_expand_user failed, at /tmp/binarydeb/ros-galactic-rcl-logging-interface-2.1.4/src/logging_dir.c:81'

with this new error message:

  'Failed to get logging directory, at /tmp/binarydeb/ros-galactic-rcl-logging-spdlog-2.1.4/src/rcl_logging_spdlog.cpp:84'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1665610492.766088023] [rclcpp]: rcl context unexpectedly not shutdown during cleanup
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to configure logging: Failed to get logging directory, at /tmp/binarydeb/ros-galactic-rcl-logging-spdlog-2.1.4/src/rcl_logging_spdlog.cpp:84
```

#### Additional information 
I thought this would a logging configuration issue but setting ROS_LOG_DIR and ROS_HOME did not fix the issue. For context, here is the information I am referring to: 

- [Logging directory configuration](https://docs.ros.org/en/galactic/Tutorials/Demos/Logging-and-logger-configuration.html#id8)

I found these two webpages that were relevant to this issue: 

- [Systemd does not work with ROS2](https://answers.ros.org/question/367935/systemd-does-not-work-with-ros2/)
- [Unable to Create ros Log File #59 ](https://github.com/osrf/car_demo/issues/59) (pinging @tfoote)

christophebedard, 2022-10-13:
> Within the driver code of Package 2, write the code necessary for starting a child process using Python's subprocess library.

Can you share this code?

From the error messages, it looks like `rcutils_expand_user` failed. This is most likely due to the fact that the `HOME` environment variable isn't set, probably because your subprocess doesn't inherit the parent process' environment. This would explain why setting `ROS_LOG_DIR` or `ROS_HOME` didn't fix the issue either (you didn't say that you were setting those environment variables for the subprocess explicitly, so I just assume you set them in your terminal).

matthew-ay, 2022-10-14:
Hey Chris, 

It was the `HOME` environmental variable not being set for the child process. I had tried this fix previously but I implemented the fix incorrectly! 

This time around: I ran `export` for the parent, used the same `HOME` environmental variable when creating the child process, and that fixed the issue. I'll work on a more robust solution but I'm now longer stuck. 

Thanks for your help and time! 
Matt

closed 2022-10-14 by matthew-ay