ros2/ros2 #1335: Latest ROS2 patch breaks services calling services

jaredsjohansen, 2022-10-18:
## Bug report
Prior to the latest [ROS 2 foxy patch](https://github.com/ros2/ros2/releases/tag/release-foxy-20220928), a service call could call another service and return properly.  Now, it does not.  I suspect something in the latest ROS2 foxy patch broke something (although I have not been able to determine the root cause).

**Why would I want a service to call a service?**  Think of a human using a GCS.  They ask the robotic leader of a swarm to convey a task to each member of the swarm.  After receiving the information, each member of the swarm reports their results through the chain of command to the GCS.

#### Required Info:
- Operating System: Ubuntu 20.04
- Installation type: binaries
- Version or commit hash: foxy
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
1. Unzip this file: [service_from_service_code.zip](https://github.com/ros2/ros2/files/9815583/service_from_service_code.zip). It will contain two rosnodes: `service_from_service_client.py` (representative of the swarm leader) and `service_from_service_server.py` (representative of a swarm member).
2. In Terminal 1, run the server: `python3 service_from_service_server.py`.
3. In Terminal 2, run the client: `python3 service_from_service_client.py`.
4. In Terminal 3, run the following command: `ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{a: 3, b: 4}"`.  (This is a simple service call as a sanity check.  You should get a response with a sum of 7.)
5. In Terminal 3, run the following command: `ros2 service call /add_two_ints_proxy example_interfaces/srv/AddTwoInts "{a: 3, b: 4}"`. 
 (This is a service call that calls another service call.)

#### Expected behavior
Terminal 3 issues a service request to the client (on Terminal 2).   The client receives the request and sends a service request to the server (on Terminal 1).  The server processes the service request and returns the service response to the client (on Terminal 2).  The client (on Terminal 2) receives the response and sends it back to the original callee (on Terminal 3).

#### Actual behavior
Terminal 3 issues a service request to the client (on Terminal 2).   The client receives the request and sends a service request to the server (on Terminal 1).  The server processes the service request and returns the service response to the client (on Terminal 2).  The client (on Terminal 2) _never processes the response (and hence never responds to the initial service request on Terminal 3)._

#### Additional information
None

clalancette, 2022-10-19:
This is a duplicate of https://github.com/ros2/rclpy/issues/1016 , and was fixed in https://github.com/ros2/rclpy/pull/1017 .  That should be released soon.

That said, you really shouldn't be calling a service from inside a service.  In the asynchronous programming model of ROS 2, it is not guaranteed to work.

closed 2022-10-19 by clalancette

jaredsjohansen, 2022-10-19:
Thanks, @clalancette, for the link with the root cause/fix -- I'll check that out!

Also, in regards to the service-within-a-service, I understand that it is not a recommended usage.  Given the co-ops I described, is there a proper approach to handle that type of scenario?

jack-frampton-dx, 2022-10-21:
@clalancette 
May I ask why you "really shouldn't" be calling a service from within another service callback? And why it is not guaranteed to work? Even if the client's inside the service callback are set to a reentrant callback group?

clalancette, 2022-10-21:
> May I ask why you "really shouldn't" be calling a service from within another service callback? And why it is not guaranteed to work? Even if the client's inside the service callback are set to a reentrant callback group?

Imagine you have a MultiThreadedExecutor with 2 threads and 2 separate subscription callbacks that are both in a reentrant callback group.

Data for the first subscription comes in, and is serviced by the first executor thread.  Let's assume that the callback wants to call a service.

Immediately after the first data comes in (before the first subscription finishes), data for the second subscription comes in and is serviced by the second executor thread.  Let's assume that this callback *also* wants to call a service.

Now you deadlock.  There is no 3rd executor thread to respond to the service responses from both of the above subscription callbacks.  This is a bit contrived (on most machines you'll probably have more than 2 executor threads), but the problem scales to any number of threads and subscription callbacks that want to call services.