ros2/ros2 #1337: ament_cmake not found when using Ninja generator and Android toolchain

mpconte, 2022-10-24:
## Bug report

**Required Info:**

- Operating System:
Windows 10
- Installation type:
  - binaries
- Version or commit hash:
  - humble
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
When attempting to build a ros2 cpp application that include the following the CMakeLists.txt file (generated by ros2 sample script **ros2 pkg create --build-type ament_cmake <package_name>**):

```
cmake_minimum_required(VERSION 3.5)
project(my_cpp_pkg)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

I run this command:

`cmake -DCMAKE_TOOLCHAIN_FILE=D:\workspace\android.toolchain.cmake -G Ninja ..`

and source the ros2 installation on Windows (i.e. **call D:\dev\ros2_humble\install\local_setup.bat**)

With the following toolchain file:

```
set (CMAKE_SYSTEM_NAME Android)
set (CMAKE_SYSTEM_VERSION 31) # Android SDK API version
set (CMAKE_ANDROID_API 31) # Android SDK API version
set (CMAKE_ANDROID_NDK "D:/Android/Sdk/ndk/25.1.8937393")
set (CMAKE_ANDROID_ARCH_ABI x86_64)

```

#### Expected behavior
I'm expecting ament cmake to be found and build the sample app

#### Actual behavior
```

-- The C compiler identification is MSVC 19.29.30146.0
-- The CXX compiler identification is MSVC 19.29.30146.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

-- Configuring incomplete, errors occurred!
See also "D:/my_cpp_pkg/bin_ninja/CMakeFiles/CMakeOutput.log".
```

#### Additional information

clalancette, 2022-10-28:
This isn't one of our target platforms, so I'm not surprised it doesn't work.

That said, if you figure out how to make it work, we'd be happy to review patches that improve the situation.

mpconte, 2022-10-31:
Hi @clalancette, are there plans to support ROS2 builds for the Android NDK, particularly on the arm64 platform like most modern Android devices are?

clalancette, 2022-10-31:
Actually, I forgot that we actually have projects using this already.  See https://github.com/sloretz/sensors_for_ros for an example that does run on Android.

mpconte, 2022-10-31:
Hi @clalancette, thanks for the reference. I did try to build ros2 humble from source with the ros.deps file in the master branch of the provided repo but I do receive the following error with the arm64-v8a ABI:

```
ld: error: undefined symbol: getifaddrs
>>> referenced by ifaddrs.c:192 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/ifaddrs/posix/ifaddrs.c:192)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/ifaddrs/posix/ifaddrs.c.o:(ddsrt_getifaddrs)

ld: error: undefined symbol: freeifaddrs
>>> referenced by ifaddrs.c:234 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/ifaddrs/posix/ifaddrs.c:234)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/ifaddrs/posix/ifaddrs.c.o:(ddsrt_getifaddrs)
>>> referenced by ifaddrs.c:234 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/ifaddrs/posix/ifaddrs.c:234)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/ifaddrs/posix/ifaddrs.c.o:(ddsrt_getifaddrs)
>>> referenced by ifaddrs.c:234 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/ifaddrs/posix/ifaddrs.c:234)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/ifaddrs/posix/ifaddrs.c.o:(ddsrt_getifaddrs)

ld: error: undefined symbol: pthread_attr_setinheritsched
>>> referenced by threads.c:302 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/threads/posix/threads.c:302)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/threads/posix/threads.c.o:(ddsrt_thread_create)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/core/CMakeFiles/ddsc.dir/build.make:2353: lib/libddsc.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:499: src/core/CMakeFiles/ddsc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
---
Failed   <<< cyclonedds [4min 34s, exited with code 2]
Aborted  <<< ament_cmake [52.1s]
Aborted  <<< ament_cmake_pycodestyle [50.6s]
Aborted  <<< ament_cmake_pyflakes [49.1s]
Aborted  <<< ament_cmake_pclint [58.5s]
Aborted  <<< ament_lint_common [55.1s]

Summary: 59 packages finished [5min 7s]
  1 package failed: cyclonedds
  5 packages aborted: ament_cmake ament_cmake_pclint ament_cmake_pycodestyle ament_cmake_pyflakes ament_lint_common
  33 packages had stderr output: ament_cmake_clang_format ament_cmake_copyright ament_cmake_core ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_export_definitions ament_cmake_export_dependencies ament_cmake_export_include_directories ament_cmake_export_interfaces ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_export_targets ament_cmake_flake8 ament_cmake_gen_version_h ament_cmake_gmock ament_cmake_google_benchmark ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_lint_cmake ament_cmake_mypy ament_cmake_nose ament_cmake_pep257 ament_cmake_pytest ament_cmake_python ament_cmake_target_dependencies ament_cmake_uncrustify ament_cmake_xmllint ament_lint_auto cyclonedds osrf_testing_tools_cpp test_interface_files test_osrf_testing_tools_cpp
  66 packages not processed
```
 
As well as the following errors when trying to build with the x86_64 ABI:

```
ld: error: undefined symbol: newlocale
>>> referenced by string.c:458 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:458)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isalnum)
>>> referenced by string.c:458 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:458)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isalpha)
>>> referenced by string.c:458 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:458)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isblank)
>>> referenced 15 more times

ld: error: undefined symbol: isalnum_l
>>> referenced by string.c:71 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:71)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isalnum)

ld: error: undefined symbol: isalpha_l
>>> referenced by string.c:72 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:72)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isalpha)

ld: error: undefined symbol: isblank_l
>>> referenced by string.c:73 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:73)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isblank)

ld: error: undefined symbol: iscntrl_l
>>> referenced by string.c:74 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:74)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_iscntrl)

ld: error: undefined symbol: isgraph_l
>>> referenced by string.c:75 (/mnt/d/ros2-humble/src/cyclonedds/src/idl/src/string.c:75)
>>>               CMakeFiles/idl.dir/src/string.c.o:(idl_isgraph)

ld: error: undefined symbol: islower_l
[... 67 lines cut ...]
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/ifaddrs/posix/ifaddrs.c.o:(ddsrt_getifaddrs)
>>> referenced by ifaddrs.c:234 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/ifaddrs/posix/ifaddrs.c:234)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/ifaddrs/posix/ifaddrs.c.o:(ddsrt_getifaddrs)

ld: error: undefined symbol: pthread_attr_setinheritsched
>>> referenced by threads.c:302 (/mnt/d/ros2-humble/src/cyclonedds/src/ddsrt/src/threads/posix/threads.c:302)
>>>               CMakeFiles/ddsc.dir/__/ddsrt/src/threads/posix/threads.c.o:(ddsrt_thread_create)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/core/CMakeFiles/ddsc.dir/build.make:2353: lib/libddsc.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:499: src/core/CMakeFiles/ddsc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
---
[... cut ...]
  23 packages had stderr output: ament_cmake ament_cmake_clang_tidy ament_cmake_core ament_cmake_export_definitions ament_cmake_export_include_directories ament_cmake_export_libraries ament_cmake_export_link_flags ament_cmake_gen_version_h ament_cmake_gmock ament_cmake_google_benchmark ament_cmake_gtest ament_cmake_include_directories ament_cmake_libraries ament_cmake_nose ament_cmake_pclint ament_cmake_pycodestyle ament_cmake_pytest ament_cmake_python ament_cmake_version cyclonedds osrf_testing_tools_cpp test_interface_files test_osrf_testing_tools_cpp
  65 packages not processed

```