ros2/ros2 #1341: [Humble] Cannot echo `/statistics` topic

EduPonz, 2022-11-04:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- Installation type: Binaries (`ros:humble` docker image)
- Version or commit hash: latest as of now ([9bf1a5b14fd9](https://hub.docker.com/layers/library/ros/humble/images/sha256-9bf1a5b14fd9b2266178a5b95667fa2d8e861d335fdc9a0cdd7daf074f5939b1?context=explore))
- DDS implementation: Fast DDS (rmw_fastrtps)
- Client library (if applicable): I believe `ros2 topic echo` leverages `rclpy`

#### Steps to reproduce issue

```dockerfile
FROM ros:humble

SHELL ["/bin/bash", "-c"]

WORKDIR /root/demos/src
RUN git clone https://github.com/ros2/demos.git

WORKDIR /root/demos/
RUN source /opt/ros/humble/setup.bash && \
    colcon build --event-handlers=console_direct+ --packages-select topic_statistics_demo

RUN echo 'source /opt/ros/humble/setup.bash' >> ~/.bashrc
RUN echo 'source /root/demos/install/setup.bash' >> ~/.bashrc
```

1. Build the image with:
    ```bash
    docker build -f Dockerfile -t ros2_cli_statistics_bug .
    ```
1. Run the image:
    ```bash
    docker run -it --rm ros2_cli_statistics_bug:latest /bin/bash
    ```
1. From within the container, run the `display_topic_statistics` application from the `topic_statistics_demo` package. This will create a publisher and a subscriber on the `/topic_statistics_chatter` topic. The subscriber will publish statistics onto the `/statistics` every second. The demo also creates a subscriber to the `/statistics` topic and displays the statistics on the screen, all of which works alright.

    ```bash
    ros2 run topic_statistics_demo display_topic_statistics string
    ```

1. Open another shell on the same container with:

    ```bash
    docker exec -it <container-id> /bin/bash
    ```

1. Try to echo the `/statistics` topic from within the new shell. At this point, `ros2 topic list` shows all the topics correctly, and echoing `/topic_statistics_chatter` is working, but echoing `/statistics` does not work.

    ```bash
    source ~/.bashrc
    ros2 topic list # <------------------------------ OK
    ros2 topic echo /topic_statistics_chatter # <---- OK
    ros2 topic echo /statistics # <------------------ NOK
    ```

#### Expected behavior

I would expect `ros2 topic echo /statistics` to work, as I have not found documentation stating otherwise, please correct me if I'm wrong.

#### Actual behavior

`ros2 topic echo /statistics` fails with the following error:

```
Unable to convert call argument to Python object (compile in debug mode for details)
```

clalancette, 2022-11-04:
> ####
> 
> I would expect `ros2 topic echo /statistics` to work, as I have not found documentation stating otherwise, please correct me if I'm wrong.

You are correct, it should work.

And I confirmed it here; I'm seeing the same problem on Humble.

The good news is that this seems to work just fine on Rolling, so there must be some fix that went in between Humble and Rolling.

fujitatomoya, 2022-11-04:
Error happens on the following.

https://github.com/ros2/rclpy/blob/6e4d2ba292f8db5c667b9c789a6235aaf00450fa/rclpy/src/rclpy/subscription.cpp#L135-L138

statistics include `NaN` data in the message, when it is being converted with `rosidl_python`, it detects the following.

```bash
Unable to convert call argument to Python object (compile in debug mode for details)
```

I think we need to backport https://github.com/ros2/rosidl_python/pull/167 to humble.

fujitatomoya, 2022-11-07:
> and https://github.com/ros2/rosidl_python/pull/185 .

Nice catch 👍 thanks, i missed it.

EduPonz, 2022-11-07:
Hi @fujitatomoya @clalancette ,

I'm not familiar at all with python overlays so I'm not sure whether I'm missing something, but I attempted to test with @clalancette backport and it seems to not solve the issue for me. I've update my Dockerfile and followed the instructions from [my previous comment](https://github.com/ros2/ros2/issues/1341#issue-1435811708) with a similar result.

```dockerfile
FROM ros:humble

SHELL ["/bin/bash", "-c"]

WORKDIR /root/demos/src
RUN git clone https://github.com/ros2/demos.git && \
    git clone -b clalancette/humble-backport-nan-fixes https://github.com/ros2/rosidl_python.git

WORKDIR /root/demos/
RUN source /opt/ros/humble/setup.bash && \
    colcon build \
        --event-handlers=console_direct+ \
        --packages-up-to \
            topic_statistics_demo \
            rosidl_generator_py \
        --cmake-args \
            -DBUILD_TESTING=OFF

RUN echo 'source /opt/ros/humble/setup.bash' >> ~/.bashrc
RUN echo 'source /root/demos/install/setup.bash' >> ~/.bashrc
```

fujitatomoya, 2022-11-07:
I did confirm with cherry-pick https://github.com/ros2/rosidl_python/pull/167 for humble worked in my environment.

I will try again with clean environment, and get back to you.

clalancette, 2022-11-07:
> I'm not familiar at all with python overlays so I'm not sure whether I'm missing something, but I attempted to test with @clalancette backport and it seems to not solve the issue for me. I've update [...]

It's not going to work with an overlay; the problem is in the code that is generated for the individual message files, so you basically need to rebuild everything from source.

fujitatomoya, 2022-11-07:
@clalancette thanks for the comment, i thought that too, just in case i will be making sure that backport branch work as expected.

fujitatomoya, 2022-11-07:
using https://github.com/ros2/ros2/blob/humble/ros2.repos (https://github.com/ros2/ros2/commit/00f3ba9f73916a5eab322710edaaf197f0f10e31) + https://github.com/ros2/rosidl_python/tree/clalancette/humble-backport-nan-fixes, it works.

```bash
### Full Build ROS 2 workspace
root@tomoyafujita:~/ros2_ws/humble_ws# colcon build --symlink-install
...
### Source workspace
root@tomoyafujita:~/ros2_ws/humble_ws# source install/local_setup.bash

### Terminal-1
root@tomoyafujita:~/ros2_ws/humble_ws# ros2 run topic_statistics_demo display_topic_statistics string
[INFO] [1667849291.029731424] [string_talker]: Talker starting up
[INFO] [1667849291.032305895] [string_listener]: Listener starting up
[INFO] [1667849291.034515478] [statistics_listener]: TopicStatisticsListener starting up
[INFO] [1667849296.033679138] [statistics_listener]: Statistics heard:
Metric name: message_age source: string_listener unit: ms
Window start: 32872144 end: 33103797
avg: nan
max: nan
min: nan
sample_count: 0.000000
std_dev: nan
[INFO] [1667849296.033992183] [statistics_listener]: Statistics heard:
Metric name: message_period source: string_listener unit: ms
Window start: 32872144 end: 33103797
avg: 999.941886
max: 1000.145085
min: 999.740975
sample_count: 4.000000
std_dev: 0.170316
[INFO] [1667849301.033405374] [statistics_listener]: Statistics heard:
Metric name: message_age source: string_listener unit: ms
Window start: 33103797 end: 33005929
avg: nan
max: nan
min: nan
sample_count: 0.000000
std_dev: nan
[INFO] [1667849301.033696264] [statistics_listener]: Statistics heard:
Metric name: message_period source: string_listener unit: ms
Window start: 33103797 end: 33005929
avg: 1000.013569
max: 1000.267229
min: 999.723682
sample_count: 5.000000
std_dev: 0.177514

### Terminal-2
root@tomoyafujita:~/ros2_ws/humble_ws# ros2 topic echo /statistics
measurement_source_name: string_listener
metrics_source: message_age
unit: ms
window_start:
  sec: 1667849296
  nanosec: 33103797
window_stop:
  sec: 1667849301
  nanosec: 33005929
statistics:
- data_type: 1
  data: .nan
- data_type: 3
  data: .nan
- data_type: 2
  data: .nan
- data_type: 5
  data: 0.0
- data_type: 4
  data: .nan
---
measurement_source_name: string_listener
metrics_source: message_period
unit: ms
window_start:
  sec: 1667849296
  nanosec: 33103797
window_stop:
  sec: 1667849301
  nanosec: 33005929
statistics:
- data_type: 1
  data: 1000.0135688
- data_type: 3
  data: 1000.267229
- data_type: 2
  data: 999.723682
- data_type: 5
  data: 5.0
- data_type: 4
  data: 0.17751378370976767
---
```

EduPonz, 2022-11-08:
Hi guys!

I just confirmed that the backports work when building humble from sources with @clalancette's PR. I leave here the Dockerfile for completion. Again, thank you guys!

<details>
<summary>Dockerfile</summary>

```dockerfile
FROM ubuntu:22.04

# Needed for a dependency that forces to set timezone
ENV TZ=Europe/Madrid
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

SHELL ["/bin/bash", "-c"]

WORKDIR /ros2_humble_ws

RUN apt update && \
    apt install -y \
        curl \
        gnupg \
        lsb-release

RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
    echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

RUN apt update && \
    apt install -y \
        build-essential \
        cmake \
        git \
        python3-colcon-common-extensions \
        python3-flake8 \
        python3-flake8-docstrings \
        python3-pip \
        python3-pytest \
        python3-pytest-cov \
        python3-rosdep \
        python3-setuptools \
        python3-vcstool \
        wget

RUN apt install -y \
    python3-flake8-blind-except \
    python3-flake8-builtins \
    python3-flake8-class-newline \
    python3-flake8-comprehensions \
    python3-flake8-deprecated \
    python3-flake8-import-order \
    python3-flake8-quotes \
    python3-pytest-repeat \
    python3-pytest-rerunfailures

RUN mkdir -p src && \
    vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src

RUN cd src/ros2/rosidl_python && git checkout clalancette/humble-backport-nan-fixes && cd -

RUN rosdep init
RUN rosdep update
RUN rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"

RUN colcon build \
    --event-handlers=console_direct+ \
    --packages-up-to \
        ros2run \
        ros2topic \
        topic_statistics_demo

RUN echo 'source /ros2_humble_ws/install/setup.bash' >> ~/.bashrc
```
</details>

fujitatomoya, 2022-11-08:
@clalancette i think we can close this issue.

clalancette, 2022-11-08:
Yes, good point.  Closing, thanks.

closed 2022-11-08 by clalancette