ros2/ros2 #1348: Windows builds failing

nuclearsandwich, 2022-11-19:
## Bug report

**Required Info:**

- Operating System:
  - Windows (so far)
- Installation type:
  - source
- Version or commit hash:
  - rolling
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

An upgrade to CMake 3.25.0 on our Windows hosts has resulted in failures when building rviz_assimp_vendor on the ROS 2 CI farm.

Early failures include these jobs although all Windows CI is currently failing to build.

Windows packaging: https://ci.ros2.org/job/packaging_windows/2672/
Nightly Repeated: https://ci.ros2.org/job/nightly_win_rep/2808/

Here's a log excerpt.

```
CMake Error at C:/ci/ws/build/rviz_assimp_vendor/assimp-4.1.0-prefix/src/assimp-4.1.0-stamp/assimp-4.1.0-build-Release.cmake:49 (message):
    Command failed: 1
     'C:/Program Files/CMake/bin/cmake.exe' '--build' '.' '--config' 'Release'
    See also
      C:/ci/ws/build/rviz_assimp_vendor/assimp-4.1.0-prefix/src/assimp-4.1.0-stamp/assimp-4.1.0-build-*.log
c:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: Custom build for 'C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-mkdir.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-download.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-update.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-patch.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-configure.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-build.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\78e9b3a1ee729cfcea0f95967341f51c\assimp-4.1.0-install.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\96521307d7f0ed6922de4eb9fc9135fc\assimp-4.1.0-complete.rule;C:\ci\ws\build\rviz_assimp_vendor\CMakeFiles\90909c6e448bc2e2fbf78dad3dda9f5f\assimp-4.1.0.rule;C:\ci\ws\src\ros2\rviz\rviz_assimp_vendor\CMakeLists.txt' exited with code 1. [C:\ci\ws\build\rviz_assimp_vendor\assimp-4.1.0.vcxproj]
```

nuclearsandwich, 2022-11-19:
While we figure out what's causing this issue and how best to resolve it we're pinning CMake 3.24.3 in the meantime.

clalancette, 2023-01-09:
I think that with the latest CMake 3.25.1 and our unpin, this is no longer an issue.  So I'm going to close this.  Feel free to reopen if I misinterpreted the issue.

closed 2023-01-09 by clalancette