ros2/ros2 #1354: API docs discoverability issues

hidmic, 2022-12-02:
It has been reported (see https://github.com/ros2/rclcpp/issues/2054#issuecomment-1335749145) that API docs may not be as easy to be found as they should be. Specially after [docs.ros2.org](https://docs.ros2.org/latest/api/) deprecation.

hidmic, 2022-12-02:
@clalancette I'm inclined to think this would not be a problem if googling `'package api docs'` did not bring old stuff. Can we improve that somehow?

kscottz, 2022-12-05:
Could we do something simple like add another custom domain like api.ros.org (with patterns like `api.ros.org/humble/rclcpp`, `api.ros.org/foxy/rclpy` and so on)? 

We could also include a link on the sidebar of docs.ros.org that says something like "Looking for Doxygen API docs?"

clalancette, 2022-12-22:
> Could we do something simple like add another custom domain like api.ros.org (with patterns like `api.ros.org/humble/rclcpp`, `api.ros.org/foxy/rclpy` and so on)?

This is the opposite of the way I think we should go.  That is, we already have docs scattered around, and we've been (slowly) trying to converge them.  We already have a hierarchy for all of these docs under https://docs.ros.org/en/<distro>/p , so I think we should just use that.

That said, I agree that discoverability could be improved.  We did consider putting a link in the docs.ros.org sidebar, but decided not to because it was redundant with what is on https://index.ros.org (like https://index.ros.org/p/rclcpp/github-ros2-rclcpp/#rolling, the "API docs" link).  Maybe we should reconsider that decision.

SteveMacenski, 2023-01-31:
Would it be possible to add the API docs for the core client libraries to the main website (e.g. https://docs.ros.org/en/humble/index.html)? Right now, its virtually impossible to find the **most critical** library docs (e.g. rclcpp, rclpy), not even concerning myself with autodocs of arbitrary packages. I had to go from discourse posts to tickets to links to find it, which is surprising since the docs are hosted on a `https://docs.ros.org/en/humble/...` url, just not linking up (unless I missed something?).

Would it be possible to have a new "API Documentation" page that links up to https://docs.ros.org/en/humble/p/ at the very least (and why `p`?) so its findable? I think @kscottz suggested the same above. What I'd suggest is to have a generic link there but also curate a list of URLs to common needs (e.g. rclcpp, rclpy, actions, lifecycl0e, tf2, ...). 

Is there a way to get doxygen style hosted instead of or in addition to the readthedocs styling for core packages? Honestly its hard to read and [I'm not the only one](https://discourse.ros.org/t/plans-for-ros-2-documentation-particularly-api-docs/28638/11?u=smac). I've stopped even trying to use the Rolling or Humble API docs because I can't find them and calling Galactic or Foxy "good enough". Googling "rclcpp humble docs API" doesn't come up with anything and the landing ROS 2 docs pages has no link either.

SteveMacenski, 2023-02-01:
P.S. I'll open a PR for the "API Documentation" page with a link to https://docs.ros.org/en/humble/p/ and what I feel is a curated list if that's something you're open to @clalancette. That's a half measure, but gets the ball rolling so that at least those willing to put a little legwork into finding it can do so faster

clalancette, 2023-02-01:
> Would it be possible to add the API docs for the core client libraries to the main website (e.g. https://docs.ros.org/en/humble/index.html)?

We considered doing this at the beginning, and decided against it since you can get to the package documentation from the index page (like from https://index.ros.org/p/rclcpp/github-ros2-rclcpp/#humble -> API Docs).

That said, I'm not sure that is the correct call, since discoverability has continually been a problem.  I'd be OK with adding a link to the top-level https://docs.ros.org/en/humble/p/ from the main https://docs.ros.org site.  If we wanted to, we could also apply a bit of styling to that directory listing so it isn't so ugly (but that's just icing).  @tfoote @kscottz thoughts?

> Is there a way to get doxygen style hosted instead of or in addition to the readthedocs styling for core packages?

In short, not easily.  The way that rosdoc2 works is that it runs doxygen and imports everything into sphinx (so we can combine with other pages that might be in the package).  So you can have alternate sphinx themes, but not really a doxygen theme.

> Though I'd argue to just rehost in the same way as before during the intermediate time as things are being sorted out since Humble is LTS

I don't think we should do this; it is a step back.  If there are problems with the current documentation, then lets make changes to rosdoc2 and make it work for all packages.

SteveMacenski, 2023-02-01:
> We considered doing this at the beginning, and decided against it since you can get to the package documentation from the index page (like from https://index.ros.org/p/rclcpp/github-ros2-rclcpp/#humble -> API Docs).

The index page doesn't come up on many google search results first page. For example:
- rclcpp API --> Nope
- rclcpp docs --> Nope
- rclcpp --> Yes

I don't totally agree with the decision not to have direct links to API docs since its on the index page, but I get the intent. But separate of the general ROS packages on the index that are released, I think the "core" docs do deserve separate consideration. They're not federated packages like SLAM Toolbox that maybe-maybe-not a particular user cares at all about. They're the core of what it means to "use ROS 2" as an end user and without access to that information, there's not much they can do but google around and look at source locations like `ros2/tutorials` `ros2/examples` or even `ros2/rclcpp` directly. 

The vision of what I think would be good is to have a tab in the main page which points you to the location of all the API docs for a distribution, but also at the top has the client library and key utility docs that have traditionally been posted separately on those doxygen pages (which I think you agree are separately important since they used to be separately hosted). But in the absence of that, at least direct links to the key libraries would be good with a note about how to get to the other API docs via Index (or a link buried at the bottom with a link to the full nasty formatted list).

> If we wanted to, we could also apply a bit of styling to that directory listing so it isn't so ugly (but that's just icing)

100% love that idea, can't help there, readthedocs is about the extent of my web knowledge.

> The way that rosdoc2 works is that it runs doxygen and imports everything into sphinx (so we can combine with other pages that might be in the package). So you can have alternate sphinx themes, but not really a doxygen theme.

I think this goes into the categorization of "special consideration for end-user libraries that 'are ROS'". Again, not a web person, but is it that difficult to generate the doxygen html that you'd make locally and dump it on a web server with the `index.html` pointing to a URL we can have on that `API Docs` tab of the ROS 2 main docs site? That way, its separate from Index but still around and findable. Doxygen doesn't require the code to be compiled, so I think a github action could even just clone rclcpp and rclpy into its container to generate the files and `<do magic web things>` to hook it up. 

> I don't think we should do this; it is a step back.

I don't disagree. Just providing alternatives. Another option is to link to Index on the main docs website and tell users to find the API docs at the package of interest there.

tfoote, 2023-02-06:
Yes, we definitely need to do better to promote the visibility of the API docs. Strategic linking out from the generic docs I think makes sense. The main one that I generally want to avoid is starting to create lists in the manually curated package lists that will get out of date and basically form defacto blessing lists without a formal process for inclusion or exclusion. 

There's a good amount of SEO things we can do for the site too to make it so that your above searches don't come back empty.

I've been working on elements off and on such as https://github.com/ros-infrastructure/rosindex/pull/257  and https://github.com/ros-infrastructure/roswiki/pull/343/files  I'd like to add to the index.ros.org to the wiki PackageHeader for greater cross discoverability too.

SteveMacenski, 2023-02-07:
> The main one that I generally want to avoid is starting to create lists in the manually curated package lists that will get out of date and basically form defacto blessing lists without a formal process for inclusion or exclusion.

I think that's a super fair concern. I was only thinking of the core user-facing libraries (rclcpp, rclpy, rclc?) as 'special' outstandingly useful and core docs. Potentially TF2?

tfoote, 2023-02-09:
Yeah, I would support adding references for rclcpp and rclpy likely integrated with the existing tutorials and concept overviews. Things like tf2 and others we want to move towards the docs being self contained in the package.

SteveMacenski, 2023-02-10:
I can live with that. I can open a PR that adds a new Docs page with links to those and then redirects to Index for seeing others. I still think though that other points discussed above are necessary to close out the ticket (e.g. nicely formatting + linking the index' root page, SEO problems)

hidmic, 2023-03-01:
@clalancette @tfoote as another data point, I can't seem to find ROS 2 Humble documentation for interfaces (ie. messages, services, actions). Another discoverability issue? Or is `rosdoc2` lacking the feature entirely? I see https://github.com/ros2/ros2_generate_interface_docs, but maybe it got dropped when migrating sites?

clalancette, 2023-03-01:
> Or is `rosdoc2` lacking the feature entirely?

Yeah, it lacks the feature entirely right now.

> I see https://github.com/ros2/ros2_generate_interface_docs, but maybe it got dropped from build jobs when migrating sites?

We could consider integrating this with rosdoc2, that might be a good way to go.

jmachowinski, 2023-09-19:
First I would like to mention, that it took me month to figure out that https://docs.ros2.org/latest/api/ is outdated as it is not stated on the page itself.

I would see it as a great improvement, if everything at https://docs.ros2.org/latest/api/ would be removed, and either a automatic forward, or a page stating, that the current documentation can be found under the new location be put there instead.

It would perhaps also make sense, to forward https://docs.ros2.org/iron/ etc to docs.ros.org

clalancette, 2023-09-19:
> I would see it as a great improvement, if everything at https://docs.ros2.org/latest/api/ would be removed, and either a automatic forward, or a page stating, that the current documentation can be found under the new location be put there instead.

Yes, the goal is to do an automatic forward to the current documentation.

jmachowinski, 2023-09-19:
> 
> Yes, the goal is to do an automatic forward to the current documentation.

If this is the goal, what is currently blocking this ?

clalancette, 2023-09-19:
> If this is the goal, what is currently blocking this ?

Finding time to do it.