ros2/ros2 #1355: External logging level configuration

fujitatomoya, 2022-12-05:
## Feature request

### Feature description

Ability to configure logging level externally at runtime. 

> In the future there will be a generalized approach to external configuration of loggers at runtime

see https://docs.ros.org/en/rolling/Tutorials/Demos/Logging-and-logger-configuration.html#logger-level-configuration-externally

[image]

### Requirements

- ROS 2 application (including ros2cli) can change the logging level for specific node during runtime.
- Node developer can `opt-in` this option via `Node:Options`. (in default, this is disabled.)
- Node is only be able to set the logging level of logger name that belongs to that node process space.

### Implementation Consideration

- GUI tools are not in scope of this issue, this is dedicated to node interface.
- Expose additional service for each node to set and get the logging level of logger name at runtime.
  - service path: `/nodename/get_logger_level` and `/nodename/set_logger_level`.
- This could be one of the `Node::Options` that user `opt-in` if they want to. Since this is debug feature and adding extra service, this should not be `opt-out`.
- If using components node, requesting to change logger level to Node-A would be effectively changing logger level in Node-B.
  - This is because that logger level hash map is maintained by in the process space.
  - logger names and levels are maintained agnostic from node configuration, logging level will be applied once the corresponding node is created.
- Not only for setting the logging level, but also user should be able to know the current logging level via service?
  - Probably yes, it is likely that user application might want to check the current logging level during runtime.
  - `rcl_create_node_logger_name` is internally issued and creates logger name for that node, but user application can create other `rclcpp::Logger` if needed.
- ROS 2 system debugging level should be accepted as well by default, such as `rclcpp`, `rcl` and `rclpy`.
  - This could be really useful to debug ROS 2 core system during application running.
- This feature requires service file must have the field to specify the `logger_name` and `logger_level`.
- Supporting user callback to be called on service request?
  - Probably no, what is the actual use case for this?

### Sample Demo

- https://github.com/ros2/demos/blob/rolling/logging_demo/src/logger_config_component.cpp

### Tasks

- [ ] [rcl](https://github.com/ros2/rcl) bypass rcutils_logging_xxx.
- [ ] [rclcpp](https://github.com/ros2/rclcpp) add option and service implementation.
- [ ] [rclpy](https://github.com/ros2/rclpy) add option and service implementation.
- [ ] [demos](https://github.com/ros2/demos) update demo sample accordingly using this feature.
- [ ] [ros2_documentation](https://github.com/ros2/ros2_documentation) update doc [logger-level-configuration-externally](https://docs.ros.org/en/rolling/Tutorials/Demos/Logging-and-logger-configuration.html#logger-level-configuration-externally) accordingly.

fujitatomoya, 2022-12-05:
@clalancette @ivanpauno @wjwwood what do you think?

CC: @iuhilnehc-ynos @Barry-Xu-2018 @ros2/team

Barry-Xu-2018, 2022-12-06:
This design will create 2 services (get/set loglevel) for each node.  Now we are considering another design which use 2 service for serval node like logging_demo behaviour (ros2 service call /config_logger logging_demo/srv/ConfigLogger "{logger_name: 'logger_usage_demo', level: INFO}")

# Runtime Logging Level Reconfiguration Design 

1. add `GetLoggerLevels.srv`, `SetLoggerLevels.srv` to  `rcl_interfaces` for get/set logger levels,

    `GetLoggerLevels.srv`:
    ```
    # A list of logger names to get.
    string[] names

    ---
    # List of logger levels which is the same order as the provided names. If a
    # logger was not found, the value will have 0 (unknown) as the value.
    LoggerLevel[] levels
    ```

    `SetLoggerLevels.srv`:
    ```
    # logger levels
    LoggerLevel[] levels

    ---
    # Indicates whether setting each logger succeeded or not and why.
    SetLoggerLevelResult[] results
    ```

   and add `LoggerLevel.msg`, `SetLoggerLevelResult.msg` for exchange result.

    `LoggerLevel.msg`:
    ```
    # The logger name.
    string name

    # logger level: unknown, debug, info, warn, error, fatal
    #                  0       1      2     3     4      5
    uint8 level
    ```

    `SetLoggerLevelResult.msg`:
    ```
    # True when succeed, false when failed.
    bool successful

    # Reason why the setting was either successful or a failure.
    string reason
    ```

2. In `rclcpp`, add `src/rclcpp/logger_service.cpp`, `include/rclcpp/logger_service.hpp` for logger service class, in this class, it will create 2 services named as `get_logger_levels` and `set_logger_levels`

   the process in callback of service
   * get_logger_level handler -> rclcpp::Logger::get_node_logger() -> rcl::rcl_node_get_logger_name()

   * set_logger_level handler -> rclcpp::Logger::set_level() -> rcutils::rcutils_logging_set_logger_level()

3. In `rclcpp/src/rclcpp/node_options.cpp`, let's add a option of `start_logger_services` which used for start/stop logger services. By default, it is set `false`.

4. Add LoggerService init work in `rclcpp::node_interfaces::NodeLogging`

clalancette, 2022-12-06:
I'm on PTO this week, so I don't have much time to comment.  But I definitely want to talk about the two new services before we implement anything here.  We already have way too much discovery traffic at startup time; this is going to make things much worse.  We should consider ways to reduce or eliminate some of services so we can improve on the startup discovery traffic.

fujitatomoya, 2022-12-22:
I will bring this topic to either `Client` or `Middleware` WG in next year 🍹

Barry-Xu-2018, 2023-01-08:
In order to use 2 service for several nodes like logging_demo behavior, update the design https://github.com/ros2/ros2/issues/1355?notification_referrer_id=NT_kwDOApkn_7M1MDA1Mjg5ODgyOjQzNTkxNjc5&notifications_query=reason%3Aparticipating#issuecomment-1338642822

# Runtime Logging Level Reconfiguration Design

## Service Interfaces

* GetLoggerLevels.srv

    ```
    # A list of logger names to get.
    string[] names

    ---
    # List of logger levels which is the same order as the provided names. If a
    # logger was not yet set, the value will have 'unknown 0' as the value.
    LoggerLevel[] levels
    ```

* SetLoggerLevels.srv

    ```
    # A list of loggers to set.
    LoggerLevel[] levels

    ---
    # Indicates whether setting each logger succeeded or not and why.
    SetLoggerLevelsResult[] results
    ```

* LoggerLevel.msg

    ```
    # The logger name.
    string name

    # logger level: UNKNOWN, DEBUG, INFO, WARN, ERROR, FATAL
    #                  0      10     20    30    40      50
    uint8 level
    ```

* SetLoggerLevelsResult.msg

    ```
    # True when succeed, false when failed.
    bool successful

    # Reason while successful is false.
    string reason
    ```

You can put them into `rcl_interfaces`.

## Design Instruction

1. Add new option `enable_log_service` to NodeOptions

    This is a debug feature, so the default value is `false`.

    ```c++
    NodeOptions::enable_log_service_{false}
    ```

    When you create a `Node` with `enable_log_service` enabled, the `Node` will create a log service with it, the log service path looks like below:

    * /node_name/get_logger_levels

    The `node_name` will be the `Node's` name, which depend on the user, the operation code looks like below:

    ```c++
    class YourNode: public rclcpp::Node
    {
      ...
    };
    
    rclcpp::NodeOptions options = rclcpp::NodeOptions();
    options.enable_log_service(true);
    auto node = std::make_shared<YourNode>(options);
    ...
    ```

    Notice: while user create more nodes, if NodeOptions enable `enable_log_service`, a log service will be created under Node. The function of log service is the same, which can get/set logger for all nodes. So user has responsibility to decide where log service is created. 

2. Start log service

    In the Node initial stage, we can start logger services when `enable_log_service_` enabled. The implementation is like `use_sim_time` feature https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/time_source.cpp#L382-L435

    In order to avoid creating many threads, we should consider using one thread for different internal subscription, service, etc. So related common code will be moved to a new class.

3. New `LoggerService` class

    Create a new file for `LoggerService` class, which included followings:

    * GetLoggerLevels service handler

    ```c++
    ...
        for (auto &n : request->names) {
          level = rcutils_logging_get_logger_level(n.c_str());
        }
    ...
    ```

    * SetLoggerLevels service handler

    ```c++
    ...
    for (auto & l : request->levels) {
        ret = rcutils_logging_set_logger_level(l.name.c_str(), l.level);
      }
    ...
    ```

fujitatomoya, 2023-01-08:
@Barry-Xu-2018 thanks for taking the initiative on this. it looks good to me as basic design.

@clalancette what do you think? instead of having configuration logging service for each node, do you have any proposals? I knew that you are concerned to have additional services to support this feature.

clalancette, 2023-01-24:
> @clalancette what do you think? instead of having configuration logging service for each node, do you have any proposals?

> ```c++
> NodeOptions::enable_log_service_{false}
> ```

I think if the default is for this to be off, then it is fine.  I still think we should reduce the number of services we offer by default (like combining all of the parameter services into one), but that can come separately.

EduPonz, 2023-01-24:
> I still think we should reduce the number of services we offer by default (like combining all of the parameter services into one), but that can come separately.

This would make total sense from the DDS point of view, and I believe all the piece are there to make this somehow more transparent and less cumbersome to the users, may using DDS topic instances (having the node name as topic key) and a union data type. It may be possible to do this on the rmws since it pertains the ROS 2 builtins. Definitely a discussion worth having!

fujitatomoya, 2023-01-25:
@Barry-Xu-2018 thanks for iterating with us, i think current basic design is okay. so let's proceed the implementation.

fujitatomoya, 2023-01-25:
> I still think we should reduce the number of services we offer by default (like combining all of the parameter services into one), but that can come separately.

Just FYI, https://github.com/ros-infrastructure/rep/pull/358/files#r1065416872 mentioned `lazy service server` which could be useful to address this situation.

bijoua29, 2023-02-22:
Thanks for this feature. It has been a long-time coming.
Seems like you can only set the logger level for all nodes in a component node i.e., you cannot individually set the logger level for a node in a component node. That seems very limiting and might drive people not to use component nodes.

fujitatomoya, 2023-02-24:
> Seems like you can only set the logger level for all nodes in a component node i.e., you cannot individually set the logger level for a node in a component node.

Why? can you explain this a little bit more about this? i might be missing something.

after component manager loads the any nodes, that will also have service to change the logging level to the logger. it should be okay with current design and interfaces?

fujitatomoya, 2023-02-28:
@Barry-Xu-2018 any update for this implementation? just checking i am not missing anything here.

Barry-Xu-2018, 2023-03-01:
@fujitatomoya 

Internal review was finished. Code is being updated based on review comments.

jrutgeer, 2023-03-02:
Conceptually, the loglevel is a parameter of a node (or actually, of a Logger).
So why implement this as services and not as a parameter?

bijoua29, 2023-03-06:
> Why? can you explain this a little bit more about this? i might be missing something.
>
> after component manager loads the any nodes, that will also have service to change the logging level to the [...]

I am going by the statement 'If using components node, requesting to change logger level to Node-A would be effectively changing logger level in Node-B'

fujitatomoya, 2023-03-06:
> So why implement this as services and not as a parameter?

it is doable with parameters, but simply taking advantage of logger hash map backend via service would be good to avoid the data duplication to have those data in parameters.

bijoua29, 2023-03-06:
> > I am going by the statement 'If using components node, requesting to change logger level to Node-A would be effectively changing logger level in Node-B'
> 
> 😅 Ah, i came up with that, sorry.
> 
> (this is not only for component case but multiple nodes in the same process space.) what i mean here is that basically having node name and subordinate loggers are the cases most likely, and this is no problem even with nodes loaded into component. (e.g Node `foo` with logger name with `foo`, `foo.bar`. and Node `hoge` with logger name with `hoge`, `hoge.fuga`)
> 
> but if we have global logger in the process space such as `rclcpp::get_logger("global_logger")` and then calling service to change the log level for that will affect the every nodes that uses this global logger. which I think this should be okay, since user knows that this is global logger and created.

Ah, so what you mean is that if a global logger is used by multiple nodes, then you are effectively changing the logger level of multiple nodes. But if each node has it's own logger in a component case, then you can change the logger level of each node individually - correct?

fujitatomoya, 2023-03-06:
@bijoua29 yeah that is correct. sorry for the confusion and thanks for reminding me!!!

jrutgeer, 2023-03-07:
@fujitatomoya

Thank you for your answer.

>simply taking advantage of logger hash map backend via service would be good to avoid the data duplication to have those data in parameters.

I understand.

Some further thoughts though:

Given the service takes the "name" and "loglevel" and adapts the hasmap accordingly, there's no strict need to have this service in each node. One such service per process suffices.
In fact, it is rather straightforward to define a node which provides such a service, see [here](https://github.com/jrutgeer/ROS2_assorted_tests/blob/main/logger_test/src/logger_set.cpp) for an example (it is just a proof of concept with an empty service). 

This same node could e.g. also deal with reading loglevels from a configuration file, (as an alternative to [this](https://github.com/ros2/rcutils/pull/347)) as well as provide a publisher for non-node log messages (so these are also published to /rosout), see [this remark](https://github.com/ros2/rclcpp/issues/1694#issuecomment-1453492628).

So questions:
1. Is it really desirable to implement this in the ROS 2 core? Or should this rather be a "convenience utility" (which could still be part of the official ROS distribution, but not necessarily of rclcpp / rcutils / rcl)?,
2. If it is implemented in ROS 2 core, why implement it as a service per node and not with one node (which could be auto-instantiated in the background)?

fujitatomoya, 2023-03-07:
> Is it really desirable to implement this in the ROS 2 core? Or should this rather be a "convenience utility" (which could still be part of the official ROS distribution, but not necessarily of rclcpp / rcutils / rcl)?

I am not really sure about what you mean here? are you saying `convenience utility` that is not implemented in ROS 2  but provide those packages as ROS 2 distribution like community package?

> If it is implemented in ROS 2 core, why implement it as a service per node and not with one node (which could be auto-instantiated in the background)?

I think logger level should be managed under node instance to request the service to that node. (specifically the logger bound to /rosout.)

We can have a system node as you suggested, to manage this kind of configuration, but how do we manage the followings?

- the node is also exposed to ROS network from `every single process space`. if there are 100 processes, there will be 100 system nodes in ROS network.
- how do we manage the name and namespace for this system node globally in ROS distributed system and process space? e.g. how user knows that which system node is responsible to specific logger in user's node?

to be honest, i do not think this is good user experience at this moment.

jrutgeer, 2023-03-07:
> how do we manage the name and namespace for this system node globally in ROS distributed system and process space? e.g. how user knows that which system node is responsible to specific logger in user's node?

Indeed, this is a fair argument. You would need to know which node is in which process in order to set its loglevel.

However even with services, this still holds for named loggers. You will need to know a node in the same process to set its loglevel. I think this is acceptable, and conclude is that it is generally better to use node loggers if possible, as named loggers have certain functional restrictions (e.g. no publish to /rosout and less straightforward to set the loglevel).

Thank you!

fujitatomoya, 2023-03-07:
> You will need to know a node in the same process to set its loglevel.

why? i think it is agnostic from process space, but node that are created by user application? i was thinking that user experience is really different... i might be mistaken though.

jrutgeer, 2023-03-07:
Because there is an rcutils hash table per process. So if node1 is in one process and node2 in another , and you do a setloglevel service call on node1 for a named logger (e.g. "rcl" or "rclcpp") this will only affect the log level for the corresponding process, and not for the process of node2.

fujitatomoya, 2023-03-08:
@jrutgeer thanks for iterating, just checking if i understand precisely.

> So if node1 is in one process and node2 in another , and you do a setloglevel service call on node1 for a named logger (e.g. "rcl" or "rclcpp") this will only affect the log level for the corresponding process, and not for the process of node2.

Isn't this the expected behavior? and user's perspective, user can only care about node1 to request the log level for logger. If user want to changing log level for node2, they can call service to node2, right?

i was thinking that downside could be the multiple nodes running in the single process, in that case, if we allow user to change the log level for named logger such as `rcl` or `rclcpp` via service provided nodes, it will be set effectively entire process space. (related to https://github.com/ros2/ros2/issues/1355#issuecomment-1456848788)

llapx, 2023-03-08:
#https://github.com/ros2/rclcpp/pull/2122
#https://github.com/ros2/rcl_interfaces/pull/154

llapx, 2023-03-08:
#https://github.com/ros2/rclcpp/pull/2122

fujitatomoya, 2023-03-08:
@llapx thanks for the PRs, I will be working on review.

@Barry-Xu-2018 @iuhilnehc-ynos can you review when you are available?

jrutgeer, 2023-03-08:
> Isn't this the expected behavior? and user's perspective, user can only care about node1 to request the log level for logger. If user want to changing log level for node2, they can call service to node2, right?

@fujitatomoya 
I think, given the current [logging documentation](http://docs.ros.org/en/rolling/Concepts/About-Logging.html), for an end user it is not an obvious conclusion (without looking at source code) that a "logger" created by `node->get_logger()` has a different behavior than a logger created by `rclcpp::get_logger("name")`. Both are  instances of the same `rclcpp::Logger` class. 

Obviously, _if_ you know about ROS 2 internal implementation (i.e. about the per-process hash tables etc), _then_ it is obvious that "node loggers" and "named loggers" have different behavior.

I think _in general_ it is best practise to always try to avoid such implicit differences. But in this case I think it is fair to conclude that this is an inherent difference between node loggers and named loggers, as nodes are exposed in the ROS network (and hence become process-agnostic) whereas named loggers don't have an associated node and hence are not process-agnostic.

Thank you for this interesting discussion, I have learned from it.

Barry-Xu-2018, 2023-03-08:
> @Barry-Xu-2018 @iuhilnehc-ynos can you review when you are available?  

We had reviewed it internally, except test code.   
Now we consider whether another way to avoid change or less change for interface of node_base.

jrutgeer, 2023-03-08:
@fujitatomoya 

Another topic: are you aware of https://github.com/ros2/rcutils/issues/397?

This is about crashes in case of concurrent calls to `rcutils_logging_get_logger_level` and `rcutils_logging_set_logger_level`.

`rcutils_logging_get_logger_level` is called on each log call.

Currently, concurrent calls to `get` and `set` are not likely to happen, as loglevels are typically only set during initialization (from the command line arguments), and at that time there are no logging calls yet.

However, given the new `set_logger_level` service, it will be more likely for `get` and `set` to be called concurrently.
So if  https://github.com/ros2/rcutils/issues/397 is not addressed as well, I think using the `set_logger_level` service can lead to sporadic crashes?

fujitatomoya, 2023-03-08:
> I think, given the current [logging documentation](http://docs.ros.org/en/rolling/Concepts/About-Logging.html), for an end user it is not an obvious conclusion (without looking at source code) that [...]

yeah, that is true. i am aware of that.

> Another topic: are you aware of https://github.com/ros2/rcutils/issues/397?

I will take a look, thanks for the heads-up.

probably we would want to discuss on these topics with dedicated issue.

bijoua29, 2023-03-10:
> I think, given the current [logging documentation](http://docs.ros.org/en/rolling/Concepts/About-Logging.html), for an end user it is not an obvious conclusion (without looking at source code) that [...]

I'm curious. Is this difference in behavior described anywhere? If not, what are the general differences?

jrutgeer, 2023-03-10:
Refer to [this document](https://github.com/jrutgeer/ROS2_assorted_docs/blob/main/ROS2_core/ROS2_logging.md#a-few-peculiarities-about-the-rosout-publisher) that I recently wrote.

bijoua29, 2023-03-10:
> Refer to [this document](https://github.com/jrutgeer/ROS2_assorted_docs/blob/main/ROS2_core/ROS2_logging.md#a-few-peculiarities-about-the-rosout-publisher) that I recently wrote.

Perfec. Thanks for writing this documentation.

bijoua29, 2023-03-12:
Do you have a rough idea when this feature will be completed? Are you targeting a specific ROS distro release?

fujitatomoya, 2023-03-12:
we are targeting Iron release, https://docs.ros.org/en/rolling/The-ROS2-Project/Roadmap.html

bijoua29, 2023-03-12:
> we are targeting Iron release, https://docs.ros.org/en/rolling/The-ROS2-Project/Roadmap.html

That's great. Looking forward to using it.

fujitatomoya, 2023-03-22:
@llapx @Barry-Xu-2018 do we have PR for rclpy for this feature?

fujitatomoya, 2023-04-03:
@Barry-Xu-2018 please check the issue header https://github.com/ros2/ros2/issues/1355#issue-1477343045 if i miss anything related to this new feature?

fujitatomoya, 2023-04-07:
@mjcarroll @clalancette could you help us to merge the followings? CI is green, and they all must go in together.

- https://github.com/ros2/rcl_interfaces/pull/154
- https://github.com/ros2/rclcpp/pull/2122
- https://github.com/ros2/rclpy/pull/1102

Barry-Xu-2018, 2023-04-18:
The implementation of rclcpp and rclpy had been reviewed.   
So I updated demos to show how to use logger service.   
ros2/demos#611

Barry-Xu-2018, 2023-04-19:
Document change has been submitted.  
ros2/ros2_documentation#3477

fujitatomoya, 2023-07-14:
@clalancette @Barry-Xu-2018 i will go ahead to close this. all task has been completed.

closed 2023-07-14 by fujitatomoya