ros2/ros2 #1358: ros2 topic lose data in reliable mode

GazeboXu, 2022-12-12:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
  - ubuntu 
- Installation type:
  - binary
  - source
- Version or commit hash:
  - humble
  - foxy
- DDS implementation:
  - fast-dds
  - Eclipse Cyclone DDS 
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

pubsub.hpp
```
#ifndef PUBSUB_HPP_4C802155646547D98711A7D27337EB3E
#define PUBSUB_HPP_4C802155646547D98711A7D27337EB3E

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

#include <thread>
#include <memory>
#include <cstdint>
#include <mutex>
#include <condition_variable>

#define TOPIC_NAME  "HelloWorldTopic"
#define TEST4REAL
#define ROS2FOXY

class PubSub {
public:
    explicit PubSub(std::shared_ptr<rclcpp::Node> node, const std::string &topicName, int slaveCount = 0);

private:
    void OnSubCallback(const std_msgs::msg::String &info);
    void OnSubCallback2(const std_msgs::msg::String::SharedPtr msg);
    void run();
    std::shared_ptr<rclcpp::Node>   node_;
    std::mutex mutex_;
    std::condition_variable condition_;
    rclcpp::Publisher<std_msgs::msg::String>::SharedPtr     publisher_;
    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr  subscription_;
    int             slaveCount_;
    int             tmpCount_;
    std::uint64_t   index_;
    bool            isWaiting_;
    std::shared_ptr<std::thread>    thread_;

};
#endif
```

pubsub.cpp
```
#include "pubsub.hpp"
#include <chrono>

using namespace std::chrono_literals;
using std::placeholders::_1;

PubSub::PubSub(std::shared_ptr<rclcpp::Node> node, const std::string &topicName, int slaveCount)
    : node_(node), slaveCount_(slaveCount), index_(0), isWaiting_(false) {
    rclcpp::PublisherOptions pubOptions;

#ifndef ROS2FOXY
    pubOptions.use_default_callbacks = false;
    pubOptions.event_callbacks.deadline_callback = [this](rclcpp::QOSDeadlineOfferedInfo &lostInfo) {
        RCLCPP_WARN(this->node_->get_logger(), "publisher deadline number: %d", lostInfo.total_count);
    };
    pubOptions.event_callbacks.liveliness_callback = [this](rclcpp::QOSLivelinessLostInfo &lostInfo) {
        RCLCPP_WARN(this->node_->get_logger(), "publisher liveliness number: %d", lostInfo.total_count);
    };
#endif

    rclcpp::QoS qos(10);
    qos.reliable();
    qos.transient_local();
    qos.keep_all();
    publisher_ = node->create_publisher<std_msgs::msg::String>(topicName, qos, pubOptions);

    rclcpp::SubscriptionOptions subOptions;
#ifndef ROS2FOXY
    subOptions.use_default_callbacks = false;
    subOptions.event_callbacks.message_lost_callback = [this](rclcpp::QOSMessageLostInfo &lostInfo) {
        RCLCPP_WARN(this->node_->get_logger(), "subscibtion lost number: %ld", lostInfo.total_count);
    };
    subOptions.event_callbacks.deadline_callback = [this](rclcpp::QOSDeadlineRequestedInfo &lostInfo) {
        RCLCPP_WARN(this->node_->get_logger(), "subscibtion deadline number: %d", lostInfo.total_count);
    };
    subOptions.event_callbacks.liveliness_callback = [this](rclcpp::QOSLivelinessChangedInfo &lostInfo) {
        RCLCPP_WARN(this->node_->get_logger(), "subscibtion liveliness number: %d", lostInfo.not_alive_count);
    };
    subscription_ = node_->create_subscription<std_msgs::msg::String>(topicName, qos, std::bind(&PubSub::OnSubCallback, this, _1), subOptions);
#else
    subscription_ = node_->create_subscription<std_msgs::msg::String>(topicName, qos, std::bind(&PubSub::OnSubCallback2, this, _1), subOptions);
#endif

    if(slaveCount_ > 0) {
        thread_ = std::make_shared<std::thread>(std::bind(&PubSub::run, this));
    } else {
        RCLCPP_INFO(node_->get_logger(), "Subscriber %s ready", node_->get_name());
    }
}

void PubSub::run() {
    RCLCPP_INFO(node_->get_logger(), "Publisher preparing...");
    std::this_thread::sleep_for(5s);
    RCLCPP_INFO(node_->get_logger(), "Publisher ready");
    while(true) {
        tmpCount_ = 0;
        isWaiting_ = false;

        auto message = std_msgs::msg::String();
        message.data = "send " + std::to_string(++index_);

        RCLCPP_INFO(node_->get_logger(), "begin Publishing: '%s'", message.data.c_str());
        publisher_->publish(message);
        RCLCPP_INFO(node_->get_logger(), "end Publishing: '%s'", message.data.c_str());

        std::unique_lock<std::mutex> lock(mutex_);
        if(slaveCount_ != tmpCount_) {
            isWaiting_ = true;
            condition_.wait(lock);
        }
    }
}

void PubSub::OnSubCallback(const std_msgs::msg::String &info) {
    RCLCPP_INFO(node_->get_logger(), "enter OnSubCallback info: %s", info);

    auto it = info.data.find(" ");
    auto cmd = info.data.substr(0, it);
    if(slaveCount_ > 0 && cmd == "back") {
        int index = std::stoi(info.data.substr(++it));
        auto needNotify = false;
        {
            std::unique_lock<std::mutex> lock(mutex_);
            if(index != index_) {
                RCLCPP_ERROR(node_->get_logger(), "OnSubCallback %ld != %ld", index, index_);
                rclcpp::shutdown();
            }
            tmpCount_++;
            needNotify = tmpCount_ == slaveCount_;
        }
        if(needNotify && isWaiting_) {
            condition_.notify_one();
        }
    } else if(slaveCount_ == 0 && cmd == "send") {
        auto message = std_msgs::msg::String();
        message.data = "back " + info.data.substr(++it);
        publisher_->publish(message);
    }
    RCLCPP_INFO(node_->get_logger(), "leave OnSubCallback info: %s", info);
}

void PubSub::OnSubCallback2(const std_msgs::msg::String::SharedPtr msg) {
    OnSubCallback(*msg);
}
```
the full program is on github: [https://github.com/GazeboXu/cpp_pubsub.git](https://github.com/GazeboXu/cpp_pubsub.git)

after compiled the program, use command `ros2 run cpp_pubsub listener node1` and `ros2 run cpp_pubsub listener node2` the run two slaves, and then use command `ros2 run cpp_pubsub talker 2` to run a master.

#### Expected behavior
the talker and listener will run forever

#### Actual behavior
sometimes, the talker and listener will get stuck due to the message lose

#### Additional information
Hello, there. We recently are tortured by a stupid problem for days. We need help from you.

We use ROS2 topic to send command and get response, so we need the messages(command and response) not be dropped (or must be delivered successfully). Of course we use a reliable QoS, but unluck, we did encounter data lose. Our detail use scenario like this, suppose we have one master and two salves, we want a command to send to all the slaves, we need each slave send back the corresponding response, if the master can not get all responses, it will not forward.

Firstly, we doubt the bottom dds has problem, so we modified a fast dds example to do a test, one publisher send messages to two subscribers very quickly, we do observe message lose when use KEEP_LAST_HISTORY_QOS, after some investigation, we knew this behavior is reasonable as [https://www.rti.com/blog/top-10-reaso...](https://www.rti.com/blog/top-10-reasons-for-dropped-dds-messages) Number 2 said. After switching to KEEP_ALL_HISTORY_QOS, we do not observe message lose. And when switched to ROS2 topic, the test result is same. But because as I said above, we use a send ,wait and send mode, we think this should not be our case.

The test program is on github( [https://github.com/GazeboXu/cpp_pubsu...](https://github.com/GazeboXu/cpp_pubsub.git) ). there is a macro in file pubsub.hpp TEST4REAL controls whether use real mode(send and wait). When in real mode, firstly run two slaves(ros2 run cpp_pubsub listener node1. ros 2 run cpp_pubsub listener node2), then run a master(ros2 run cpp_subpub talker 2). eventually, the output will stop, you will find some messages didn't receive by some nodes). It looks like if when one publisher and some subscribers, the message will not lose, but if many publishers and many subcsribers, the message will lose.

Any advices will be appreciated!

GazeboXu, 2023-01-31:
It seems this bug  disappeared on the new release -- "ROS 2 Humble Hawksbill - Patch Release 2".

clalancette, 2023-01-31:
> It seems this bug disappeared on the new release -- "ROS 2 Humble Hawksbill - Patch Release 2".

Thanks for the follow-up, it is appreciated.  In that case, I'm going to close this out for now.  If it occurs again, please feel free to reopen.

GazeboXu, 2023-02-01:
@clalancette , thanks for your hard work, but from the changelog of rclcpp, I didn't find relative info. Anyway, the bug seems dsiappeared now.