ros2/ros2 #1359: Improve handling of dynamic discovery

gbiggs, 2022-12-14:
In ROS 2 Iron Irwini, we hope to improve how discovery is managed by ROS 2. In brief:

- Broadcast discovery "distance" will be controllable by the user, in terms of off, localhost-only, and subnet-only.
- Discovery static peers will be easily specified by the user, allowing specific hosts to be discovered even when broadcast discovery is off.

For the full details, please see [this Discourse post](https://discourse.ros.org/t/proposed-changes-to-how-ros-performs-discovery-of-nodes/27640).

Below are the individual PRs that make up this feature.

- [ ] [rcl#1023](https://github.com/ros2/rcl/pull/1023)
- [ ] [rmw#338](https://github.com/ros2/rmw/pull/338)
- [ ] [rmw_fastrtps#653](https://github.com/ros2/rmw_fastrtps/pull/653)
- [ ] [rmw_cyclonedds#429](https://github.com/ros2/rmw_cyclonedds/pull/429)

arjo129, 2023-01-31:
Work for this is going on here: https://github.com/osrf/iron_irwini_discovery_modes_prototype/pull/2

wjwwood, 2023-03-29:
- Decide if we should keep or delete RMW_AUTOMATIC_DISCOVERY_RANGE_DEFAULT https://github.com/ros2/rmw/pull/338#discussion_r1093863765
  - decision from @sloretz and I speaking off-line, we'll make a new `SYSTEM_DEFAULT` enum which delegates configuration to the xml config of the middleware you're using, and if no enum is set, then we'll use `LOCALHOST` (which is set at the `rcl` level if no env var is set and/or options in the discovery options struct passed to context init)
- Fast-DDS ignore_participant implementation merged: https://github.com/eProsima/Fast-DDS/pull/3412
  - @sloretz try an idea to not need `ignore_participant`, if that doesn't pan out...
  - @wjwwood to try running without Fast-DDS patch, to see what doesn't work, in the hope that we could merge with these known issues and get this Fast-DDS change later to fix them
  - If neither are ok, then we just try to get the Fast-DDS stuff merged

sloretz, 2023-03-29:
Windows canary - building and testing all packages up to rcl (this build excludes rmw_connextdds and test_discovery) [image]

sloretz, 2023-03-31:
Another Windows canary - testing `test_discovery`
[image]

GIST:  https://gist.github.com/sloretz/37e0a5eeecda22b56f5ca70874e6281e

sloretz, 2023-03-31:
Full CI excluding connext

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-03-31:
Another full build excluding connext - this time seeing the affect of some improvements to rmw_fastrtps_*

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-03-31:
Targeted windows CI looking for more info on CycloneDDS test failures (build: `--packages-up-to test_communication test_discovery` test: `--packages-select test_communication test_discovery`, special repos file, only cyclonedds): [image]

Full CI (first one with connext changes!) to see where everything is after today's changes
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-03-31:
Full CI re-run with changes to rmw_cyclonedds

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-03-31:
Rurun again with changes to rmw_connextdds
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-04-01:
After making connext change, full CI excluding Cyclone since I'm seeing strange errors on windows with Cyclone

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-04-04:
Full CI, this time with all three RMW implementations, after making the change of the maximum participant ID to 32.

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

clalancette, 2023-04-04:
Just as an FYI, we also need https://github.com/ros/ros_environment/tree/gbiggs/discovery-peers-specification so we don't get a warning on every command:

```
[WARN] [1680633323.129234187] [rcl]: ROS_LOCALHOST_ONLY is deprecated. Use ROS_AUTOMATIC_DISCOVERY_RANGE and ROS_STATIC_PEERS instead.
```

I'll go ahead and open a PR with that change.

clalancette, 2023-04-04:
I did some testing here with RMW_IMPLEMENTATION=rmw_connextdds and the set of PRs listed.  In my particular case, I had two different containers, with two different IP addresses on the same bridged network (subnet).  I then ran a talker in each container, and used the `ros2` command-line tools to do various pieces of introspection.

For the most part, things work as advertised.  By default, neither container could see the topics and nodes from the other one.  Adding in `ROS_STATIC_PEERS` for the IP address of the other container made them be able to discover each other and transfer data.

However, there is some weirdness around the `ros2daemon`.  In particular, when it initially launches, it launches with the environment of whichever command first needed it.  So, for instance, if you run `ros2 node list` with `ROS_STATIC_PEERS` unset initially, then it won't find nodes from the other container (as expected).  But if you then do `ROS_STATIC_PEERS=<IP> ros2 node list`, it *still* won't find nodes from the other container until you do a `ros2 daemon stop`, and then run the command again.  The problem happens in reverse as well.

I think we need a follow-up here where we have `ros2daemon` detect if the `ROS_STATIC_PEERS` and `ROS_AUTOMATIC_DISCOVERY_RANGE` are different from when it started, and restart itself if needed.

sloretz, 2023-04-05:
test_launch_ros tests are hanging with rmw_fastrtps (increasing maxAutoParticipantIndex for rmw_fastrtps fixes it, but neither Cyclone nor Connext need the same increase), so tonight's CI is with Cyclone and connext, with Cyclone on the 0.10.x branch to see if that resolves the issues on windows

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-04-06:
Full CI tonight
* [repos file](https://gist.githubusercontent.com/sloretz/37e0a5eeecda22b56f5ca70874e6281e/raw/2ffd0ef6d9c95635d29b3520a3619cef65e370e5/test_discovery_oobe.repos)
* Includes https://github.com/eProsima/Fast-DDS/pull/3437
* Includes https://github.com/ros2/ros2/pull/1402

Jobs

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-04-06:
There are still some issues with the discovery range being Localhost
* With Fast-DDS, creating and deleting participants causes the participant ID to climb without limit https://github.com/eProsima/Fast-DDS/pull/3437
   * This is probably only a problem with large amounts of unit tests in a single process that create/delete contexts with overlap - like test_launch_ros
* With CycloneDDS, even after bumping to 0.10.x, there are lots of test failures on Windows where Cyclone can no longer send unicast discovery messages

As a workaround I've changed the default discovery range to SUBNET - which is the same as the current behavior, but with the ability to set static peers. I'm going to try full CI to see if that would be an option to get these changes in.
*  ros_environment: https://github.com/ros/ros_environment/pull/37/commits/c5d513f6b9ab560cb1a25c8ad0cb8df0099ab804
* rcl: https://github.com/ros2/rcl/pull/1023/commits/ee00c56c7a230520e1965c3e9528b7fef8af2a99

Jobs

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

sloretz, 2023-04-06:
RHEL and Windows Debug full CI

* RHEL: [image]
* Windows Debug: [image]

clalancette, 2023-04-06:
> As a workaround I've changed the default discovery range to SUBNET - which is the same as the current behavior, but with the ability to set static peers. I'm going to try full CI to see if that would be an option to get these changes in.

Just as a follow-up; @sloretz and I talked about this in-person, and I think getting this code in but leaving it as SUBNET is a good compromise for now.  Then we can consider fixing the issues for J-Turtle, and at that point changing the default to LOCALHOST.

sloretz, 2023-04-07:
Full CI plus after pulling rolling changes and fixing warning on windows in rmw_connextdds_common

[repos file](https://gist.githubusercontent.com/sloretz/bdeaeba4134b79309dff7b3302114940/raw/8048fe8429416f5138e2c7caeefbbf23fad048fd/test_dds_oobe_pre_rmw_freeze.repos)

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]
* RHEL: [image]
* Windows debug: [image]

sloretz, 2023-04-07:
Full CI - this time with Fast-DDS 2.10.x

[repos file](https://gist.githubusercontent.com/sloretz/bdeaeba4134b79309dff7b3302114940/raw/75e29892de53a792c61b7efda3edc7eb445de5c8/test_dds_oobe_pre_rmw_freeze.repos)

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]
* RHEL [image]
* Windows Debug: Windows [image]

sloretz, 2023-04-08:
The only windows test [failure appears to be flaky on the nightly jobs](https://ci.ros2.org/view/nightly/job/nightly_win_rep/2951/testReport/junit/launch_testing_examples.launch_testing_examples/check_multiple_nodes_launch_test/launch_testing_examples_check_multiple_nodes_launch_test/history/) CI LGTM :tada:

clalancette, 2023-04-10:
@sloretz @gbiggs @mxgrey @arjo129 Thank you so much for all of your work getting this in!

If I can make two more requests:
1. Add a release note to https://github.com/ros2/ros2_documentation/blob/rolling/source/Releases/Release-Iron-Irwini.rst explaining what this work is and what the changes are from Humble.
2. Close this issue, and open up a new one explaining the follow-up work to be done to make this work with LOCALHOST as the default.

fujitatomoya, 2023-05-15:
> ROS 2 documentation about discovery options

https://github.com/ros2/ros2_documentation/pull/3596

> ROS Iron release notes for discovery change

https://github.com/ros2/ros2_documentation/pull/3519

fujitatomoya, 2023-06-10:
@Yadunund i think we can close this.

closed 2023-06-11 by Yadunund