ros2/ros2 #1360: ROS 2 maintainer reassignment or update?

fujitatomoya, 2022-12-19:
## Question

Since [OSRC Team is Joining Intrinsic](https://discourse.ros.org/t/the-osrc-team-is-joining-intrinsic-and-what-it-means-for-the-ros-community/28764) is taking place, we will reassign or update the maintainer for ROS 2 such as `rmw`, `rcl`, `rclcpp` and so on? probably maintainers can be the same people but email addresses in the `pacakge.xml`s need to be update?

### Related Resource

- https://discourse.ros.org/t/the-osrc-team-is-joining-intrinsic-and-what-it-means-for-the-ros-community/28764

CC: @tfoote @clalancette

clalancette, 2022-12-19:
Let's hold off on doing anything here until the new year.  We'll have a better idea on what we need to do then.

fujitatomoya, 2022-12-19:
No pressure, just checking.

gbiggs, 2022-12-20:
In the short term, the maintainers will remain as they currently are, except where someone is no longer able to perform that maintainer duty. In the longer term, OSRF is setting up a more structured process for determining maintainers in order to involve the wider community more actively. We cannot say exactly when this process will be in place yet as there are many things currently going on that require our attention first.