ros2/ros2 #1362: SetuptoolsDeprecationWarning: setup.py install is deprecated

Hongbo-Miao, 2022-12-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Humble

#### Steps to reproduce issue

```shell
ros2 pkg create --build-type=ament_python --node-name=hm_python_node --dependencies=rclpy hm_python_package
colcon build
```

#### Expected behavior

No warning.

#### Actual behavior

Because `setup.py` is deprecated, so I got warning:

```shell
colcon build --packages-select=hm_python_package
Starting >>> hm_python_package
--- stderr: hm_python_package                   
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< hm_python_package [0.64s]

Summary: 1 package finished [0.77s]
  1 package had stderr output: hm_python_package
```

#### Implementation considerations

Here has more info about this deprecated `setup.py`: https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html
It is not something critical, but it would be great to get rid of it. 😃

ijnek, 2022-12-24:
This seems to be a duplicate of https://github.com/ament/ament_cmake/issues/382

Hongbo-Miao, 2022-12-24:
Thanks @ijnek ! Yes, it is. 😃

closed 2022-12-24 by Hongbo-Miao