ros2/ros2 #1364: ROS2 humble desktop installation causes unmet dependencies issue

RandyChen233, 2023-01-02:
I know this issue was discussed [here](https://github.com/ros-drivers/joystick_drivers/issues/241) and [here](https://github.com/ros2/ros2/issues/1272) . However, even after upgrading `apt`, I was still not able to install ROS Humble. 
Here is my Ubuntu version:
```
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
```
Here is how the error pops up : 
```
(ICON_lab) randychen@randychen-Delta-15-A5EFK:~/projects$ sudo apt update && sudo apt upgrade
Hit:1 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease                                            
Hit:3 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease                      
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
#
# News about significant security updates, features and services will
# appear here to raise awareness and perhaps tease /r/Linux ;)
# Use 'pro config set apt_news=false' to hide this and future APT news.
#
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(ICON_lab) randychen@randychen-Delta-15-A5EFK:~/projects$ sudo apt install ros-humble-desktop
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libarmadillo10 : Depends: liblapack3 but it is not installable or
                           liblapack.so.3
 libarpack2 : Depends: libgfortran5 (>= 8) but it is not installable
              Depends: liblapack3 but it is not installable or
                       liblapack.so.3
 libarpack2-dev : Depends: liblapack-dev but it is not installable or
                           liblapack.so
 libdbus-1-dev : Depends: libdbus-1-3 (= 1.12.20-2ubuntu4) but 1.12.20-2ubuntu4.1 is to be installed
 libglib2.0-dev : Depends: libglib2.0-0 (= 2.72.1-1) but 2.72.4-0ubuntu1 is to be installed
                  Depends: libglib2.0-bin (= 2.72.1-1)
                  Depends: libglib2.0-dev-bin (= 2.72.1-1)
 libhdf5-fortran-102 : Depends: libgfortran5 (>= 8) but it is not installable
 libhdf5-hl-fortran-100 : Depends: libgfortran5 (>= 8) but it is not installable
 libhdf5-openmpi-fortran-102 : Depends: libgfortran5 (>= 8) but it is not installable
 libhdf5-openmpi-hl-fortran-100 : Depends: libgfortran5 (>= 8) but it is not installable
 libjson-c-dev : Depends: libjson-c5 (= 0.15-2build4) but 0.15-3~ubuntu1.22.04.1 is to be installed
 libopenmpi-dev : Depends: gfortran-11 but it is not installable or
                           gfortran-mod-15
                  Recommends: libcoarrays-openmpi-dev but it is not installable
 libpcre2-dev : Depends: libpcre2-8-0 (= 10.39-3build1) but 10.39-3ubuntu0.1 is to be installed
                Depends: libpcre2-32-0 (= 10.39-3build1) but 10.39-3ubuntu0.1 is to be installed
 libpoppler-dev : Depends: libpoppler118 (= 22.02.0-2) but 22.02.0-2ubuntu0.1 is to be installed
 libpq-dev : Depends: libpq5 (= 14.2-1ubuntu1) but 14.5-0ubuntu0.22.04.1 is to be installed
 libpulse-dev : Depends: libpulse0 (= 1:15.99.1+dfsg1-1ubuntu1) but 1:15.99.1+dfsg1-1ubuntu2 is to be installed
                Depends: libpulse-mainloop-glib0 (= 1:15.99.1+dfsg1-1ubuntu1) but 1:15.99.1+dfsg1-1ubuntu2 is to be installed
 libtiff-dev : Depends: libjbig-dev but it is not going to be installed
               Depends: libtiff5 (= 4.3.0-6) but 4.3.0-6ubuntu0.3 is to be installed
 libudev-dev : Depends: libudev1 (= 249.11-0ubuntu3) but 249.11-0ubuntu3.6 is to be installed
 libusb-1.0-0-dev : Depends: libusb-1.0-0 (= 2:1.0.25-1ubuntu1) but 2:1.0.25-1ubuntu2 is to be installed
                    Recommends: libusb-1.0-doc but it is not going to be installed
 libvtk9-dev : Depends: libfreetype6-dev
 libwayland-dev : Depends: libwayland-client0 (= 1.20.0-1) but 1.20.0-1ubuntu0.1 is to be installed
                  Depends: libwayland-server0 (= 1.20.0-1) but 1.20.0-1ubuntu0.1 is to be installed
                  Depends: libwayland-cursor0 (= 1.20.0-1) but 1.20.0-1ubuntu0.1 is to be installed
                  Depends: libwayland-egl1 (= 1.20.0-1) but 1.20.0-1ubuntu0.1 is to be installed
 libxft-dev : Depends: libfontconfig1-dev
              Depends: libfreetype6-dev
 libxml2-dev : Depends: libxml2 (= 2.9.13+dfsg-1build1) but 2.9.13+dfsg-1ubuntu0.2 is to be installed
 python3-numpy : Depends: liblapack3 but it is not installable or
                          liblapack.so.3
 python3-pil.imagetk : Depends: python3-pil (= 9.0.1-1build1) but 9.0.1-1ubuntu0.1 is to be installed
 ros-humble-libcurl-vendor : Depends: libcurl4-openssl-dev but it is not installable
 ros-humble-rviz-ogre-vendor : Depends: libfreetype6-dev
E: Unable to correct problems, you have held broken packages.

```
Any help is greatly appreciated!

labiybafakh, 2023-04-18:
For this problem, I think we should install all of those dependencies manually.

binarychen, 2025-09-13:
Manually install the dependencies:
e.g.
```
sudo apt install libusb-1.0-0=2:1.0.25-1ubuntu1
sudo apt install libusb-1.0-0-dev
```
```
sudo apt install libpulse0=1:15.99.1+dfsg1-1ubuntu1
sudo apt install libpulse-mainloop-glib0=1:15.99.1+dfsg1-1ubuntu1
```