ros2/ros2 #1366: ROS2 CALLBACK DONT TRIGGRED IN GAZEBO PLUGIN

omr-web, 2023-01-09:
I wrote a simple model plugin in GAZEBO-11. This plugin subscribe a "topic". The topic named "topic" is publishing a float data. I use ROS2 example publisher codes. When I launch publisher and gazebo, model plugin's callback sometimes don't triggred and I write to terminal  "ros2 topic echo /topic", callback start to work. Sometimes I launch publisher and gazebo-11 callback start normally but after 1-2 minutes callback function stopped,gazebo plugin continue to work.  I think the problem is caused by the model plugin because when the ros2 subscriber code runs, the callback there never stops. Here my publisher and model plugin codes.

Publisher .cpp :
==================================================
class MinimalPublisher : public rclcpp::Node
{
    public:
      float i=0;
      MinimalPublisher()
      : Node("minimal_publisher"), count_(0)
      {
        publisher_ = this->create_publisher<std_msgs::msg::Float32>("topic", 10);
        timer_ = this->create_wall_timer(
        500ms, std::bind(&MinimalPublisher::timer_callback, this));
    }

  private:
    void timer_callback()
    {
      std_msgs::msg::Float32 data;
      data.data=i++;
      RCLCPP_INFO(this->get_logger(), "Publishing: '%f'",data.data);
      publisher_->publish(data);
    }
    rclcpp::TimerBase::SharedPtr timer_;
    rclcpp::Publisher<std_msgs::msg::Float32>::SharedPtr publisher_;
    size_t count_;
 };

int main(int argc, char * argv[])
 {
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalPublisher>());
  rclcpp::shutdown();
  return 0;
 }

Model Plugin Code .cpp :
==================================================

#include "plugin_deneme.h"

int i = 0;
namespace gazebo
{

DenemePlugin::DenemePlugin()
{
        printf("const");
}

void DenemePlugin::FloatVeriCallback(std_msgs::msg::Float32 msg)
{
        RCLCPP_INFO(node_->get_logger(), " PLUGIN  CALLBACK %f \n:", msg.data);
}

void DenemePlugin::Load(gazebo::physics::ModelPtr model, sdf::ElementPtr sdf) {

    node_ = gazebo_ros::Node::Get(sdf);
    RCLCPP_INFO(node_->get_logger(), "  LOAD \n:");
    printf("load\n");

  
    model_ = model;
    world_ = model_->GetWorld();

    float_veri_subs = node_->create_subscription<std_msgs::msg::Float32>("topic", 10,
            std::bind(&DenemePlugin::FloatVeriCallback, this, std::placeholders::_1));

    // Hook into simulation update loop
    update_connection_ = gazebo::event::Events::ConnectWorldUpdateBegin(
            std::bind(&DenemePlugin::Update, this));
}

void DenemePlugin::Update() {

  RCLCPP_INFO(node_->get_logger(), "  UPDATE %f \n:", i++);
}

GZ_REGISTER_MODEL_PLUGIN(DenemePlugin)
}

Model Plugin .h:
========================================================
namespace gazebo {

class DenemePlugin : public gazebo::ModelPlugin {
    
    public: 
    
    
        DenemePlugin();

        void Load(gazebo::physics::ModelPtr model, sdf::ElementPtr sdf);
        void Update();

        rclcpp::Subscription<std_msgs::msg::Float32>::SharedPtr float_veri_subs;
        void FloatVeriCallback(std_msgs::msg::Float32 x);

        //gazebo-ros node
        gazebo_ros::Node::SharedPtr node_;

        
        gazebo::physics::ModelPtr model_;
        gazebo::physics::WorldPtr world_;

         gazebo::event::ConnectionPtr update_connection_;
  
  };

}

#endif

==========================================================

- Operating System:
  - Ubuntu 22.04
- Version or commit hash:
  - GAZEBO-11, ROS2 HUMBLE
- DDS implementation:
  -Fast RTPS

#### Expected behavior ###
 When the model plugin code starts to run, the callback function in the plugin works as long as the publisher publishes data.

#### Actual behavior
When the model plugin code starts running, the callback function doesn't work at first. But when the ros2 topic echo command is entered from the terminal ,the callback function starts to work . Sometimes it works properly but after a few minutes the callback is not triggered.

vLevente, 2023-01-20:
I have the same issue with the [Ackermann gazebo plugin](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/ros2/gazebo_plugins/src/gazebo_ros_ackermann_drive.cpp). At the beginning of the simulation it can receive cmd_vel messages and after a couple of seconds for some reason it just stops.

clalancette, 2023-02-09:
Please open questions like this on https://answers.ros.org, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

If it turns out to be a bug, feel free to open an issue on an appropriate repository.

mrjoe3012, 2023-03-04:
I see that this has been closed, but this is an issue. The plugin stops receiving callbacks (for timers and subscribers) until `ros2 topic echo` or sometimes `ros2 topic list` is called.