ros2/ros2 #1368: Issue: Failed to install package `mimick_vendor`

ryankshah, 2023-01-11:
## Bug report

**Required Info:**

- Operating System:
  - Windows 11
- Installation type:
  - From Source
- Version or commit hash:
  - For `mimick_vendor` I ran `git log -n 4 --oneline` and it output:

```
39ec45b (HEAD -> humble, tag: 0.2.8, origin/humble) 0.2.8
774396a Changelog
5826e15 support pi zero (#24)
7c2d2c7 (tag: 0.2.7) 0.2.7
```

- DDS implementation:
  -  N/A (default)
- Client library (if applicable):
  -  N/A

#### Steps to reproduce issue
Follow the documentation steps exactly as described from: https://docs.ros.org/en/humble/Installation/Alternatives/Windows-Development-Setup.html#system-requirements

This is the latter of the installation output:

```
Starting >>> tlsf
Finished <<< qt_gui [1min 22s]
Starting >>> qt_dotgraph
Finished <<< orocos_kdl_vendor [1min 58s]
Starting >>> mcap_vendor
[798.443s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'tlsf' because it has no 'install' target
Finished <<< tlsf [55.7s]
Starting >>> rttest
Finished <<< console_bridge_vendor [2min 7s]
Starting >>> urdfdom
Failed   <<< mimick_vendor [2min 4s, exited with code 1]
Aborted  <<< qt_dotgraph [50.9s]
Aborted  <<< urdfdom [6.03s]
Aborted  <<< qt_gui_py_common [1min 18s]
Aborted  <<< rttest [17.7s]
Aborted  <<< mcap_vendor [1min 8s]
Aborted  <<< libcurl_vendor [11min 40s]
Aborted  <<< rviz_ogre_vendor [11min 38s]

Summary: 109 packages finished [15min 14s]
  1 package failed: mimick_vendor
  7 packages aborted: libcurl_vendor mcap_vendor qt_dotgraph qt_gui_py_common rttest rviz_ogre_vendor urdfdom
  28 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator launch launch_testing launch_xml launch_yaml osrf_pycommon rosidl_cli rpyutils rti_connext_dds_cmake_module tracetools_read tracetools_trace
  226 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```

#### Expected behavior

Clean installation from source

#### Actual behavior

Outputs the following error:

```
Starting >>> tlsf
Finished <<< qt_gui [1min 22s]
Starting >>> qt_dotgraph
Finished <<< orocos_kdl_vendor [1min 58s]
Starting >>> mcap_vendor
[798.443s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'tlsf' because it has no 'install' target
Finished <<< tlsf [55.7s]
Starting >>> rttest
Finished <<< console_bridge_vendor [2min 7s]
Starting >>> urdfdom
Failed   <<< mimick_vendor [2min 4s, exited with code 1]
Aborted  <<< qt_dotgraph [50.9s]
Aborted  <<< urdfdom [6.03s]
Aborted  <<< qt_gui_py_common [1min 18s]
Aborted  <<< rttest [17.7s]
Aborted  <<< mcap_vendor [1min 8s]
Aborted  <<< libcurl_vendor [11min 40s]
Aborted  <<< rviz_ogre_vendor [11min 38s]

Summary: 109 packages finished [15min 14s]
  1 package failed: mimick_vendor
  7 packages aborted: libcurl_vendor mcap_vendor qt_dotgraph qt_gui_py_common rttest rviz_ogre_vendor urdfdom
  28 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator launch launch_testing launch_xml launch_yaml osrf_pycommon rosidl_cli rpyutils rti_connext_dds_cmake_module tracetools_read tracetools_trace
  226 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```

clalancette, 2023-01-11:
We don't officially support Windows 11, so you are in uncharted territory here.  I don't have a solution for you, but if you find out how to fix it, please feel free to open a pull request.  We're happy to review.

BlackXDragon, 2023-03-02:
Is there any update on how to build on Windows 11? Seems like it would benefit a lot of people to provide Windows 11 support.

clalancette, 2023-03-02:
> Is there any update on how to build on Windows 11? Seems like it would benefit a lot of people to provide Windows 11 support.

Unfortunately not.  At the moment we can't switch to Windows 11 since we have some infrastructure issues preventing CI from running.  That is being worked on, so hopefully that will be an option soon.  That said, this likely will come too late for Iron (due in May 2023), so Windows 11 support will be a J-Turtle feature.

LastStarDust, 2023-06-03:
I am experiencing this very same problem in Windows 10 as well.

james-yoo, 2023-09-22:
I could build mimick_vendor on Windows 11 with VS 2019 Community. This error is due to the short default Windows path limits (260 characters), so I resolved it by allowing a longer path as described [here](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry).

Use the PowerShell command from a terminal window with elevated privileges:
```
New-ItemProperty -Path "HKLM:\SYSTEM\CurrentControlSet\Control\FileSystem" `
-Name "LongPathsEnabled" -Value 1 -PropertyType DWORD -Force
```

clalancette, 2023-09-27:
@james-yoo Thank you, that is very helpful.  We already have this advice in http://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#character-path-limit, so I'm going to go ahead and close this out.  Feel free to comment or reopen if you are still having trouble.

closed 2023-09-27 by clalancette

harshbhatt99, 2023-10-08:
Even after setting up the maximum path length, I am still getting the same error. I am installing ROS2 Humble on Windows 10.
After installing all pre-requisites from [https://docs.ros.org/en/humble/Installation/Alternatives/Windows-Development-Setup.html](url) I am running the build command.
`colcon build --merge-install`

And I am getting this error.

`Summary: 110 packages finished [2min 32s]
  1 package failed: mimick_vendor
  7 packages aborted: ament_cmake_clang_format ament_cmake_mypy qt_dotgraph qt_gui resource_retriever tlsf urdfdom
  27 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint domain_coordinator launch launch_testing launch_xml launch_yaml osrf_pycommon rosidl_cli rpyutils tracetools_read tracetools_trace
  226 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)`

Any alternate solution?