ros2/ros2 #1370: ros2 command giving TypeErrror

lonebots, 2023-01-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 LTS
- Installation type:
  - source installation
- Version or commit hash:
  
  -  - 
- DDS implementation:
  
  -  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

To reproduce the error install ros2 Humble Hawksbill and type the command in the terminal
```
ros2
```

#### Expected behavior
list all the possible flags and options for running ros2

#### Actual behavior
It gives an error, as given below.

```bash
~$ ros2
Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.4', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 50, in main
    add_subparsers_on_demand(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/command/__init__.py", line 188, in add_subparsers_on_demand
    entry_points = get_entry_points(group_name)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/entry_points.py", line 66, in get_entry_points
    entry_points_impl = importlib_metadata.entry_points()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 1009, in entry_points
    return SelectableGroups.load(eps).select(**params)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 459, in load
    ordered = sorted(eps, key=by_group)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 1007, in <genexpr>
    dist.entry_points for dist in unique(distributions())
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 631, in entry_points
    return EntryPoints._from_text_for(self.read_text('entry_points.txt'), self)
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 381, in _from_text_for
    return cls(ep._for(dist) for ep in cls._from_text(text))
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 381, in <genexpr>
    return cls(ep._for(dist) for ep in cls._from_text(text))
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 389, in <genexpr>
    return (
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 103, in <genexpr>
    section._replace(value=Pair.parse(section.value))
  File "/usr/lib/python3.10/importlib/metadata/_collections.py", line 30, in parse
    return cls(*map(str.strip, text.split("=", 1)))
TypeError: Pair.__new__() missing 1 required positional argument: 'value'
```

----

tfoote, 2023-01-17:
Can you please provide a full description of how you have setup your workspace. Including what you've installed and where you've installed it? Also have you done anything to your python path and other things to setup your environment. 

I can not reproduce your problem. 

If you can provde a minimal docker file which shows this it would be highly informative. 

I tested this against the most recent humble release as follows

```
❯ rocker ros:humble
Extension volume doesn't support default arguments. Please extend it.
Active extensions []
Writing dockerfile to /tmp/tmpotz9083s/Dockerfile
vvvvvv

FROM ros:humble
USER root

^^^^^^
Building docker file with arguments:  {'path': '/tmp/tmpotz9083s', 'rm': True, 'nocache': False, 'pull': False}
building > Step 1/2 : FROM ros:humble
building >  ---> c144c1e970a3
building > Step 2/2 : USER root
building >  ---> Running in 27fd47869b2b
building > Removing intermediate container 27fd47869b2b
building >  ---> a8e8e762d2b1
building > Successfully built a8e8e762d2b1
Executing command: 
docker run --rm -it a8e8e762d2b1 
root@2dcae6a18c0a:/# ros2
usage: ros2 [-h] [--use-python-default-buffering]
            Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

options:
  -h, --help            show this help message and exit
  --use-python-default-buffering
                        Do not force line buffering in stdout and instead use
                        the python default buffering, which might be affected
                        by PYTHONUNBUFFERED/-u and depends on whatever stdout
                        is interactive or not

Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
  daemon     Various daemon related sub-commands
  doctor     Check ROS setup and other potential issues
  interface  Show information about ROS interfaces
  launch     Run a launch file
  lifecycle  Various lifecycle related sub-commands
  multicast  Various multicast related sub-commands
  node       Various node related sub-commands
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  topic      Various topic related sub-commands
  wtf        Use `wtf` as alias to `doctor`

  Call `ros2 <command> -h` for more detailed usage.
root@2dcae6a18c0a:/# 
```

I"ll also note that I saw you posted something that looks like it could be a similar issue with python metadata having invalid syntax: https://robotics.stackexchange.com/questions/24306/colcon-build-exception 

And you've recently been changing versions of setuptools: https://robotics.stackexchange.com/questions/24230/setuptoolsdeprecationwarning-in-ros2-humble You'll need to document what you've done well enough to reproduce your issue for people here to be able to help you.

lonebots, 2023-01-17:
I have the ros2 humble directly installed in my ubuntu system, it is not on the docker. I  followed the official ros2 documentation [installation](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html) guide. Then I set the following environment variables in the `.bashrc` file 
```bash
# ros-humble-source
source /opt/ros/humble/setup.bash

# colcon_cd command setup
source /usr/share/colcon_cd/function/colcon_cd.sh
export _colcon_cd_root=/opt/ros/humble/

# colcon tab completion
# source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash
source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash
```

Apart from this basic setup, I haven't manually altered any of the configurations or python3 paths. 

> And you've recently been changing versions of setuptools: https://robotics.stackexchange.com/questions/24230/setuptoolsdeprecationwarning-in-ros2-humble 

Yeah, I had this issue of deprecation warning, which I was able to resolve after the downgrading the python3 `setuptools` package. but later on I have upgraded it again and followed build process with the `warning`, 

> I"ll also note that I saw you posted something that looks like it could be a similar issue with python metadata having invalid syntax: https://robotics.stackexchange.com/questions/24306/colcon-build-exception

The issue appeared with `colcon build`  first, and later when I tried using the `ros2` command, I was getting this error mentioned in this issue. 

I didn't understand how this occurred, but I will try to figure it out and help you reproduce it.  I tried following methods to rectify it, but nothing worked for me.
 - I uninstalled and reinstalled the `ros2-humble-hawksbill` in my local system.
 - Tried removing python3-colcon-common-extensions` package and install it again

lonebots, 2023-01-17:
I have found out that this is not a bug and instead it was caused by a `source` added to the `.bashrc` file pointing a custom ros2 workspace that I have created. Thank you. The issue got resolved once I removed those lines from the bash file and run the `ros2` command from a fresh terminal. Thank you!

closed 2023-01-17 by clalancette