ros2/ros2 #1372: Building Rolling and Humble from source fails with

Achllle, 2023-01-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu22.04 (Jammy)
- Installation type:
  - source
- Version or commit hash:
  - [ros2.repos commit](https://github.com/ros2/ros2/commit/07c27c934a6a1269fec2828f4c1a4ff40430d44d)
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Essentially follow the instructions at [Linux Development Setup for Rolling](https://index.ros.org/doc/ros2/Installation/Rolling/Linux-Development-Setup/). I created a Dockerfile that executes these in [this gist](https://gist.github.com/Achllle/e55a5f34c02720963898ef807ae396ac)

#### Expected behavior

Build completes and the example can be run inside the container

#### Actual behavior

`colcon build --symlink-install` fails at `FortranCInterface` which seems to be a CMake dependency. 

Before that, I get several variations of this error which seems unrelated:
```
#0 1.686 Traceback (most recent call last):
#0 1.686   File "<string>", line 1, in <module>
#0 1.686   File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 10, in <module>
#0 1.686     import distutils.core
#0 1.686   File "/usr/lib/python3/dist-packages/numpy/distutils/__init__.py", line 24, in <module>
#0 1.686     from . import ccompiler
[2023-01-24T17:40:35.425Z] #0 1.686   File "/usr/lib/python3/dist-packages/numpy/distutils/ccompiler.py", line 9, in <module>
#0 1.686     from distutils.ccompiler import (
#0 1.686 ImportError: cannot import name 'compiler_class' from partially initialized module 'distutils.ccompiler' (most likely due to a circular import) (/usr/lib/python3/dist-packages/numpy/distutils/ccompiler.py)
#0 1.690 [1.143s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'python_setup_py' in 'lib/python3/dist-packages/numpy': Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
#0 1.690 Traceback (most recent call last):
#0 1.690   File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 142, in _identify
#0 1.690     retval = extension.identify(_reused_descriptor_instance)
#0 1.690   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 48, in identify
#0 1.690     config = get_setup_information(setup_py)
#0 1.690   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
#0 1.690     _setup_information_cache[hashable_env] = _get_setup_information(
#0 1.690   File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 281, in _get_setup_information
#0 1.690     result = subprocess.run(
#0 1.690   File "/usr/lib/python3.10/subprocess.py", line 524, in run
#0 1.690     raise CalledProcessError(retcode, process.args,
[2023-01-24T17:40:35.426Z] #0 1.690 subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
```
Compilation finally fails with the following:
```
[2023-01-24T17:23:26.876Z] #0 22.04 Starting >>> IntelFortranImplicit
#0 22.22 --- stderr: FortranCInterface
#0 22.22 CMake Error at CMakeLists.txt:5 (project):
#0 22.22   No CMAKE_Fortran_COMPILER could be found.
#0 22.22 
#0 22.22   Tell CMake where to find the compiler by setting either the environment
#0 22.22   variable "FC" or the CMake cache entry CMAKE_Fortran_COMPILER to the full
#0 22.22   path to the compiler, or to the compiler name if it is in the PATH.
#0 22.22 
#0 22.22 
#0 22.22 ---
#0 22.22 Failed   <<< FortranCInterface [0.51s, exited with code 1]
#0 22.35 Aborted  <<< IntelFortranImplicit [0.31s]
[2023-01-24T17:23:26.876Z] #0 22.41 Aborted  <<< ament_pycodestyle [1.65s]
...
#0 28.97 
#0 28.97 Summary: 3 packages finished [28.3s]
#0 28.97   1 package failed: FortranCInterface
[2023-01-24T17:23:26.876Z] #0 28.97   11 packages aborted: IntelFortranImplicit ament_cppcheck ament_lint ament_package ament_pycodestyle google_benchmark_vendor googletest-distribution gtest iceoryx_hoofs osrf_pycommon osrf_testing_tools_cpp
#0 28.97   8 packages had stderr output: FortranCInterface IntelFortranImplicit ament_cppcheck ament_lint ament_package ament_pycodestyle google_benchmark_vendor osrf_pycommon
#0 28.97   345 packages not processed
------
[2023-01-24T17:23:26.876Z] error: failed to solve: executor failed running [/bin/sh -c colcon build --symlink-install]: exit code: 1
```
The same errors occur when using `humble` rather than `rolling`.

when I add `gfortran` to the `apt install` list, I get another error:
```
#0 24.61 Starting >>> lit
#0 25.19 [24.725s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'IntelFortranImplicit' because it has no 'install' target
#0 25.19 Finished <<< IntelFortranImplicit [2.07s]
#0 25.25 Starting >>> my-test-package
#0 26.09 --- stderr: lit
[2023-01-24T18:06:27.997Z] #0 26.09 CMake Warning (dev) in CMakeLists.txt:
#0 26.09   No project() command is present.  The top-level CMakeLists.txt file must
#0 26.09   contain a literal, direct call to the project() command.  Add a line of
#0 26.09   code such as
#0 26.09 
[2023-01-24T18:06:27.997Z] 
#0 26.09     project(ProjectName)
#0 26.09 
#0 26.09   near the top of the file, but after cmake_minimum_required().
#0 26.09 
#0 26.09   CMake is pretending there is a "project(Project)" command on the first
#0 26.09   line.
[2023-01-24T18:06:27.997Z] 
#0 26.09 This warning is for project developers.  Use -Wno-dev to suppress it.
#0 26.09 
#0 26.09 CMake Error at CMakeLists.txt:4 (configure_lit_site_cfg):
#0 26.09   Unknown CMake command "configure_lit_site_cfg".
#0 26.09 
[2023-01-24T18:06:27.997Z] #0 26.09 
#0 26.09 CMake Warning (dev) in CMakeLists.txt:
#0 26.09   No cmake_minimum_required command is present.  A line of code such as
#0 26.09 
#0 26.09     cmake_minimum_required(VERSION 3.22)
#0 26.09 
[2023-01-24T18:06:27.997Z] #0 26.09   should be added at the top of the file.  The version specified may be lower
#0 26.09   if you wish to support older CMake versions for this project.  For more
#0 26.09   information run "cmake --help-policy CMP0000".
#0 26.09 This warning is for project developers.  Use -Wno-dev to suppress it.
#0 26.09 
#0 26.09 ---
#0 26.09 Failed   <<< lit [1.48s, exited with code 1]
```
There's no explicit error to be found and I don't know what `lit` or `my-test-package` are.

#### Additional information

* I couldn't determine what depends on Fortran with certainty and searching the `ros2` org didn't yield any results, so this issue might be elsewhere or with documentation.
* I looked at the official docker images but they install the binary packages from `apt` rather than build from source.
* question while I'm here: If I want to develop on one of the component libraries (say `rclpy`), is building everything from source the recommended approach? Is there a way to use the prebuilt docker images `ros:rolling` and then clone just `rclpy`

clalancette, 2023-01-24:
Please open questions like this on https://answers.ros.org, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2023-01-24 by clalancette

Achllle, 2023-01-24:
@clalancette Can you explain why this is a question? I followed the instructions on [the docs](https://index.ros.org/doc/ros2/Installation/Rolling/Linux-Development-Setup/) and it fails which isn't a question, it seems like a bug to me. I did include a question at the end which I'm happy to post on answers instead.

> Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

Those are all included in the original post. I also linked to a gist that contains the Dockerfile I used.

clalancette, 2023-01-24:
> @clalancette Can you explain why this is a question?

We know that this works since we build it nightly: https://ci.ros2.org/view/nightly/job/nightly_linux_release/ .  So there is likely something in your setup.

Achllle, 2023-01-24:
The CI doesn't seem to build the same way that the documentation does, so perhaps this points to the documentation? I posted on answers [here](https://answers.ros.org/question/411771/building-ros2-rolling-from-source-issues-with-gfortran-lit/).

clalancette, 2023-01-24:
Likely this is a problem in colcon or setuptools.  Can you try two things please:

1.  Run this again and give us the full output.  That should help us tell if this is related to https://github.com/pypa/setuptools/issues/3791 , which we are currently tracking.
2. Try to downgrade to setuptools < 66.0, and see if that fixes the issue.

Achllle, 2023-01-24:
1. [logs.txt](https://github.com/ros2/ros2/files/10493508/logs.txt)
2. from the logs: `Setting up python3-setuptools (59.6.0-1.2ubuntu0.22.04.1)` so already at <66.0.

Achllle, 2023-01-25:
According to [this answer](https://answers.ros.org/question/411771/building-ros2-rolling-from-source-issues-with-gfortran-lit/?answer=411779#post-id-411779), colcon considers `.` as the base path, not `src`. Curious default, a warning would be helpful here.