ros2/ros2 #1373: ROS2 Documentation Error

929937690, 2023-01-24:
https://docs.ros.org/en/foxy/The-ROS2-Project/Contributing/Migration-Guide-Python.html

On this tutorial site and with all other ROS 2 versions documentation, over the ROS Parameters part, 

In ROS 1:
```
 port = rospy.get_param('port', '/dev/ttyUSB0')
 assert isinstance(port, str), 'port parameter must be a str'

 buadrate = rospy.get_param('baudrate', 115200)
 assert isinstance(port, int), 'port parameter must be an integer'

rospy.logwarn('port: ' + port)
```
In ROS 2:
```
port = node.declare_parameter('port', '/dev/ttyUSB0').value
assert isinstance(port, str), 'port parameter must be a str'

baudrate = node.declare_parameter('baudrate', 115200).value
assert isinstance(port, int), 'port parameter must be an integer'

node.get_logger().warn('port: ' + port)
```
I think it should be asserting baudrate parameter for the second part on both ROS 1 and ROS 2. 

And I do have a question about ROS 1 to ROS 2. In ROS 1, we could get a parameter and define queue_size for the publisher like this:
`self.publisher= rospy.Publisher(rospy.get_param( '/topic'), String, queue_size=10)`
How do we migrate this to ROS 2?

mjcarroll, 2023-01-25:
Thanks for reporting, I have fixed the `humble` version here: https://github.com/ros2/ros2_documentation/pull/3289

SteveMacenski, 2023-01-31:
@mjcarroll I think this can be closed :smile:

closed 2023-02-01 by clalancette