ros2/ros2 #1374: Runtime Interface Reflection Subscription Prototype (REP-2011 subset)

methylDragon, 2023-01-30:
# Description

This issue describes and lists the relevant PRs that allow subscriptions to receive messages (i.e. DynamicData) with descriptions/types constructed at runtime, and use that type to access fields of those messages without using message-specific compiled structs, as specified in https://github.com/ros-infrastructure/rep/pull/358

## Feature Description
- A C serialization support library run-time type reflection and construction, and utilities that support it
  - The associated serialization support libraries
- New rmw interfaces for:
  - Obtaining middleware-specific serialization support
  - Checking if a middleware supports:
    - Runtime type discovery
    - Taking message buffers as runtime type (i.e. DynamicData)
  - A new **middleware agnostic** runtime type `rosidl_message_type_support_t` implementation
- A new RuntimeTypeSubscription subscription class that supports a callback that takes dynamic data
  - Should support taking serialized messages and converting them into dynamic data for the callback
  - And also taking dynamic data directly from the middleware
- Necessary changes in rclcpp to allow the RuntimeTypeSubscription to work

Prototype serialization support libraries and implementations of this feature will be made for FastDDS in rmw, rcl, rclcpp, and rmw_fastrtps.

## Implementation Considerations

See [REP-2011](https://github.com/ros-infrastructure/rep/pull/358) for the feature specification.
(The REP will have to be updated slightly since the implementation of this feature exposed certain necessary refinements.)

FastDDS was chosen for the prototype implementation because it is currently the default rmw for ROS 2.

Additionally, some other REP-2011 features have barebones/minimal implementations that were created to support this feature. As other subsets of REP-2011 get implemented (e.g. the type description message and its utilities), sections of this feature will need to be swapped out with those implementations, namely:
- The type description message's C struct and how it is used
- The usage of type hashing

# Relevant PRs

- [ ] https://github.com/ros2/rmw/pull/340
- [ ] https://github.com/ros2/rcl/pull/1025
- [ ] https://github.com/ros2/rclcpp/pull/2077
- [ ] https://github.com/ros2/rmw_fastrtps/pull/655

# TODOs
- [x] Initial implementation
- [ ] Taking dynamic data directly

# Additional notes/issues
- (wip) const correctness

allenh1, 2023-02-08:
@methylDragon Hey! I was trying to run the demo locally, but I had trouble compiling:

```
Finished <<< rmw_connextdds_common [4.04s]
Starting >>> rmw_connextdds
Starting >>> rmw_connextddsmicro
Finished <<< rmw_connextdds [5.13s]            
Finished <<< rcl_interfaces [30.8s]            
Finished <<< rmw_connextddsmicro [5.36s]                                     
Finished <<< std_msgs [32.5s]                                                
Starting >>> runtime_type_support_examples_msgs
Finished <<< rmw_fastrtps_shared_cpp [16.3s]                                 
Finished <<< test_msgs [42.0s]                                               
Starting >>> rmw_fastrtps_cpp
Starting >>> rmw_fastrtps_dynamic_cpp         
Finished <<< runtime_type_support_examples_msgs [10.0s]
Finished <<< rmw_fastrtps_cpp [16.3s]                                          
Finished <<< rmw_fastrtps_dynamic_cpp [17.2s]                                
Starting >>> rmw_implementation
Finished <<< rmw_implementation [11.3s]                                
Starting >>> rcl
--- stderr: rcl                                        
/usr/lib/gcc/x86_64-pc-linux-gnu/12/../../../../x86_64-pc-linux-gnu/bin/ld: ../librcl.so: undefined reference to `rmw_get_serialization_support'
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/CMakeFiles/test_rmw_impl_id_check_exe.dir/build.make:138: test/test_rmw_impl_id_check_exe] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:336: test/CMakeFiles/test_rmw_impl_id_check_exe.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rcl [17.3s, exited with code 2]

Summary: 119 packages finished [4min 52s]
  1 package failed: rcl
  34 packages had stderr output: ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_package ament_pep257 ament_pycodestyle ament_uncrustify ament_xmllint domain_coordinator fastrtps foonathan_memory_vendor google_benchmark_vendor launch launch_testing launch_xml launch_yaml mimick_vendor osrf_pycommon performance_test_fixture rcl rcl_yaml_param_parser rcutils rmw_connextdds_common rmw_cyclonedds_cpp rosidl_cli rosidl_pycommon rosidl_typesupport_c rpyutils rti_connext_dds_cmake_module uncrustify_vendor
  4 packages not processed
```

Am I missing changes or something?

methylDragon, 2023-02-08:
> @methylDragon Hey! I was trying to run the demo locally, but I had trouble compiling:
>
> ```
> Finished <<< rmw_connextdds_common [4.04s]
> Starting >>> rmw_connextdds
> Starting >>> rmw_connextddsmicro [...]

Hey hunter! The examples won't compile if you have the other non-FastDDS middlewares in your workspace. (I see that your colcon is building rmw_cyclonedds, for example.)

The ros2.repos file I have supplied with the examples should have commented out iceoryx, connext, and cyclone. Check to make sure you didn't source ROS 2 rolling before compiling!

methylDragon, 2023-02-08:
Ah, apologies @allenh1 , it turns out that I **also** missed out pushing a change...

I need to update `rmw_implementation`. Sorry about that, let me get back to you soon.

allenh1, 2023-02-08:
> Ah, apologies @allenh1 , it turns out that I also missed out pushing a change...

@methylDragon awesome. Thanks! I'll give this a try shortly.

methylDragon, 2023-02-08:
Oh.. you also need to build ros2cli (and ros2run) also :facepalm: , my fault for working in a dirty workspace where I had run `colcon build` before

I'm refining the setup commands.

methylDragon, 2023-04-02:
https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/78091c76c4d7a6806be61f0d27a9e36d4bfa938a/runtime_interface_reflection.repos

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-02:
Another one, windows build fixed

https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/78091c76c4d7a6806be61f0d27a9e36d4bfa938a/runtime_interface_reflection.repos

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-02:
Another one https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/53ddc68894bc9a8cc1e941519a48c93a3d245020/runtime_interface_reflection.repos

I missed out using Emerson's branch of rosidl_typesupport and rosidl_typesupport_fastrtps

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-02:
Fix warnings

[gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/53ddc68894bc9a8cc1e941519a48c93a3d245020/runtime_interface_reflection.repos](https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/53ddc68894bc9a8cc1e941519a48c93a3d245020/runtime_interface_reflection.repos)

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-06:
== Before fix ==
https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/828f2208a5a0e2c66491b2badb052ba20514ef96/runtime_interface_reflection.repos

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

== After Fix ==
https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/c4690ada72b89328ad72e0e9c9e17383d32a112a/odr_fix.repos

ODR fix, fixed ODR:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-06:
Doing a run for rviz as well,

Fix is in:
- https://github.com/ros2/rviz/pull/967

== BEFORE FIX ==

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

== AFTER FIX ==

https://gist.githubusercontent.com/methylDragon/1f63b89429c327cc13002538c5294a29/raw/e2510930b8bb60c09f5f427162faf7e0d1724605/odr_fix_with_rviz.repos

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-07:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-07:
Allocator refactor is being built on top of this wave of PRs: https://github.com/ros2/ros2/issues/1405

methylDragon, 2023-04-07:
Canary build for windows warnings

* Windows [image]

methylDragon, 2023-04-08:
Pre-RMW changes:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

wjwwood, 2023-04-08:
I've approved all the prs, and windows debug is at least into running tests, so hopefully it comes back green-ish tomorrow. I'll merge all of these then.

wjwwood, 2023-04-08:
Tests lgtm, let me double check the prs since some other stuff was merged.

wjwwood, 2023-04-08:
They look ok, I'm going to merge 🤞, let's see what the nightlies say!