ros2/ros2 #1375: Missing sys/acl.h when trying to install Humble

DrBwts, 2023-01-30:
## Bug report

- Operating System:
  Windows 10 running WSL2 - Ubuntu
- Installation type:
  From source following the instructions from [here](https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html)
- Version or commit hash:
  ROS2 Humble
- DDS implementation:
 unknown - how do I find out?
- Client library (if applicable):
  problem with iceoryx_hoofs in ament_pycodestyle

I followed the instruction to install from source for Ubuntu. All is well until I run,

`colcon build --symlink-install`

Which Installs the 1st few packages with some warnings then throws a "fatal error"

```
Starting >>> ament_pycodestyle
--- stderr: iceoryx_hoofs
In file included from /root/ros2_humble/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/include/iceoryx_hoofs/internal/posix_wrapper/access_control.hpp:23,
                 from /root/ros2_humble/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/source/posix_wrapper/access_control.cpp:18:
/root/ros2_humble/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/platform/linux/include/iceoryx_hoofs/platform/acl.hpp:20:10: fatal error: sys/acl.h: No such file or directory
   20 | #include <sys/acl.h>
      |          ^~~~~~~~~~~
compilation terminated.
```

Full traceback,

```
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> fastcdr
Starting >>> ament_cppcheck
Starting >>> iceoryx_hoofs
Starting >>> osrf_pycommon
Starting >>> osrf_testing_tools_cpp
Finished <<< fastcdr [2.19s]
Starting >>> google_benchmark_vendor
Finished <<< gtest_vendor [2.63s]
Starting >>> gmock_vendor
--- stderr: ament_lint
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< ament_lint [4.14s]
Starting >>> ament_flake8
--- stderr: ament_package
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< ament_package [4.21s]
Starting >>> ament_cmake_core
--- stderr: ament_cppcheck
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< ament_cppcheck [4.73s]
--- stderr: osrf_pycommon
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< osrf_pycommon [4.76s]
Starting >>> urdfdom_headers
Starting >>> ament_pycodestyle
--- stderr: iceoryx_hoofs
In file included from /root/ros2_humble/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/include/iceoryx_hoofs/internal/posix_wrapper/access_control.hpp:23,
                 from /root/ros2_humble/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/source/posix_wrapper/access_control.cpp:18:
/root/ros2_humble/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/platform/linux/include/iceoryx_hoofs/platform/acl.hpp:20:10: fatal error: sys/acl.h: No such file or directory
   20 | #include <sys/acl.h>
      |          ^~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/iceoryx_hoofs.dir/build.make:300: CMakeFiles/iceoryx_hoofs.dir/source/posix_wrapper/access_control.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/iceoryx_hoofs.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
---
Failed   <<< iceoryx_hoofs [4.86s, exited with code 2]
Aborted  <<< urdfdom_headers [0.08s]
Aborted  <<< ament_flake8 [0.73s]
Aborted  <<< gmock_vendor [2.16s]
Aborted  <<< ament_pycodestyle [0.08s]
Aborted  <<< google_benchmark_vendor [2.61s]
Aborted  <<< osrf_testing_tools_cpp [5.12s]
Aborted  <<< ament_cmake_core [1.49s]

Summary: 6 packages finished [7.65s]
  1 package failed: iceoryx_hoofs
  7 packages aborted: ament_cmake_core ament_flake8 ament_pycodestyle gmock_vendor google_benchmark_vendor osrf_testing_tools_cpp urdfdom_headers
  5 packages had stderr output: ament_cppcheck ament_lint ament_package iceoryx_hoofs osrf_pycommon
  329 packages not processed
```

clalancette, 2023-01-30:
Did you run the step where you run:

```
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers"
```
?

That should have installed `libacl1-dev`, which is where that header comes from.

clalancette, 2023-02-16:
No response in 2 weeks, closing this.

closed 2023-02-16 by clalancette

orascheg, 2024-01-10:
Have the same problem. It still happens. Command in the first response does not work out of the box, because ROS_DISTRO not yet set.
After adding the --rosdistro option the same error happens (rosdep install --from-paths src --ignore-src -y --skip-keys --rosdistro humble "fastcdr rti-connext-dds-6.0.1 urdfdom_headers")
I have a my_ros2.repos file, which I unfortunately cannot attach here.

avgsurfman, 2024-05-14:
Same problem when trying to install ROS2 on Raspberry Pi via Cross-compilation. Happens on both Ubuntu and Debian.
The compilation terminates and I cannot proceed any further.

Wildhair2, 2024-06-09:
> That should have installed `libacl1-dev`, which is where that header comes from.

You are correct. It is far easier to install it directly (on Debian based distros) with:
`sudo apt install libacl1-dev -y`

This fixes the missing sys/acl.h that numerous programs fail on compilation (such as "pip3 install borgbackup")

cdisco, 2024-11-25:
For what it's worth I'm experiencing the same issue from a jazzy-desktop docker container.
It is being tracked in issue #1626