ros2/ros2 #1381: RuntimeError: Failed to initialize init_options: failed to load shared library

LukeKcy, 2023-02-23:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 Education
- Installation type:
  -  source
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Just trying to import this library causes the issue
from pssh.client import ParallelSSHClient, SSHClient
```

```

#### Expected behavior
Should just run code as normal, 
#### Actual behavior
`Traceback (most recent call last):
  File "c:\workspace_ws\install\lib\talker\talker-script.py", line 33, in <module>
    sys.exit(load_entry_point('talker==0.0.0', 'console_scripts', 'talker')())
  File "c:\workspace_ws\install\Lib\site-packages\talker\main.py", line 251, in main
    rclpy.init(args=args)
  File "c:\opt\ros\foxy\x64\lib\site-packages\rclpy\__init__.py", line 71, in init
    return context.init(args)
  File "c:\opt\ros\foxy\x64\lib\site-packages\rclpy\context.py", line 59, in init
    rclpy_implementation.rclpy_init(args if args is not None else sys.argv, capsule)
RuntimeError: Failed to initialize init_options: failed to load shared library 'c:\opt\ros\foxy\x64\bin/rmw_fastrtps_cpp.dll' due to LoadLibrary error: 127, at C:\a\1\s\ws\src\rcutils\src\shared_library.c:88, at C:\a\1\s\ws\src\rmw_implementation\rmw_implementation\src\functions.cpp:84, at C:\a\1\s\ws\src\rcl\rcl\src\rcl\init_options.c:61`

#### Additional information
Link to parallel ssh library[here](https://parallel-ssh.org/)
I'm able to run code using the library fine but when I try implement this with ROS2 it causes be issues, I can't even import the library. I've made sure Visual C++ Redistributable Packages for Visual Studio 2013  is downloaded based on the suggestion on the installation troubleshooting.

mjcarroll, 2023-03-01:
Can you add some more details here?  It's unclear to me how the ssh library and ROS 2 are interacting in what you have described.

LukeKcy, 2023-03-01:
My goal is to connect to 4 raspberry Pi devices and run code from each of them to operate cameras. I'm using the ssh library to connect to the devices. From there I'm sending a String message from the RPi devices to my host PC. I'm trying to use ROS 2 to send this message to a Unity subscriber which will allow me to instantiate objects in my scene based on the message. 

I know the Unity subscriber works as we used other cameras (PoE models) to complete the task and had no issues. We are trying to update the cameras for new models (which don't have PoE capability, hence the RPi devices). The use of the ssh library and ROS 2 is to allow for the messages to be sent from the RPis to the Unity listener.

The code below should allow the Unity subscriber to receive the messages from the cameras (for testing I'm working on sending one string message).
```
import` rclpy
from rclpy.node import Node
from std_msgs.msg import String
from geometry_msgs.msg import Point
import argparse
from pssh.clients import ParallelSSHClient, SSHClient
from datetime import datetime
import itertools

#parser = argparse.ArgumentParser()
#args = parser.parse_args()

class TalkerNode(Node):
    def __init__(self):
        super().__init__('talker')
        self.publisher = self.create_publisher(String, 'talker', 10)
        timer_period = 10
        self.timer = self.create_timer(timer_period, self.timer_callback)

    def timer_callback(self):
        hosts = ['192.168.1.x', '192.168.1.xx', '192.168.1.xx', '192.168.1.xx'] #IPs for devices to connect to
        client = ParallelSSHClient(hosts, user='xx', password='xxxx', forward_ssh_agent=True) #Connect and log into each device

        ##### Connect to each none parallel RPi to run the Detection script ######
        client1 = SSHClient('192.168.1.xx', user='xx', password='xxxx', forward_ssh_agent=True) #Connect and log into each device
        client2 = SSHClient('192.168.1.xx', user='xx', password='xxxx', forward_ssh_agent=True) #Connect and log into each device
        client3 = SSHClient('192.168.1.xx', user='xx', password='xxxx', forward_ssh_agent=True) #Connect and log into each device
        client4 = SSHClient('192.168.1.xx', user='xx', password='xxxx', forward_ssh_agent=True) #Connect and log into each device

        cmd = 'python PiSmartHangar/YoloTestCode.py'
        output1 = client1.run_command(cmd, use_pty=True)
        output2 = client2.run_command(cmd, use_pty=True)
        output3 = client3.run_command(cmd, use_pty=True)
        output4 = client4.run_command(cmd, use_pty=True)
        msg = String()
        A = ""
        B = ""
        C = ""
        D = ""

        for line1, line2, line3, line4 in itertools.zip_longest(output1.stdout, output2.stdout, output3.stdout, output4.stdout, fillvalue = ' '):
            if line1 is not None:
                print(line1)
                A = line1
            if line2 is not None:
                print(line2)
                B = line2
            if line3 is not None:    
                print(line3)
                C = line3
            if line4 is not None:
                print(line4)
                D = line4

            msg.data = A
            self.publisher.publish(msg)

def main(args = None):
    rclpy.init(args=args)
    talker_node = TalkerNode()
    rclpy.spin(talker_node)
    rclpy.shutdown()
```

Even when we try to use a simple test publisher/subscriber with the ssh library we end up with the same issue. Would it be possible that the ssh library isn't compatible with ROS 2? I appreciate any help that you can give with this issue.

jd509, 2023-03-13:
Hi @LukeKcy !  I plan to take up the issue, could you assign this one to me?

somekindsoffood, 2023-07-14:
I met the same issue does anyone know how to solve it？many thx。