ros2/ros2 #1389: ROS2 Humble colcon build error

snhamid, 2023-03-09:
I am using a Surface Laptop 2 running Kubuntu 22.04

I have been having this issue for quite a while now. I try to run
`colcon build --symlink-install`
from my root project directory. I get the following error:

shamid@shamid-Surface-Laptop-2:~/CLionProjects/R2ED$ sudo colcon build --symlink-install
Starting >>> humanoid_interfaces
Starting >>> humanoid_desc
Starting >>> humanoid_gazebo_plugin
Starting >>> humanoid_neck
Starting >>> lwa4p_controller
Starting >>> lwa4p_driver
Starting >>> lwa4p_hardware
Starting >>> senior_design_piano
--- stderr: humanoid_interfaces                                                                                                                                                                          
CMake Error at /opt/ros/humble/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:35 (message):
  No 'rosidl_typesupport_c' found
Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:8 (get_used_typesupports)
  /opt/ros/humble/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/humble/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /opt/ros/humble/share/action_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/action_msgs/cmake/action_msgsConfig.cmake:41 (include)
  /opt/ros/humble/share/moveit_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/moveit_msgs/cmake/moveit_msgsConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

---
Failed   <<< humanoid_interfaces [0.66s, exited with code 1]
Aborted  <<< lwa4p_controller [0.64s]
Aborted  <<< humanoid_desc [0.66s]
Aborted  <<< humanoid_neck [0.65s]
Aborted  <<< lwa4p_hardware [0.90s]                                                                                                                                       
Aborted  <<< senior_design_piano [1.03s]                                                                                                    
Aborted  <<< lwa4p_driver [1.16s]                                                                        
Aborted  <<< humanoid_gazebo_plugin [1.45s]                            

Summary: 0 packages finished [1.65s]
  1 package failed: humanoid_interfaces
  7 packages aborted: humanoid_desc humanoid_gazebo_plugin humanoid_neck lwa4p_controller lwa4p_driver lwa4p_hardware senior_design_piano
  6 packages had stderr output: humanoid_gazebo_plugin humanoid_interfaces humanoid_neck lwa4p_controller lwa4p_hardware senior_design_piano
  4 packages not processed

There are a few problems with this. 

1. My classmates are able to run this script completely fine without any errors
2. The referenced package "rosidl_typesupport_c" is installed on my computer and I have verified this multiple times.
3. This build was working completely fine until one day it stopped working entirely
I have ensured I am up to date on the repo 
I have not been able to find an answer yet. All help is greatly appreciated.

mmsoby, 2023-03-09:
I'm not really an expert at robotics but have you tried restarting your computer?

snhamid, 2023-03-09:
> I'm not really an expert at kubuntu but have you tried restarting your computer?

Yes I have, this was suggested by an expert in kubuntu

clalancette, 2023-03-16:
You should not be running colcon with sudo; that will be stripping environment variables that you'll need.

In the future, please open questions like this on https://answers.ros.org , which is our central Question and Answer site.  You'll get faster answers there.

closed 2023-03-16 by clalancette