ros2/ros2 #1398: Build error

tumani1, 2023-03-19:
## Bug report

- Operating System:
  - MacOS Monterey 12.6.3
  
  Log:
  ```
  Starting >>> urdf_parser_plugin
Finished <<< tracetools [3.52s]
Starting >>> mimick_vendor
Finished <<< launch_testing_ament_cmake [2.35s]
Starting >>> tlsf
Finished <<< ament_index_cpp [3.50s]
Starting >>> resource_retriever
[53.969s] WARNING:colcon.colcon_cmake.task.cmake.build:Could not run installation step for package 'tlsf' because it has no 'install' target
--- stderr: performance_test_fixture
Undefined symbols for architecture x86_64:
  "osrf_testing_tools_cpp::memory_tools::on_unexpected_calloc(std::__1::variant<std::__1::function<void (osrf_testing_tools_cpp::memory_tools::MemoryToolsService&)>, std::__1::function<void ()>, std::nullptr_t>)", referenced from:
      performance_test_fixture::PerformanceTest::SetUp(benchmark::State&) in performance_test_fixture.cpp.o
  "osrf_testing_tools_cpp::memory_tools::on_unexpected_malloc(std::__1::variant<std::__1::function<void (osrf_testing_tools_cpp::memory_tools::MemoryToolsService&)>, std::__1::function<void ()>, std::nullptr_t>)", referenced from:
      performance_test_fixture::PerformanceTest::SetUp(benchmark::State&) in performance_test_fixture.cpp.o
  "osrf_testing_tools_cpp::memory_tools::on_unexpected_realloc(std::__1::variant<std::__1::function<void (osrf_testing_tools_cpp::memory_tools::MemoryToolsService&)>, std::__1::function<void ()>, std::nullptr_t>)", referenced from:
      performance_test_fixture::PerformanceTest::SetUp(benchmark::State&) in performance_test_fixture.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libperformance_test_fixture.dylib] Error 1
make[1]: *** [CMakeFiles/performance_test_fixture.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< performance_test_fixture [2.42s, exited with code 2]
Aborted  <<< rosidl_parser [3.54s]
Aborted  <<< tlsf [1.13s]
Aborted  <<< urdf_parser_plugin [1.89s]
Aborted  <<< mimick_vendor [1.70s]
Aborted  <<< resource_retriever [0.91s]
Aborted  <<< python_qt_binding [3.72s]
Aborted  <<< tango_icons_vendor [18.4s]

Summary: 96 packages finished [1min 9s]
  1 package failed: performance_test_fixture
  7 packages aborted: mimick_vendor python_qt_binding resource_retriever rosidl_parser tango_icons_vendor tlsf urdf_parser_plugin
  5 packages had stderr output: cyclonedds performance_test_fixture pybind11_vendor rviz_ogre_vendor shared_queues_vendor
  227 packages not processed
  ```

clalancette, 2023-03-20:
We don't officially support macOS.  That said, this is likely fixed by the recent merging of https://github.com/osrf/osrf_testing_tools_cpp/pull/76 , so if you could give that a try, that would be helpful.

tumani1, 2023-03-20:
okay, i will try to use docker