ros2/ros2 #1400: Service callback blocks subscriber callback with MultiThreadedExecutor

leonce-m, 2023-04-02:
## Bug report

- Operating System: Ubuntu 22.04
- Installation type: binary
- Version or commit hash: humble
- DDS implementation: default
- Client library (if applicable): rclcpp

#### Steps to reproduce issue

1. Create a node with a subscriber and a service.
2. Either use two MutuallyExclusive or one Reentrant callback group.
3. Spin the node in a MultiThreadedExecutor

```#include <chrono>
#include <functional>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
#include "std_srvs/srv/empty.hpp"

using namespace std::placeholders;

class MultiThreadedNode : public rclcpp::Node
{
public:
    MultiThreadedNode()
    : Node("multi_threaded_node")
    {
        auto default_qos = rclcpp::QoS(10);
        cb_group_1_ = create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive);
        cb_group_2_ = create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive);
        cb_group_3_ = create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive);

        rclcpp::SubscriptionOptions sub_options;
        sub_options.callback_group = cb_group_1_;

        subscriber_ = create_subscription<std_msgs::msg::String>(
            "chatter",
            default_qos, 
            std::bind(&MultiThreadedNode::subscriber_callback, this, _1),
            sub_options);

        service_ = create_service<std_srvs::srv::Empty>(
            "service",
            std::bind(&MultiThreadedNode::service_callback, this, _1, _2),
            rmw_qos_profile_default,
            cb_group_2_);

        timer_ = create_wall_timer(
            std::chrono::seconds(2),
            std::bind(&MultiThreadedNode::timer_callback, this),
            cb_group_3_);
    }
private:
    void subscriber_callback(const std_msgs::msg::String::SharedPtr msg)
    {
        (void)msg;
        RCLCPP_INFO(get_logger(), ".");
    }

    void service_callback(
        const std_srvs::srv::Empty::Request::SharedPtr req,
        const std_srvs::srv::Empty::Response::SharedPtr res)
    {
        (void)req;
        (void)res;
        RCLCPP_INFO(this->get_logger(), "enter service callback");
        for (size_t i = 0; i < 10; i++)
        {
            RCLCPP_INFO(this->get_logger(), "#");
            std::this_thread::sleep_for(std::chrono::milliseconds(100));
        }
        RCLCPP_INFO(this->get_logger(), "exit service callback");
    }

    void timer_callback()
    {
        RCLCPP_INFO(this->get_logger(), "enter timer callback");
        for (int i = 0; i < 10; ++i) {
            RCLCPP_INFO(this->get_logger(), "+");
            std::this_thread::sleep_for(std::chrono::milliseconds(100));
        }
        RCLCPP_INFO(this->get_logger(), "exit timer callback");
    }

    rclcpp::CallbackGroup::SharedPtr cb_group_1_;
    rclcpp::CallbackGroup::SharedPtr cb_group_2_;
    rclcpp::CallbackGroup::SharedPtr cb_group_3_;

    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr subscriber_;
    rclcpp::Service<std_srvs::srv::Empty>::SharedPtr service_;
    rclcpp::TimerBase::SharedPtr timer_;

};

int main(int argc, char * argv[])
{
    rclcpp::init(argc, argv);

    auto node = std::make_shared<MultiThreadedNode>();
    rclcpp::executors::MultiThreadedExecutor executor;
    executor.add_node(node);
    executor.spin();

    rclcpp::shutdown();
    return 0;
}
```

#### Expected behavior

Calling the service should not block the subscriber or the timer.

#### Actual behavior

Calling the service blocks the subscriber but not the timer.

```
[INFO] [1680471757.115031455] [multi_threaded_node]: .
[INFO] [1680471757.314653746] [multi_threaded_node]: .
[INFO] [1680471757.514849401] [multi_threaded_node]: .
[INFO] [1680471757.714504254] [multi_threaded_node]: .
[INFO] [1680471757.807676140] [multi_threaded_node]: enter timer callback
[INFO] [1680471757.807809920] [multi_threaded_node]: +
[INFO] [1680471757.908003562] [multi_threaded_node]: +
[INFO] [1680471757.915152829] [multi_threaded_node]: .
[INFO] [1680471758.008272385] [multi_threaded_node]: +
[INFO] [1680471758.108555032] [multi_threaded_node]: +
[INFO] [1680471758.114981067] [multi_threaded_node]: .
[INFO] [1680471758.208821078] [multi_threaded_node]: +
[INFO] [1680471758.296828261] [multi_threaded_node]: enter service callback
[INFO] [1680471758.296945886] [multi_threaded_node]: #
[INFO] [1680471758.309048233] [multi_threaded_node]: +
[INFO] [1680471758.397140516] [multi_threaded_node]: #
[INFO] [1680471758.409268071] [multi_threaded_node]: +
[INFO] [1680471758.497408827] [multi_threaded_node]: #
[INFO] [1680471758.509478168] [multi_threaded_node]: +
[INFO] [1680471758.597679190] [multi_threaded_node]: #
[INFO] [1680471758.609651464] [multi_threaded_node]: +
[INFO] [1680471758.697816595] [multi_threaded_node]: #
[INFO] [1680471758.709744064] [multi_threaded_node]: +
[INFO] [1680471758.797935812] [multi_threaded_node]: #
[INFO] [1680471758.809839482] [multi_threaded_node]: exit timer callback
[INFO] [1680471758.898066940] [multi_threaded_node]: #
[INFO] [1680471758.998168237] [multi_threaded_node]: #
[INFO] [1680471759.098295381] [multi_threaded_node]: #
[INFO] [1680471759.198418182] [multi_threaded_node]: #
[INFO] [1680471759.298543627] [multi_threaded_node]: exit service callback
[INFO] [1680471759.305546736] [multi_threaded_node]: .
[INFO] [1680471759.514772973] [multi_threaded_node]: .
[INFO] [1680471759.714968133] [multi_threaded_node]: .
[INFO] [1680471759.807685939] [multi_threaded_node]: enter timer callback
[INFO] [1680471759.807810022] [multi_threaded_node]: +
[INFO] [1680471759.908048993] [multi_threaded_node]: +
[INFO] [1680471759.914723904] [multi_threaded_node]: .
[INFO] [1680471760.008308786] [multi_threaded_node]: +
[INFO] [1680471760.108570934] [multi_threaded_node]: +
[INFO] [1680471760.114304442] [multi_threaded_node]: .
[INFO] [1680471760.208807294] [multi_threaded_node]: +
[INFO] [1680471760.309081122] [multi_threaded_node]: +
[INFO] [1680471760.314686328] [multi_threaded_node]: .
[INFO] [1680471760.409337562] [multi_threaded_node]: +
[INFO] [1680471760.509623742] [multi_threaded_node]: +
[INFO] [1680471760.514945516] [multi_threaded_node]: .
[INFO] [1680471760.609898005] [multi_threaded_node]: +
[INFO] [1680471760.710182078] [multi_threaded_node]: +
[INFO] [1680471760.714599593] [multi_threaded_node]: .
[INFO] [1680471760.810430563] [multi_threaded_node]: exit timer callback
[INFO] [1680471760.914164703] [multi_threaded_node]: .
[INFO] [1680471761.114127971] [multi_threaded_node]: .
[INFO] [1680471761.314642288] [multi_threaded_node]: .
[INFO] [1680471761.514949373] [multi_threaded_node]: .
[INFO] [1680471761.714710297] [multi_threaded_node]: .
```

#### Additional information

Please don't tell me to use an action.

fujitatomoya, 2023-04-25:
I cannot reproduce this issue with rolling https://github.com/ros2/ros2/commit/fea3f22ebc1becc6303aebcc9f93bde5abf0f92d

we can see the subscription callback is called based on publication, service call does not block the subscription callback.

- publisher

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic pub -r 1 /chatter std_msgs/msg/String "data: Hello"
publisher: beginning loop
publishing #1: std_msgs.msg.String(data='Hello')

publishing #2: std_msgs.msg.String(data='Hello')

publishing #3: std_msgs.msg.String(data='Hello')

publishing #4: std_msgs.msg.String(data='Hello')

publishing #5: std_msgs.msg.String(data='Hello')

publishing #6: std_msgs.msg.String(data='Hello')

publishing #7: std_msgs.msg.String(data='Hello')
...<snip>
```

- service client

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 service call /service std_srvs/srv/Empty
requester: making request: std_srvs.srv.Empty_Request()

response:
std_srvs.srv.Empty_Response()

```

- [sample application](https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclcpp/src/ros2_1400.cpp)

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp ros2_1400
[INFO] [1682396340.311997818] [multi_threaded_node]: enter timer callback
[INFO] [1682396340.312371436] [multi_threaded_node]: +
[INFO] [1682396341.312505771] [multi_threaded_node]: +
[INFO] [1682396342.312876926] [multi_threaded_node]: +
[INFO] [1682396343.313222623] [multi_threaded_node]: +
[INFO] [1682396343.542140432] [multi_threaded_node]: .
[INFO] [1682396344.313487851] [multi_threaded_node]: +
[INFO] [1682396344.543880708] [multi_threaded_node]: .
[INFO] [1682396345.313842780] [multi_threaded_node]: +
[INFO] [1682396345.543880109] [multi_threaded_node]: .
[INFO] [1682396346.314206216] [multi_threaded_node]: +
[INFO] [1682396346.538394098] [multi_threaded_node]: enter service callback
[INFO] [1682396346.538434910] [multi_threaded_node]: #
[INFO] [1682396346.542449663] [multi_threaded_node]: .
[INFO] [1682396347.314472785] [multi_threaded_node]: +
[INFO] [1682396347.538611666] [multi_threaded_node]: #
[INFO] [1682396347.543570484] [multi_threaded_node]: .
[INFO] [1682396348.314789306] [multi_threaded_node]: +
[INFO] [1682396348.538956262] [multi_threaded_node]: #
[INFO] [1682396348.543820296] [multi_threaded_node]: .
[INFO] [1682396349.315117501] [multi_threaded_node]: exit timer callback
[INFO] [1682396349.539326467] [multi_threaded_node]: #
[INFO] [1682396349.543789416] [multi_threaded_node]: .
[INFO] [1682396350.311976680] [multi_threaded_node]: enter timer callback
[INFO] [1682396350.312131075] [multi_threaded_node]: +
[INFO] [1682396350.539659562] [multi_threaded_node]: #
[INFO] [1682396350.543826073] [multi_threaded_node]: .
[INFO] [1682396351.312337249] [multi_threaded_node]: +
[INFO] [1682396351.539918209] [multi_threaded_node]: #
[INFO] [1682396351.543779407] [multi_threaded_node]: .
[INFO] [1682396352.312717030] [multi_threaded_node]: +
[INFO] [1682396352.540252516] [multi_threaded_node]: #
[INFO] [1682396352.543780019] [multi_threaded_node]: .
[INFO] [1682396353.313060884] [multi_threaded_node]: +
[INFO] [1682396353.540589582] [multi_threaded_node]: #
[INFO] [1682396353.543722982] [multi_threaded_node]: .
[INFO] [1682396354.313411743] [multi_threaded_node]: +
[INFO] [1682396354.540927073] [multi_threaded_node]: #
[INFO] [1682396354.543752216] [multi_threaded_node]: .
[INFO] [1682396355.313735594] [multi_threaded_node]: +
[INFO] [1682396355.541333231] [multi_threaded_node]: #
[INFO] [1682396355.542834266] [multi_threaded_node]: .
[INFO] [1682396356.314067515] [multi_threaded_node]: +
[INFO] [1682396356.541576322] [multi_threaded_node]: exit service callback
[INFO] [1682396356.543694572] [multi_threaded_node]: .
[INFO] [1682396357.314447947] [multi_threaded_node]: +
[INFO] [1682396357.543778569] [multi_threaded_node]: .
[INFO] [1682396358.314789714] [multi_threaded_node]: +
[INFO] [1682396358.543769904] [multi_threaded_node]: .
[INFO] [1682396359.315158851] [multi_threaded_node]: exit timer callback
[INFO] [1682396360.312031223] [multi_threaded_node]: enter timer callback
[INFO] [1682396360.312176592] [multi_threaded_node]: +
```

fujitatomoya, 2023-04-25:
@leonce-m can you share more details about how you did run service client and publisher to the sample application?

leonce-m, 2023-05-02:
Ah! I used rqt to publish on the /chatter topic and call the service. Since calling the service blocks the rqt node, it interrupts the publisher.

Running the publisher and service client via terminal yields the expected result as @fujitatomoya described.

closed 2023-05-02 by leonce-m