ros2/ros2 #1405: Dynamic Subscription (REP-2011 subset): BONUS: Allocator Refactor

methylDragon, 2023-04-07:
# Description
No long detailed description this time, unfortunately...

This issue describes and lists the relevant PRs that build on top of the PRs in https://github.com/ros2/ros2/issues/1374 with non-essential, bonus changes for ease of review and decoupling.

The base PRs should be **prioritized**.

The PRs in #1374 assumed that most of the objects being introduced can only live on the heap, but this wave of PRs allows those objects to be created on the stack and then initialized.

The changes here are namely refactors across the entire stack for function signatures to take in allocators, and splitting of `create()` functions into `init()` and `create()` (likewise for destroy into `fini()` and `destroy()`.)

That is, splitting the steps of:
- Allocating an outer struct (create/new)
- Allocating and initializing the internal members of that struct (init)

And similarly for destruction:
- Deallocating the outer struct (delete/destroy)
- Deallocating and finalizing the internal members of that struct (fini)

# Relevant PRs
...

methylDragon, 2023-04-09:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

methylDragon, 2023-04-10:
Run 2 for posterity:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]
* Windows Debug [image]

methylDragon, 2023-04-10:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]
* Windows Debug

methylDragon, 2023-04-10:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]
* Windows Debug [image]

clalancette, 2023-05-02:
After discussion, the future work here is as follows:

1. Get https://github.com/ros2/rclcpp/pull/2160 reviewed and merged into Rolling.
1. Get https://github.com/ros2/rclcpp/pull/2160 backported to Iron for a patch release
1. Implement dynamic type support for Connext (probably needs a new rosidl_dynamic_typesupport_connext package).
1. Update rmw_connextdds to take advantage of the above.
1. Implement dynamic type support for CycloneDDS (probably needs a new rosidl_dynamic_typesupport_cyclonedds package).
1. Update rmw_cyclonedds to take advantage of the above.
1. Documentation and examples for the feature.

What I'll suggest is that after we get the first two items done, we close out this issue and open a new one to track the remaining work.