ros2/ros2 #1406: [bug] : QSocketNotifier: Can only be used with threads started with QThread | symbol lookup error

lonebots, 2023-04-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.02
- Installation type:
  - Source
- Version or commit hash:
  - ROS 2

#### Steps to reproduce issue

1. Follow the official documentation of ROS2 Humble for [Writing a broadcaster (Python)](http://docs.ros.org/en/humble/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Broadcaster-Py.html) 
2.  At the final stage when trying to run the `launch` file I am getting the following error. 

``` bash
[INFO] [launch]: All log files can be found below /home/jishnu/.ros/log/2023-04-07-18-58-34-999453-Legion-2153240
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [turtlesim_node-1]: process started with pid [2153241]
[INFO] [turtle_tf2_broadcaster-2]: process started with pid [2153243]
[turtlesim_node-1] QSocketNotifier: Can only be used with threads started with QThread
[turtlesim_node-1] /opt/ros/humble/lib/turtlesim/turtlesim_node: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE
[ERROR] [turtlesim_node-1]: process has died [pid 2153241, exit code 127, cmd '/opt/ros/humble/lib/turtlesim/turtlesim_node --ros-args -r __node:=sim'].
```

#### Expected behavior 
The turtlesim Node is supposed to start and the broadcaster node should be publishing the transforms.

#### Actual behavior
The error occurring. 
```bash
QSocketNotifier: Can only be used with threads started with QThread
``` 

#### Additional information

**Complete Log file**

```bash
1680873142.3912311 [INFO] [launch]: All log files can be found below /home/jishnu/.ros/log/2023-04-07-18-42-22-390651-Legion-2140630
1680873142.3913183 [INFO] [launch]: Default logging verbosity is set to INFO
1680873142.5266192 [INFO] [turtlesim_node-1]: process started with pid [2140631]
1680873142.5267770 [INFO] [turtle_tf2_broadcaster-2]: process started with pid [2140633]
1680873142.5499730 [turtlesim_node-1] QSocketNotifier: Can only be used with threads started with QThread
1680873142.6347790 [turtlesim_node-1] /opt/ros/humble/lib/turtlesim/turtlesim_node: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE
1680873142.6409402 [ERROR] [turtlesim_node-1]: process has died [pid 2140631, exit code 127, cmd '/opt/ros/humble/lib/turtlesim/turtlesim_node --ros-args -r __node:=sim'].
1680873722.7566974 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
1680873722.8312013 [turtle_tf2_broadcaster-2] Traceback (most recent call last):
1680873722.8313580 [turtle_tf2_broadcaster-2]   File "/home/jishnu/robotics/ros-git/tf2/install/tf2_python_package/lib/tf2_python_package/turtle_tf2_broadcaster", line 33, in <module>
1680873722.8316448 [turtle_tf2_broadcaster-2]     sys.exit(load_entry_point('tf2-python-package==0.0.0', 'console_scripts', 'turtle_tf2_broadcaster')())
1680873722.8317199 [turtle_tf2_broadcaster-2]   File "/home/jishnu/robotics/ros-git/tf2/install/tf2_python_package/lib/python3.10/site-packages/tf2_python_package/dtf_broadcaster.py", line 96, in main
1680873722.8317873 [turtle_tf2_broadcaster-2]     rclpy.shutdown()
1680873722.8319311 [turtle_tf2_broadcaster-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 126, in shutdown
1680873722.8320026 [turtle_tf2_broadcaster-2]     _shutdown(context=context)
1680873722.8320696 [turtle_tf2_broadcaster-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/utilities.py", line 58, in shutdown
1680873722.8321402 [turtle_tf2_broadcaster-2]     return context.shutdown()
1680873722.8322432 [turtle_tf2_broadcaster-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/context.py", line 102, in shutdown
1680873722.8323112 [turtle_tf2_broadcaster-2]     self.__context.shutdown()
1680873722.8323784 [turtle_tf2_broadcaster-2] rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:241
1680873723.0084956 [ERROR] [turtle_tf2_broadcaster-2]: process has died [pid 2140633, exit code 1, cmd '/home/jishnu/robotics/ros-git/tf2/install/tf2_python_package/lib/tf2_python_package/turtle_tf2_broadcaster --ros-args -r __node:=broadcaster1 --params-file /tmp/launch_params_uduscqkx'].

```

lonebots, 2023-04-08:
The problem was mainly because of the `VSCode`'s environment variable. A detailed solution is mentioned here  [Java: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE](https://stackoverflow.com/questions/75921414/java-symbol-lookup-error-snap-core20-current-lib-x86-64-linux-gnu-libpthread)], although the issue arise from a whole different scenario, the solution worked for me. 

**What** I tried ?**

I unset the `GTK_PATH`  environment variable for VSCode using the below command 

```bash
uset GTK_PATH
```

Since the issue was resolved I am closing the issue.

andrewkenreich, 2023-05-09:
I was able to partially recreate this - Could you guys let me know if you have a "GTK_PATH" set when you run "env" in the docker?

maybe something like this - "GTK_PATH=/snap/code/124/usr/lib/x86_64-linux-gnu/gtk-3.0"

andrewkenreich, 2023-05-09:
if so run "unset GTK_PATH" and let me know if it works now

sa-webb, 2023-05-18:
I am on Ubuntu 22.10 and `unset GTK_PATH` didn't work for me and my VS Code is not installed with snap but for some reason **it is a VS Code terminal issue**. 

For now I'm getting by with the default terminal.

haroldshields, 2023-06-16:
`unset GTK_PATH` worked for me.

gernotpokorny, 2023-06-21:
For me uninstall the snap package and instead install vscode via the .dev from the website worked for me. `unset GTK_PATH` also worked with the snap package. Using a normal terminal instead outside of vscode also works ^^

Adii-5442, 2023-09-27:
I'm currently using Ubuntu 22.04, and installed VScode from snap store.I was also getting the same error but running the command ```unset GTK_PATH``` perfectly worked for me

ruimaciel, 2023-11-06:
As this issue is the top search result that pops up when searching for this specific error message, I'm dropping by to let everyone know that it's possible to tweak Visual Studio Code to unset `GTK_PATH` for all users by going to one of your `settings.json` files (user or workspace) and update `terminal.integrated.env.linux` to set `GTK_PATH` as an empty string.

```
{
   // ...

    "terminal.integrated.env.linux": {
        "GTK_PATH": ""
    },

    // ...
}
```

bgtor, 2024-07-24:
Alternatively, if your workspace has a .env file which is referenced in settings ("python.envFile": "${workspaceFolder}/.env" is the default if not mistaken), then adding GTK_PATH="" to it also solves the issue.

hifron, 2024-10-21:
posix manpage for [pthread](https://manpages.ubuntu.com/manpages/noble/en/man7/pthreads.7.html)

apt search libpthread

```
libpthread-stubs0-dev/noble,now 0.4-1build3 amd64 [installed,automatic]

libpthread-workqueue-dev/noble 0.9.1-1build1 amd64
  thread pool library (development files)

libpthread-workqueue0/noble 0.9.1-1build1 amd64
  thread pool library

libpthreadpool-dev/noble 0.0~git20210507.1787867-2 amd64
  pthread-based thread pool for C/C++

libpthreadpool0/noble 0.0~git20210507.1787867-2 amd64
  pthread-based thread pool for C/C++

```

luca-bitvark, 2025-03-02:
I had to do this and it worked:   
vs code settings:   
```
"meld-diff.diffCommand": "/var/sw/meld.sh",
  "terminal.integrated.env.linux": {
    "GTK_PATH": "",
    "LD_LIBRARY_PATH": "",
    "GIO_MODULE_DIR": ""
  }
```
/var/sw/meld.sh (you have to make it executable):   
```
#!/bin/bash
unset GTK_PATH
unset LD_LIBRARY_PATH
unset GIO_MODULE_DIR
export LD_PRELOAD=/lib/x86_64-linux-gnu/libpthread.so.0
/usr/bin/meld "$@"

```

pelmato, 2025-04-04:
> I had to do this and it worked: vs code settings:
> 
> ```
> "meld-diff.diffCommand": "/var/sw/meld.sh",
>   "terminal.integrated.env.linux": {
>     "GTK_PATH": "",
>     "LD_LIBRARY_PATH": "",
>     "GIO_MODULE_DIR": ""
>   }
> ```

Yes, `"GIO_MODULE_DIR": null` helped me ! Thanks.