ros2/ros2 #1407: Cannot publish dynamic TF with this->get_node_topics_interface() in humble.

BlueBirdHouse, 2023-04-09:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04.2 LTS
- Installation type:
  binaries
- Version or commit hash:
  ROS humble
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue

Make a dynamic publisher as we always do in foxy.
```
rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr node_topics_handle = this->get_node_topics_interface();

std::unique_ptr<tf2_ros::TransformBroadcaster> tf_broadcaster_ = nullptr;
tf_broadcaster_ = std::make_unique<tf2_ros::TransformBroadcaster>(node_topics_handle);
```

#### Expected behavior

The dynamic TF broadcaster object can be initialized. 

#### Actual behavior

It will give you an error.
"passed non-default qos overriding options without providing a parameters interface"

#### Additional information

[image]

fujitatomoya, 2023-04-14:
@iuhilnehc-ynos if you have bandwidth, could you take a look at this?

BlueBirdHouse, 2023-04-14:
Can you be specific? How much bandwidth are you suggesting? Why a programming problem is related to bandwidth? You have not left any links or materials.

iuhilnehc-ynos, 2023-04-14:
From ([NodeT](https://github.com/ros2/geometry2/blob/foxy/tf2_ros/include/tf2_ros/transform_broadcaster.h#L52), [A sample](https://github.com/ros2/geometry2/blob/foxy/tf2_ros/test/test_transform_broadcaster.cpp#L44))
, I think the original intention of `TransformBroadCaster` is not expected to accept a parameter that type is `rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr` even if it worked in `foxy` as it used some trick and the `qos_overriding_options` is not imported at that time. 

Maybe the design intention I mentioned above, there is no deprecated time window to let users update even in the [`galatic`](https://github.com/ros2/geometry2/blob/galactic/tf2_ros/include/tf2_ros/transform_broadcaster.h#L56-L73).

I suggest using the pointer/shared_ptr of `rclcpp::Node` directly, because to support `rclcpp::node_interfaces::NodeTopicsInterface` in [`rolling`](https://github.com/ros2/geometry2/blob/rolling/tf2_ros/include/tf2_ros/transform_broadcaster.h#L59-L77) is impossible.

iuhilnehc-ynos, 2023-04-14:
@BlueBirdHouse 

> Can you be specific? How much bandwidth are you suggesting? Why a programming problem is related to bandwidth? You have not left any links or materials.

@fujitatomoya just asked if I have extra time to deal with this issue.

MiguelCompany, 2023-04-14:
> @BlueBirdHouse
> 
> > Can you be specific? How much bandwidth are you suggesting? Why a programming problem is related to bandwidth? You have not left any links or materials.
> 
> @fujitatomoya just asked if I have extra time to deal with this issue.

Think of it as _**workforce** bandwidth_

iuhilnehc-ynos, 2023-04-14:
To add a new constructor for `TransformBroadcaster`,

```
  template<class AllocatorT = std::allocator<void>>
  TransformBroadcaster(
    rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr node_topics,
    const rclcpp::QoS & qos = DynamicBroadcasterQoS(),
    const rclcpp::PublisherOptionsWithAllocator<AllocatorT> & options =
      rclcpp::PublisherOptionsWithAllocator<AllocatorT>())
  {
    publisher_ = rclcpp::create_publisher<tf2_msgs::msg::TFMessage>(
      node_topics, node_topics, "/tf", qos, options);
  }
```

with a deprecated message might be a solution. (not sure if there are side effects)

BTW. I don't think maintainers will agree with it.

BlueBirdHouse, 2023-04-14:
I understand that using rclcpp::Node will certainly work; however, I feel it is less convenient. I would like to isolate different functions in different classes without passing the shared printer node everywhere.
For example, I can have a main node class inherited from rclcpp::Node in the main CPP file. Then, I can have different sub-classes in different HPP files. Each file or sub-class publishes different topics. But, all these topics should be associated with the main class. In Foxy, I use this->get_XXX_interfaces() to pass the shared_ptr to sub-classes. However, I cannot use this trick to build a TransformBroadcaster object in Humble. If this->get_XXX_interfaces() will be completely impossible in rolling as you suggested. How can we publish different topics in different sub-classes and associated them with the main node class? Can you show me some example codes?

iuhilnehc-ynos, 2023-04-14:
According to https://github.com/ros2/rclcpp/blob/humble/rclcpp/include/rclcpp/create_publisher.hpp#L84-L86,
In humble, you could try with
```patch
rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr node_topics_handle = this->get_node_topics_interface();

std::unique_ptr<tf2_ros::TransformBroadcaster> tf_broadcaster_ = nullptr;
- tf_broadcaster_ = std::make_unique<tf2_ros::TransformBroadcaster>(node_topics_handle);
+ tf_broadcaster_ = std::make_unique<tf2_ros::TransformBroadcaster>(node_topics_handle, tf2_ros::DynamicBroadcasterQoS(), rclcpp::PublisherOptions()); 
```

but I don't think it could work in `rolling` any more.

BlueBirdHouse, 2023-04-15:
@iuhilnehc-ynos 
It works! That is amazing! 
But, would you please provide me with a detailed explanation?
How could you know it should be written like this?
If it was me, I may barely guess that the second parameter is **tf2_ros::DynamicBroadcasterQoS()**. For the third one, I may try to use **rclcpp::PublisherOptionsWithAllocator**.
It shows that I lack the ability to understand the hits from the library file. Are there some books or lectures that can help to improve these kinds of abilities?

iuhilnehc-ynos, 2023-04-17:
> It works! That is amazing!

Good to know.

> How could you know it should be written like this?
> If it was me, I may barely guess that the second parameter is **tf2_ros::DynamicBroadcasterQoS()**. For the third one, I may try to use **rclcpp::PublisherOptionsWithAllocator**.

Using `rclcpp::PublisherOptionsWithAllocator` can work too. The [rclcpp::PublisherOptions](https://github.com/ros2/rclcpp/blob/6c4afb3a702617d263079bf9a8c304fc4e0f34b3/rclcpp/include/rclcpp/publisher_options.hpp#L138) is just a type alias.

> It shows that I lack the ability to understand the hits from the library file. Are there some books or lectures that can help to improve these kinds of abilities?

I think you're on the right way, at least from asking about this kind of issue.

closed 2023-04-19 by BlueBirdHouse