ros2/ros2 #1411: Does ros2 humble support zero copy default? and don't need using specified API

Lannister-Xiaolin, 2023-04-15:
There is no official example for zero copy, and for image transport and other large size data, zero copy  is very neccessary!!!

Here is some example(galactic) from internet:

https://github.com/ZhenshengLee/ros2_shm_msgs/blob/humble/README_cn.md

There is some trouble for me:

      1、Do we need to using api like:  can_loan_messages, to using zero copy in humble?

      2、 Can we offer some official zero copy examples ?

fujitatomoya, 2023-04-15:
for questions, we have central QA site with https://answers.ros.org/questions/. probably you can get better answer there.

fujitatomoya, 2023-04-15:
> 1、Do we need to using api like:  can_loan_messages, to using zero copy in humble?

Yes. that is the interface to get the memory from the middleware to write the data directly.

> 2、 Can we offer some official zero copy examples ?

Always you can. I am happy to review if you are willing to make contribution.

btw, https://github.com/ros2/demos/blob/humble/demo_nodes_cpp/src/topics/talker_loaned_message.cpp there is already demonstration code for zero copy.
if you think that is not good enough, probably we can start creating issue to explain what can be addressed or added?

Lannister-Xiaolin, 2023-04-17:
Thanks, I will ask

> Yes. that is the interface to get the memory from the middleware to write the data directly.
>
>
> Always you can. I am happy to review if you are willing to make contribution.
>
> btw, [...]

Thanks, I will ask questions on QA site

closed 2023-04-17 by Lannister-Xiaolin

Lannister-Xiaolin, 2023-04-21:
> Yes. that is the interface to get the memory from the middleware to write the data directly.
>
>
> Always you can. I am happy to review if you are willing to make contribution.
>
> btw, [...]

I have a sugestion that we can specified in document or examples that for subscribes, we dont't need call any special api like publisher, otherwise It would make us confused.

fujitatomoya, 2023-04-21:
@Lannister-Xiaolin sorry i am not sure if understand your suggestion.

are you saying that it would be better to support zero copy API only on subscription side? why is that? I think that is gonna be more burden for user application since it is likely to have multiple subscription endpoints.

i believe at least, we do need to get the physical shared memory area to write the data for user application, unless there will be copy on publisher side. I believe that `LoanedMessage` interface is mandatory.

Or is this suggestion to add `LoanedMessage` tutorial in the documentation? i think that makes sense to me, since there is no such description about `LoanedMessage`.

ZhenshengLee, 2024-05-09:
> Here is some example(galactic) from internet:
> 
> https://github.com/ZhenshengLee/ros2_shm_msgs/blob/humble/README_cn.md

the repo works for humble too, feel free to create issues in the repo if it doesn't work. @Lannister-Xiaolin