ros2/ros2 #1413: Abnormal output with ros2 node list

Rrooach, 2023-04-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
 - source code
- Version or commit hash:
 - humble
- DDS implementation:
 - fast-rtps
- Client library (if applicable):
  - 

#### Steps to reproduce issue
Basically, I installed ros2-humble on a remote server, and I installed and ran turtlebot3 and moveit2. However, after I killed moveit2 and turtlebot (I use ps aux to ensure that all ros related process has been killed), and I ran ```ros2 node info```, I still get output with turtlebot‘s and moveit's node info. I cannot clean this output, even I rebuild the entire ros environment, what could possibly cause the problem? 
Also, at first, I believe it is moveit2 problem only, since I only run moveit2 and only get moveit2-related node names. But after I run turtlebot and reproduce the same problem,  I realize there may be something wrong with the ros2 itself. 
#### Expected behavior
no output
#### Actual behavior
output with many node name
>>> ros2 node list            
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
/camera_driver
/gazebo
/moveit_cpp_tutorial
/moveit_cpp_tutorial
/moveit_cpp_tutorial
/moveit_cpp_tutorial
/moveit_cpp_tutorial_private_94247147953280
/moveit_cpp_tutorial_private_94485151199984
/robot_state_publisher
/static_transform_publisher
/static_transform_publisher
/transform_listener_impl_55b79fbe6cd0
/transform_listener_impl_55ef09d59ff0
/turtlebot3_diff_drive
/turtlebot3_imu
/turtlebot3_joint_state
/turtlebot3_laserscan

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2023-04-17:
> Basically, I installed ros2-humble on a remote server, and I installed and ran turtlebot3 and moveit2. However, after I killed moveit2 and turtlebot (I use ps aux to ensure that all ros related process has been killed), and I ran ros2 node info

It could be related to `ros2 daemon` still caches ros2 graph information, and `ros2 node list` queries to ros2 daemon to get those information to print.

Do you have reproducible environment with this problem? if that so, could you try `ros2 node list --no-daemon` to see if you still have the same problem?

Rrooach, 2023-04-18:
Thank for the tips, I use ```ros2 node list --no-daemon```, Indeed this command reduces many node information, but I still get the following output: 
```
>>ros2 node list --no-daemon                                                                      130 ↵
/static_transform_publisher
/static_transform_publisher
```

Also, 
> It could be related to `ros2 daemon` still caches ros2 graph information, and `ros2 node list` queries to ros2 daemon to get those information to print. 

Is this indicate a bug related to the ros2 cache functionality?