ros2/ros2 #1416: importlib.metadata.PackageNotFoundError: ros2cli

Cherub1Zhuang, 2023-04-25:
## Bug report
**Required Info:**
- Operating System:
  - Windows 10
- Installation type:
  - ource
- Version or commit hash:
  -  release version
#### Steps to reproduce issue

```
In the Windows environment, when installing ROS2, the previous installation follows the doc and is executed at the end
ros2 run demo_ nodes_ cpp talker
The following error occurred during:

Traceback (most recent call last):
  File "C:\dev\ros2_humble\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.9.13', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_humble\Scripts\ros2-script.py", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "C:\Python38\lib\importlib\metadata.py", line 504, in distribution
    return Distribution.from_name(distribution_name)
  File "C:\Python38\lib\importlib\metadata.py", line 177, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: ros2cli

```

clalancette, 2023-05-04:
Can you please give us more detailed information on exactly the steps you took?  As far as we know this works OK elsewhere, so there may be something in your environment that is subtly different.

clalancette, 2023-05-22:
No response in 3 weeks here, so I'm going to close this out.

closed 2023-05-22 by clalancette

Toskies, 2023-11-23:
do you resolve this question?   I have the same question.

mengxyokok, 2024-08-26:
the same qustion

yihangy, 2024-12-18:
the same qustion

wittrup, 2025-01-18:
This should be reopened.  Have the same question.
Followed Installation guide [Windows (binary)](https://docs.ros.org/en/jazzy/Installation/Windows-Install-Binary.html#windows-binary)
```
C:\Users\wittr>ros2 run demo_nodes_cpp talker
Traceback (most recent call last):
  File "C:\dev\ros2_jazzy\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.32.2', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_jazzy\Scripts\ros2-script.py", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "C:\Python38\lib\importlib\metadata.py", line 504, in distribution
    return Distribution.from_name(distribution_name)
  File "C:\Python38\lib\importlib\metadata.py", line 177, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: ros2cli
```