ros2/ros2 #1417: Specific Interface for ros2 msg

jelly-FF, 2023-04-25:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu18.04
- Installation type:
  - binaries
- Version or commit hash:
  - dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

3 interface are available on the device
- eth50 [ip is not fixed]: for internet visit
- eth1 192.168.1.103
- eth11 192.168.2.103
[image]

```
RMW_IMPLEMENTATION=rmw_fastrtps_cpp FASTRTPS_DEFAULT_PROFILES_FILE=fastdds.xml ros2 run [my-node] [my-node]
```
fastdds.xml
```
<?xml version="1.0" encoding="UTF-8" ?>
<profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
<transport_descriptors>
<transport_descriptor>
 <transport_id>udp</transport_id>
<type>UDPv4</type>
<interfaceWhiteList>
 <address>192.168.2.103</address>
  </interfaceWhiteList>
  </transport_descriptor>
 </transport_descriptors>
<!-- Default publisher profile -->
        <publisher profile_name="default publisher profile" is_default_profile="true">
            <qos>
                <publishMode>
                    <kind>SYNCHRONOUS</kind>
                </publishMode>
            </qos>
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </publisher>

        <!-- Default subscriber profile -->
        <subscriber profile_name="default subscriber profile" is_default_profile="true">
            <historyMemoryPolicy>PREALLOCATED_WITH_REALLOC</historyMemoryPolicy>
        </subscriber>
<participant profile_name="participant_profile_ros2" is_default_profile="true">
  <rtps>
  <name>profile_for_ros2_node</name>
<useBuiltinTransports>false</useBuiltinTransports>
<userTransports>
<transport_id>udp</transport_id>
</userTransports>
 </rtps>
</participant>
</profiles>
```

#### Expected behavior
ros2 msg are all sent by eth11

#### Actual behavior
rmw_fastrtps xml config file seems not work.
ros2 msg are sent by eth50, eth1, eth11.

When ros2 msg data size is large, eth50 will be occupied.
It will lead problem that ### device can not visit internet through eth50.
[image]

#### Additional information

fujitatomoya, 2023-04-25:
@jelly-FF thanks for creating issue.

1st, ROS dashing is E.O.L since May 2021, please see the supported distribution https://docs.ros.org/en/rolling/Releases.html.
So i am not sure if this is supported with ROS dashing, or Fast-DDS version supports xml configuration. could you try either Foxy or Humble to see if you have the same problem?

AFAIS, you xml configuration file seems to be correct, https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/whitelist.html

jelly-FF, 2023-04-26:
@fujitatomoya Thanks for the quick reply!

Our Operating System is ubuntu18.04.  It seems that neither Foxy nor Humble is supported on ubuntu18.04.
That's why we choose dashing instead [sad].

jelly-FF, 2023-04-26:
@fujitatomoya Besides, please check whether ip routing table is correct. Thanks!

clalancette, 2023-05-04:
It is difficult to tell here, as Dashing is so old.  We don't even know if these options work on that version.

My suggestion here would be at least to switch to Ubuntu 20.04, as that will let you use Foxy, and/or build Humble from source.

If you really can't switch to Ubuntu 20.04, you can also try compiling Foxy on Ubuntu 18.04; the instructions at http://docs.ros.org/en/foxy/Installation/Alternatives/Ubuntu-Development-Setup.html should largely work, though there may be some differences.

jelly-FF, 2023-05-09:
@fujitatomoya @clalancette 
I have tried compiling humble on Ubuntu 18.04, and using fastdds.xml to configure whitelist. It still does not work.

**Required Info:**

- Operating System:
  - ubuntu18.04
- Installation type:
  - from source
- Version or commit hash:
  - humble
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp_lifecycle

#### Steps to reproduce issue
```
RMW_IMPLEMENTATION=rmw_fastrtps_cpp FASTRTPS_DEFAULT_PROFILES_FILE=fastdds.xml ros2 run [my-node] [my-node]
```

#### Expected behavior
ros2 msg are all sent by eth11

#### Actual behavior
Lifecycle node cannot be seen by `ros2 lifecycle nodes`. Lifecycle node cannot be set activate. Ros2 msg cannot be sent.

P.S. Lifecycle node status can be set successfully without fastdds.xml config.
```
ros2 run [my-node] [my-node]
```

fujitatomoya, 2023-05-09:
> I have tried compiling humble on Ubuntu 18.04, and using fastdds.xml to configure whitelist.

I did not know this is even possible, see https://docs.ros.org/en/humble/Releases/Release-Humble-Hawksbill.html#supported-platforms. supported platform for humble is ubuntu 22.04 and 20.04 as tier 3...

Can you try to use docker or containers to bind host network interface? in that case, you can verify the problem with ubuntu22.04/humble for sure.

jelly-FF, 2023-05-10:
@fujitatomoya 

Without fastdds.xml config, I can compile and run humble lifecycle node successfully.

Our processor is NVIDIA Jetson AGX Xavier, so I am not able to use docker or containers.

fujitatomoya, 2023-05-10:
> Without fastdds.xml config, I can compile and run humble lifecycle node successfully.

okay, but that is technically unsupported. (I just did not know that humble can be compiled on Ubuntu 18.04...)

> Our processor is NVIDIA Jetson AGX Xavier, so I am not able to use docker or containers.

how about using `ubuntu22.04` docker container on NVIDIA Jetson AGX Xavier host system?

clalancette, 2026-10-26:
No response here in 3 years, so closing.  If you are still having issues, please open a new issue.

closed 2026-10-26 by clalancette