ros2/ros2 #1421: colcon build uses wrong version of python

alonsnir, 2023-05-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Humble
- DDS implementation:
  - Package: rti-connext-dds-6.0.1 Version: 6.0.1.25-1
- Client library (if applicable):
  - colcon I guess

#### Steps to reproduce issue

```
Python set via pyenv 
$ pyenv versions
  system
  3.6.15
  3.8.10
  3.8.15
  3.9.13
* 3.10.11 (set by PYENV_VERSION environment variable)

colcon build
Starting >>> robot_interfaces
--- stderr: robot_interfaces
CMake Error at /opt/ros/humble/share/python_cmake_module/cmake/Modules/FindPythonExtra.cmake:174 (message):
  execute_process(/home/dev/.pyenv/shims/python3.6 -c 'from sysconfig import
  get_config_var;print(get_config_var('SOABI'))') returned error code No such
  file or directory

Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:23 (find_package)
  /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:28 (rosidl_generate_interfaces)

---
Failed   <<< robot_interfaces [3.31s, exited with code 1]

```

#### Expected behavior

#### Actual behavior

#### Additional information
Initially I was using pipenv, but just in case have switched to the global Python3 version 

----

clalancette, 2023-05-02:
This is essentially a duplicate of #1094 .  In short, we can't use venvs currently, and someone needs to sit down with them and figure it out.

closed 2023-05-02 by clalancette

alonsnir, 2023-05-02:
@clalancette thanks for getting back

Please note, I have switched to non virtual, but the issue still there. Any idea how to repair it?

alonsnir, 2023-05-02:
@clalancette this is not #1094 duplicate

alonsnir, 2023-05-03:
Apparently the problem was with my environment. 
Once all virtual environment has been removed and ros and colcon reinstalled all back to normal

geoeo, 2024-02-06:
This is not specific to venvs. I get a similar but different error with no venv but multiple python versions. For example, I build 3.6 from source on ubuntu 22. Now it always defaults to 3.6 for some reason

```
haubenstockm@L1AAS97:~/git/ros2/allan_ws$ colcon build
Starting >>> px4_msgs
[Processing: px4_msgs]                             
[Processing: px4_msgs]                                     
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
--- stderr: px4_msgs                                        
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(_sre.o): warning: relocation against `PyExc_RecursionError' in read-only section `.text'
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(bytearrayobject.o): relocation R_X86_64_PC32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/px4_msgs__rosidl_generator_py.dir/build.make:3386: rosidl_generator_py/px4_msgs/libpx4_msgs__rosidl_generator_py.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:475: CMakeFiles/px4_msgs__rosidl_generator_py.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```