ros2/ros2 #1423: build error

goooodday, 2023-05-06:
Mac info.

MacOS Ventura 13.3.19(a)
MacBook Pro 16-inch, 2019
2.3GHz 8core Intel Core i9

ROS2 Humble

bellow, command & error message

$cd ~/ros2_humble/                                                                              
colcon build --symlink-install --packages-skip-by-dep python_qt_binding

<< error message >>

<pre>
Starting >>> ament_cmake_pycodestyle
Starting >>> ament_cmake_pyflakes
--- stderr: mcap_vendor
In file included from /Users/hyeongabshin/ros2_humble/src/ros2/rosbag2/mcap_vendor/src/main.cpp:16:
In file included from /Users/hyeongabshin/ros2_humble/build/mcap_vendor/_deps/mcap-src/cpp/mcap/include/mcap/mcap.hpp:3:
In file included from /Users/hyeongabshin/ros2_humble/build/mcap_vendor/_deps/mcap-src/cpp/mcap/include/mcap/reader.hpp:710:
/Users/hyeongabshin/ros2_humble/build/mcap_vendor/_deps/mcap-src/cpp/mcap/include/mcap/reader.inl:6:10: fatal error: 'zstd_errors.h' file not found
#include <zstd_errors.h>
         ^~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/mcap.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/mcap.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< mcap_vendor [2.25s, exited with code 2]
Aborted  <<< ament_cmake_xmllint [0.43s]
Aborted  <<< ament_cmake_cppcheck [0.42s]
</pre>

clalancette, 2023-05-18:
This generally falls under the fact that macOS is a Tier-3 platform (so is only community supported).  We aren't likely to have time to look into it, but if you find a solution please feel free to open a pull request.  We are happy to review them.

james-rms, 2023-05-25:
Hi @goooodday , does your build work if you force the zstd package to be built from source? ie. with `--cmake-args  -DFORCE_BUILD_VENDOR_PKG=ON`