ros2/ros2 #1429: ros2 daemon causes cli issue when switching distros

mikeferguson, 2023-05-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04, AMD64
- Installation type:
  - Binaries for both Humble and Iron
- DDS implementation:
  - Cyclone and FastRTPS show same behavior

#### Steps to reproduce issue

 * In Humble - "ros2 launch demo_nodes_cpp talker_listener.launch.py"
 * In Humble - "ros2 node list" then works
 * Shut all that down, switch to an Iron environment
 * In Iron - "ros2 launch demo_nodes_cpp talker_listener_launch.py" (note the . is now a _ before launch)
 * In Iron - "ros2 node list" fails - traceback is below
 * To get things working - "ros2 daemon stop" - and then "ros2 node list" works properly in Iron

#### Expected behavior

Should list nodes.

#### Actual behavior

```
ubr@ubr1:~/iron$ ros2 node list
Traceback (most recent call last):
  File "/opt/ros/iron/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.25.1', 'console_scripts', 'ros2')())
  File "/opt/ros/iron/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/iron/lib/python3.10/site-packages/ros2node/command/node.py", line 37, in main
    return extension.main(args=args)
  File "/opt/ros/iron/lib/python3.10/site-packages/ros2node/verb/list.py", line 38, in main
    node_names = get_node_names(node=node, include_hidden_nodes=args.all)
  File "/opt/ros/iron/lib/python3.10/site-packages/ros2node/api/__init__.py", line 61, in get_node_names
    node_names_and_namespaces = node.get_node_names_and_namespaces()
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1122, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1464, in __request
    response = self.__transport.request(
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1166, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1182, in single_request
    return self.parse_response(resp)
  File "/usr/lib/python3.10/xmlrpc/client.py", line 1354, in parse_response
    return u.close()
  File "/usr/lib/python3.10/xmlrpc/client.py", line 668, in close
    raise Fault(**self._stack[0])
xmlrpc.client.Fault: <Fault 1: "<class 'rclpy._rclpy_pybind11.RCLError'>:Failed to get node names: empty node name returned by the RMW layer, at ./src/rcl/graph.c:360">
```

#### Additional information

Ported from https://github.com/osrf/ros2_test_cases/issues/1056

clalancette, 2023-05-17:
Thanks.

In general, this is kind of a funny problem since it can only really come up in odd years (or when people build from source).  That is, at least installing from binaries, you would never run into this across Iron and J-Turtle, since the former is on Ubuntu 22.04 and the latter will be on Ubuntu 24.04.

The easy way to work around this is to do:

```
ros2 daemon stop
ros2 daemon start
```

which should kill of the old one and start up the new one.

There may also be a way we could take care of this automagically, by noticing that the daemon is running but with an old version.  We'd need to have the ROS distribution information available in the daemon, and we'd have to have the code recognize that there is an old one.  Then when we go to check whether the daemon is running, we could further check that it is the right "version" and kill off the old one if it isn't.

Anyway, those are just some thoughts about it.

mikeferguson, 2023-05-17:
I don't personally need anything else out of this - but it took me a while to figure out when I ran into it and so I posted on Twitter a warning so others wouldn't waste time on it - @kscottz asked me to ticket it (on the test party repo) and suggested we should include something in the release notes.

fujitatomoya, 2023-05-17:
@mikeferguson good catch.

as @clalancette mentioned, this is one of the corner case, i think. and the problem is technically `cross-distribution` communication, which is not supported or recommended. I am not sure what we can do this... maybe adding documentation in ROS 2 daemon section or installation troubleshooting?