ros2/ros2 #1430: Colcon build error i windows 10 ros2 Humble

sudarsan4, 2023-05-23:
C:\humble>colcon build --merge-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> fastcdr
Starting >>> ament_cppcheck
Starting >>> iceoryx_hoofs
Starting >>> osrf_pycommon
Starting >>> osrf_testing_tools_cpp
Finished <<< gtest_vendor [4.24s]
Starting >>> gmock_vendor
Finished <<< fastcdr [5.52s]
Starting >>> google_benchmark_vendor
Finished <<< iceoryx_hoofs [6.61s]
Starting >>> iceoryx_posh
--- stderr: ament_lint
c:\python38\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_lint [7.47s]
Starting >>> ament_flake8
--- stderr: osrf_pycommon
c:\python38\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< osrf_pycommon [7.61s]
Finished <<< gmock_vendor [3.67s]
Starting >>> urdfdom_headers
Starting >>> ament_pycodestyle
--- stderr: ament_cppcheck
c:\python38\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_cppcheck [8.20s]
--- stderr: ament_package
c:\python38\lib\site-packages\setuptools\command\easy_install.py:156: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
c:\python38\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ament_package [9.44s]
Starting >>> ament_cmake_core
--- stderr: ament_cmake_core
Traceback (most recent call last):
  File "C:\humble\src\ament\ament_cmake\ament_cmake_core\cmake\core\package_xml_2_cmake.py", line 22, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at cmake/core/ament_package_xml.cmake:95 (message):
  execute_process(C:/Program Files (x86)/Microsoft Visual
  Studio/Shared/Python39_86/python.exe
  C:/humble/src/ament/ament_cmake/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  C:/humble/src/ament/ament_cmake/ament_cmake_core/package.xml
  C:/humble/build/ament_cmake_core/ament_cmake_core/package.cmake) returned
  error code 1
Call Stack (most recent call first):
  cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  CMakeLists.txt:15 (ament_package_xml)

---
Failed   <<< ament_cmake_core [2.12s, exited with code 1]
Aborted  <<< ament_pycodestyle [3.94s]
Aborted  <<< ament_flake8 [4.55s]
Aborted  <<< google_benchmark_vendor [6.52s]
Aborted  <<< urdfdom_headers [4.36s]
Aborted  <<< osrf_testing_tools_cpp [32.3s]
Aborted  <<< iceoryx_posh [30.8s]

Summary: 8 packages finished [44.9s]
  1 package failed: ament_cmake_core
  6 packages aborted: ament_flake8 ament_pycodestyle google_benchmark_vendor iceoryx_posh osrf_testing_tools_cpp urdfdom_headers
  5 packages had stderr output: ament_cmake_core ament_cppcheck ament_lint ament_package osrf_pycommon
  328 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)

C:\humble>

Yadunund, 2023-05-23:
Did you follow the step in the [Installation Guide](https://docs.ros.org/en/humble/Installation/Alternatives/Windows-Development-Setup.html) that requires you to [install dependencies](https://docs.ros.org/en/humble/Installation/Alternatives/Windows-Development-Setup.html#install-dependencies) including `catkin_pkg`?

HosseinlGholami, 2023-07-05:
I do have the same problem. and the answer is yes I have that package.
I guess these codes are run on Venv or ... that could not use pythons library 

`
Traceback (most recent call last):
  File "C:\hubmble\src\ament\ament_cmake\ament_cmake_core\cmake\core\package_xml_2_cmake.py", line 22, in <module>
    from catkin_pkg.package import parse_package_string
ModuleNotFoundError: No module named 'catkin_pkg'
CMake Error at cmake/core/ament_package_xml.cmake:95 (message):
  execute_process(C:/ProgramData/anaconda3/python.exe
  C:/hubmble/src/ament/ament_cmake/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  C:/hubmble/src/ament/ament_cmake/ament_cmake_core/package.xml
  C:/hubmble/build/ament_cmake_core/ament_cmake_core/package.cmake) returned
  error code 1
Call Stack (most recent call first):
  cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  CMakeLists.txt:15 (ament_package_xml)

---
Failed   <<< ament_cmake_core [1.66s, exited with code 1]
Aborted  <<< iceoryx_hoofs [6.56s]
Aborted  <<< ament_flake8 [0.61s]
Aborted  <<< ament_pyflakes [0.16s]
Aborted  <<< gmock_vendor [1.22s]
Aborted  <<< fastcdr [7.20s]
Aborted  <<< osrf_testing_tools_cpp [20.7s]
Summary: 8 packages finished [25.5s]

  1 package failed: ament_cmake_core
  6 packages aborted: ament_flake8 ament_pyflakes fastcdr gmock_vendor iceoryx_hoofs osrf_testing_tools_cpp
  6 packages had stderr output: ament_cmake_core ament_cppcheck ament_lint ament_package ament_pycodestyle osrf_pycommon
  329 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)

C:\hubmble>pip install catkin_pkg
Requirement already satisfied: catkin_pkg in c:\python311\lib\site-packages (0.5.2)
Requirement already satisfied: docutils in c:\python311\lib\site-packages (from catkin_pkg) (0.20.1)
Requirement already satisfied: python-dateutil in c:\python311\lib\site-packages (from catkin_pkg) (2.8.2)
Requirement already satisfied: pyparsing in c:\python311\lib\site-packages (from catkin_pkg) (3.1.0)
Requirement already satisfied: setuptools in c:\python311\lib\site-packages (from catkin_pkg) (59.6.0)
Requirement already satisfied: six>=1.5 in c:\python311\lib\site-packages (from python-dateutil->catkin_pkg) (1.16.0)
`

cottsay, 2023-07-13:
You have multiple Python interpreters installed. Your pip command is using Python 3.11 while CMake is finding and using Python 3.8.

You could do any of the following:
1. Install `catkin_pkg` and other Python dependencies using Python 3.8 to match the interpreter CMake is finding.
2. Instruct CMake to specifically use the Python 3.11 interpreter using CMake variables.
3. Uninstall any additional interpreters so that there is only a single option.

mjcarroll, 2025-03-29:
Closing as stale, feel free to re-open if you have further information or questions.

closed 2025-03-29 by mjcarroll