ros2/ros2 #1433: ROS Humble install dependancy issue on Ubuntu 22.04.2

kuralme, 2023-05-27:
I use ubuntu 22.04.2, recently installed and enabled realtime-kernel. Having the appropriate libgbm1 version, got trouble installing ROS humble. Also tried to install iron irwini as well with no avail. Here is my system information and command output.
```
Static hostname: KME
       Icon name: computer-laptop
         Chassis: laptop
      Machine ID: ed0952f1284e4364b5e0998ad2d27214
         Boot ID: 92462c28388a4fe2af18a824569641c9
Operating System: Ubuntu 22.04.2 LTS              
          Kernel: Linux 5.15.0-1038-realtime
    Architecture: x86-64
 Hardware Vendor: ASUSTeK COMPUTER INC.
  Hardware Model: X550JX
```
```
kuralme@KME:~$ sudo apt install ros-humble-desktop
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgbm-dev : Depends: libgbm1 (= 22.2.5-0ubuntu0.1~22.04.1) but 22.2.5-0ubuntu0.1~22.04.2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

I would appreciate help, thank you in advance.

zzgames, 2023-05-27:
I was having the same problem but with sdl2. These 2 commands helped me:
```bash
sudo apt install libgbm1=22.2.5-0ubuntu0.1~22.04.1
sudo apt install libglapi-mesa=22.2.5-0ubuntu0.1~22.04.1
```

kuralme, 2023-05-29:
I don't think its a good solution to downgrade it. The problem you had is most likely caused because, to downgrade libgbm1 tens of packages is to be removed including ubuntu-desktop. Are you sure 'everything' is back to normal just reinstalling it?

```
...
The following packages will be REMOVED:
  apturl cheese evolution-data-server gdm3 gir1.2-gst-plugins-base-1.0
  gir1.2-mutter-10 gir1.2-rb-3.0 gir1.2-webkit2-4.0 gnome-calendar
  gnome-control-center gnome-initial-setup gnome-online-accounts
  gnome-remote-desktop gnome-session-bin gnome-shell
  gnome-shell-extension-appindicator gnome-shell-extension-desktop-icons-ng
  gnome-shell-extension-ubuntu-dock gnome-startup-applications gnome-todo
  gnome-user-docs gstreamer1.0-clutter-3.0 gstreamer1.0-gl gstreamer1.0-gtk3
  libcheese-gtk25 libcheese8 libclutter-1.0-0 libclutter-gst-3.0-0
  libclutter-gtk-1.0-0 libcogl-pango20 libcogl-path20 libcogl20
  libedataserverui-1.2-3 libegl-dev libegl-mesa0 libegl1 libgl1-mesa-dev
  libgles-dev libglvnd-dev libgoa-backend-1.0-1 libgstreamer-gl1.0-0
  libmutter-10-0 libwebkit2gtk-4.0-37 libyelp0 nautilus-share
  rhythmbox-plugins shotwell totem totem-plugins ubuntu-desktop
  ubuntu-desktop-minimal ubuntu-docs ubuntu-release-upgrader-gtk
  ubuntu-session update-manager update-notifier xorg xserver-xorg
  xserver-xorg-core xserver-xorg-input-all xserver-xorg-input-libinput
  xserver-xorg-input-wacom xserver-xorg-video-all xserver-xorg-video-amdgpu
  xserver-xorg-video-ati xserver-xorg-video-fbdev xserver-xorg-video-intel
  xserver-xorg-video-nouveau xserver-xorg-video-qxl xserver-xorg-video-radeon
  xserver-xorg-video-vesa xserver-xorg-video-vmware yelp zenity
The following NEW packages will be installed:
  policykit-1-gnome
The following packages will be DOWNGRADED:
  libgbm1
0 upgraded, 1 newly installed, 1 downgraded, 74 to remove and 0 not upgraded.
Need to get 59.4 kB of archives.
After this operation, 127 MB disk space will be freed.
```
I'll look for a more advisable solution for now, if not wait for developers to update

zzgames, 2023-05-29:
> Are you sure 'everything' is back to normal just reinstalling it?

Ubuntu was fresh new but yeah, everything was fine

Kooshal71, 2023-05-30:
> ```shell
> sudo apt install libgbm1=22.2.5-0ubuntu0.1~22.04.1
> sudo apt install libglapi-mesa=22.2.5-0ubuntu0.1~22.04.1
> ```

I tried to install the packages but i got this error:

>```
>The following packages have unmet dependencies:
> libegl1 : Depends: libegl-mesa0 but it is not going to be installed
>E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
>```

Did you have to change some other files in your case?

Kooshal71, 2023-05-30:
> I'll look for a more advisable solution for now, if not wait for developers to update

How long does it usually take for such an update to come out?

fujitatomoya, 2023-05-30:
CC: @razr @christophebedard @artivis any thoughts?

tfoote, 2023-05-31:
If you're upgrading the gbm libraries you need to make sure to upgrade the `-dev` package as well to be compatible. This is a dependency of one or more packages. And until that's installable on your system the ROS packages which depend on it will not be installable. 

As `libgbm-dev` appears to be pinned together with `libgbm1`  I would suggest that you look to your source of the upgraded library to also provide the `-dev` package for the new version too. 

Without further information this sounds like an external issue and not a ROS issue and I'll suggest closing it.

kuralme, 2023-05-31:
Hi.
After activating pro following the [tutorial](https://ubuntu.com/pro/tutorial), I enabled it by "sudo pro enable realtime-kernel" command. Here is my source list:
```
kuralme@KME:~$ apt policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 http://packages.microsoft.com/repos/code stable/main armhf Packages
     release o=code stable,a=stable,n=stable,l=code stable,c=main,b=armhf
     origin packages.microsoft.com
 500 http://packages.microsoft.com/repos/code stable/main arm64 Packages
     release o=code stable,a=stable,n=stable,l=code stable,c=main,b=arm64
     origin packages.microsoft.com
 500 http://packages.microsoft.com/repos/code stable/main amd64 Packages
     release o=code stable,a=stable,n=stable,l=code stable,c=main,b=amd64
     origin packages.microsoft.com
 500 https://esm.ubuntu.com/realtime/ubuntu jammy/main amd64 Packages
     release v=22.04,o=UbuntuRealtimeKernel,a=jammy,n=jammy,l=UbuntuRealtimeKernel,c=main,b=amd64
     origin esm.ubuntu.com
 500 https://esm.ubuntu.com/infra/ubuntu jammy-infra-updates/main amd64 Packages
     release v=22.04,o=UbuntuESM,a=jammy-infra-updates,n=jammy,l=UbuntuESM,c=main,b=amd64
     origin esm.ubuntu.com
 500 https://esm.ubuntu.com/infra/ubuntu jammy-infra-security/main amd64 Packages
     release v=22.04,o=UbuntuESM,a=jammy-infra-security,n=jammy,l=UbuntuESM,c=main,b=amd64
     origin esm.ubuntu.com
 500 https://esm.ubuntu.com/apps/ubuntu jammy-apps-security/main i386 Packages
     release v=22.04,o=UbuntuESMApps,a=jammy-apps-security,n=jammy,l=UbuntuESMApps,c=main,b=i386
     origin esm.ubuntu.com
 500 https://esm.ubuntu.com/apps/ubuntu jammy-apps-security/main amd64 Packages
     release v=22.04,o=UbuntuESMApps,a=jammy-apps-security,n=jammy,l=UbuntuESMApps,c=main,b=amd64
     origin esm.ubuntu.com
 500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
     release o=ROS,a=jammy,n=jammy,l=ROS jammy,c=main,b=amd64
     origin packages.ros.org
 500 https://nvidia.github.io/libnvidia-container/stable/ubuntu18.04/amd64  Packages
     release v=1.0,o=https://nvidia.github.io/libnvidia-container,a=bionic,n=bionic,l=NVIDIA CORPORATION <cudatools@nvidia.com>,c=
     origin nvidia.github.io
 500 https://download.docker.com/linux/ubuntu jammy/stable amd64 Packages
     release o=Docker,a=jammy,l=Docker CE,c=stable,b=amd64
     origin download.docker.com
 500 http://dl.google.com/linux/chrome/deb stable/main amd64 Packages
     release v=1.0,o=Google LLC,a=stable,n=stable,l=Google,c=main,b=amd64
     origin dl.google.com
 500 http://security.ubuntu.com/ubuntu jammy-security/multiverse i386 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=multiverse,b=i386
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=multiverse,b=amd64
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/universe i386 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=universe,b=i386
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=universe,b=amd64
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/restricted i386 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=restricted,b=i386
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=restricted,b=amd64
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/main i386 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=main,b=i386
     origin security.ubuntu.com
 500 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy-security,n=jammy,l=Ubuntu,c=main,b=amd64
     origin security.ubuntu.com
 100 http://de.archive.ubuntu.com/ubuntu jammy-backports/universe i386 Packages
     release v=22.04,o=Ubuntu,a=jammy-backports,n=jammy,l=Ubuntu,c=universe,b=i386
     origin de.archive.ubuntu.com
[... 40 lines cut ...]
     release v=22.04,o=Ubuntu,a=jammy,n=jammy,l=Ubuntu,c=universe,b=i386
     origin de.archive.ubuntu.com
 500 http://de.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy,n=jammy,l=Ubuntu,c=universe,b=amd64
     origin de.archive.ubuntu.com
 500 http://de.archive.ubuntu.com/ubuntu jammy/restricted i386 Packages
     release v=22.04,o=Ubuntu,a=jammy,n=jammy,l=Ubuntu,c=restricted,b=i386
     origin de.archive.ubuntu.com
 500 http://de.archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy,n=jammy,l=Ubuntu,c=restricted,b=amd64
     origin de.archive.ubuntu.com
 500 http://de.archive.ubuntu.com/ubuntu jammy/main i386 Packages
     release v=22.04,o=Ubuntu,a=jammy,n=jammy,l=Ubuntu,c=main,b=i386
     origin de.archive.ubuntu.com
 500 http://de.archive.ubuntu.com/ubuntu jammy/main amd64 Packages
     release v=22.04,o=Ubuntu,a=jammy,n=jammy,l=Ubuntu,c=main,b=amd64
     origin de.archive.ubuntu.com
Pinned packages:
     nsight-compute -> 2021.3.1.4~11.5.1-1ubuntu1 with priority -1
     nsight-systems -> 2021.3.3.2~11.5.1-1ubuntu1 with priority -1

```
Can the problem be Nvidia driver somehow? I was not able install nvidia container toolkit and generic driver in conventional ways, so i followed these steps mentioned here and installed it manually with few adjustments.

[image]

Anirudh-UG, 2023-05-31:
Hi i am actually facing a similar error...

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgbm-dev : Depends: libgbm1 (= 22.2.5-0ubuntu0.1~22.04.1) but 22.2.5-0ubuntu0.1~22.04.2 is to be installed
E: Unable to correct problems, you have held broken packages.

artivis, 2023-05-31:
@kuralme could you please post the output of,

```bash
apt-cache policy libgdm1
apt-cache show -a libgdm1
apt-cache policy libgdm-dev
apt-cache show -a libgdm-dev
```

kuralme, 2023-06-01:
Sure, here are the outputs:
```
kuralme@KME:~$ apt-cache policy libgdm1
libgdm1:
  Installed: 42.0-1ubuntu7.22.04.2
  Candidate: 42.0-1ubuntu7.22.04.2
  Version table:
 *** 42.0-1ubuntu7.22.04.2 500
        500 http://de.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     42.0-1ubuntu6 500
        500 http://de.archive.ubuntu.com/ubuntu jammy/main amd64 Packages
```
```
kuralme@KME:~$ apt-cache show -a libgdm1
Package: libgdm1
Architecture: amd64
Version: 42.0-1ubuntu7.22.04.2
Priority: optional
Section: libs
Source: gdm3
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 302
Depends: dconf-gsettings-backend | gsettings-backend, libc6 (>= 2.34), libglib2.0-0 (>= 2.44.0), libsystemd0
Breaks: gdm3 (<< 3.18.0-3~)
Replaces: gdm3 (<< 3.18.0-3~)
Filename: pool/main/g/gdm3/libgdm1_42.0-1ubuntu7.22.04.2_amd64.deb
Size: 61950
MD5sum: ff41c2ab05435493ec976fc359c0e6fb
SHA1: 2d06d89029bbf1997c7bc2bf8e51fb88ae390cb7
SHA256: 330ef438f86ef7b24899900139aebb3dbdf2a2eea6ff4135764e3f95dcd63309
SHA512: 28930f35f7b284848d67cf2e0e4eb9ccdc388348a1ed9b080c6e7918a41f07daa1ea6a1f4d645fe24ccc1de7b729574337e18be7a34aed0c8ff437ef0104822c
Homepage: https://wiki.gnome.org/Projects/GDM/
Description-en: GNOME Display Manager (shared library)
 GDM provides the equivalent of a "login:" prompt for X displays: it
 asks for a login and starts graphical sessions.
 .
 It supports multiple seats and switching between multiple users.
 .
 The greeter is based on the GNOME libraries and applications, and its
 look and design are the same as those of a GNOME session.
 .
 This package contains the shared library required by gnome-shell.
Description-md5: fbcf9de06b83ab57fb37bde788d9034b
Task: ubuntu-desktop-minimal, ubuntu-desktop

Package: libgdm1
Architecture: amd64
Version: 42.0-1ubuntu6
Priority: optional
Section: libs
Source: gdm3
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 302
Depends: dconf-gsettings-backend | gsettings-backend, libc6 (>= 2.34), libglib2.0-0 (>= 2.44.0), libsystemd0
Breaks: gdm3 (<< 3.18.0-3~)
Replaces: gdm3 (<< 3.18.0-3~)
Filename: pool/main/g/gdm3/libgdm1_42.0-1ubuntu6_amd64.deb
Size: 62514
MD5sum: 396d0327330b82a9d998d4ced23f1a4c
SHA1: eec1617de953f591d4710ea45f88be5e45d6c106
SHA256: 5b89009b381e2a90129a00b786cad0d11b7ed0bc52aa640ed57d50848822608e
SHA512: 2b617876ad407063a8b3c6a1f5c55b9dd82b113bd0ea21f86f917a1bb8ec52782b537c385dfb1ab3dc9aee2910db441ef13992028a63bb5e3d757aae4c9884f5
Homepage: https://wiki.gnome.org/Projects/GDM/
[... 65 lines cut ...]
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 491
Depends: gir1.2-gdm-1.0 (= 42.0-1ubuntu6), libgdm1 (= 42.0-1ubuntu6), libglib2.0-dev (>= 2.44.0)
Filename: pool/main/g/gdm3/libgdm-dev_42.0-1ubuntu6_amd64.deb
Size: 20030
MD5sum: 53e1e390fb72ecf9752e7b1c4d17775a
SHA1: 78a2e6879c98000b1892b1b50c3a5bbb28372465
SHA256: ae58a4ce3c02ebe405300cbfc8972302aa45930bfb06af93f1a56a6c810ac91b
SHA512: ee202d10355b0c5e9520442b638c2d7bcf8d3205ad83bd2cb4401e34e7974d526fcfd34e966d1a8764ee8b902f3e0aa25aa6be726421b789b23e043d3f22af7e
Homepage: https://wiki.gnome.org/Projects/GDM/
Description-en: GNOME Display Manager (development files)
 GDM provides the equivalent of a "login:" prompt for X displays: it
 asks for a login and starts graphical sessions.
 .
 It supports multiple seats and switching between multiple users.
 .
 The greeter is based on the GNOME libraries and applications, and its
 look and design are the same as those of a GNOME session.
 .
 This package contains header and development files.
Description-md5: 669514f6396aac0aea32903df3bc22ad

```

MirkoFerrati, 2023-06-01:
@kuralme in your first message the output was 
```
The following packages have unmet dependencies:
 libgbm-dev : Depends: libgbm1 (= 22.2.5-0ubuntu0.1~22.04.1) but 22.2.5-0ubuntu0.1~22.04.2 is to be installed
E: Unable to correct problems, you have held broken packages.
```
suggesting that libgbm1 needed to be installed, while apt-cache show 
```
libgdm1:
  Installed: 42.0-1ubuntu7.22.04.2
```
and same for the libgdm-dev library, also on 
```
libgdm-dev:
  Installed: 42.0-1ubuntu7.22.04.2
```
If there is a chain of dependencies that eventually breaks, you need to find which package has the wrong dependency so you can forward the issue to the right package maintainer. 
I believe that (given the apt-cache output from 2 hours ago) the command you used to install ros-humble-desktop should print a *different* output now, can you try? 
So be sure to have latest gdm 22.04.02 installed, and try to install ros on top and see the error, if you are lucky you will see a package requiring libgdm 22.04.01 in the output

kuralme, 2023-06-01:
@MirkoFerrati Unfortunately, the output is exactly the same as in my first command.

MirkoFerrati, 2023-06-01:
@kuralme  sorry I don't have the setup to reproduce locally, so I am just trying to help with hints. My idea was to follow  dependencies with apt-get install one by one. The approach from https://askubuntu.com/questions/707333/how-to-debug-an-unmet-dependencies-problem  but at this point I would try with the 
```
apt-get -s -o Debug::pkgProblemResolver=yes install ros-humble-desktop
```
as suggested in one of the solutions at the same link. Let's see the full output :)

brean, 2023-06-01:
I am having the same issue.
Downgrading libgdm to 22.04.01 will lead to problems with gnome, so that's not an option.

```
$ apt-get -s -o Debug::pkgProblemResolver=yes install ros-humble-desktop
NOTE: This is only a simulation!
      apt-get needs root privileges for real execution.
      Keep also in mind that locking is deactivated,
      so don't depend on the relevance to the real current situation!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Starting pkgProblemResolver with broken count: 1
Starting 2 pkgProblemResolver with broken count: 1
Investigating (0) libgbm-dev:amd64 < none -> 22.2.5-0ubuntu0.1~22.04.1 @un puN Ib >
Broken libgbm-dev:amd64 Depends on libgbm1:amd64 < 22.2.5-0ubuntu0.1~22.04.2 @ii pK > (= 22.2.5-0ubuntu0.1~22.04.1)
  Considering libgbm1:amd64 202 as a solution to libgbm-dev:amd64 0
Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgbm-dev : Depends: libgbm1 (= 22.2.5-0ubuntu0.1~22.04.1) but 22.2.5-0ubuntu0.1~22.04.2 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Small workaround for me: instead of full ros-humble-desktop I was able to just install ros-humble-ros-base and what you need ( rviz, ros-humble-rqt, ros-humble-ros-gz-sim ...), maybe that's enough for someone else as well.

kuralme, 2023-06-01:
Its ok @MirkoFerrati . My output is quite similar to @brean  's. 
```
kuralme@KME:~$ sudo apt -s -o Debug::pkgProblemResolver=yes install ros-humble-desktop
[sudo] password for kuralme: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Starting pkgProblemResolver with broken count: 1
Starting 2 pkgProblemResolver with broken count: 1
Investigating (0) libgbm-dev:amd64 < none -> 22.2.5-0ubuntu0.1~22.04.1 @un puN Ib >
Broken libgbm-dev:amd64 Depends on libgbm1:amd64 < 22.2.5-0ubuntu0.1~22.04.2 @ii pK > (= 22.2.5-0ubuntu0.1~22.04.1)
  Considering libgbm1:amd64 88 as a solution to libgbm-dev:amd64 0
Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgbm-dev : Depends: libgbm1 (= 22.2.5-0ubuntu0.1~22.04.1) but 22.2.5-0ubuntu0.1~22.04.2 is to be installed
E: Unable to correct problems, you have held broken packages.
```
I would also like to avoid downgrading the package since it would likely to broke many other dependants, but maybe downgrading the OS would be simpler solution.

brean, 2023-06-01:
@kuralme what do you mean with "downgrading the OS"? Switch to ROS 2 foxy and Ubuntu 20.04? ros is always tied closely to a specific ubuntu version, so downgrading isn't possible, so humble needs to work again with 22.04.

kuralme, 2023-06-02:
@brean No, I was referring to installing Ubuntu 22.04.1

marcosjl31, 2023-06-06:
I'm hitting the same problem when installing ROS2 humbe desktop on top of ubuntu 22.04.02

Downgrading to 22.04.01 is not an option.

Yoshiya-Fujii, 2023-06-06:
I am also facing the same issue. I am unable to install ROS2 Humble on Ubuntu 22.04.02. As a result, I cannot install Autoware, which relies on ROS2 Humble, and the development of autonomous driving is at a standstill. I hope that this issue will be resolved soon.

arsenyvasilyev, 2023-06-06:
I have a similar issue. I tracked it down to the `libgbm-dev` and `libgbm1` packages having had version `22.04.2_amd64.deb` deleted from the repo. I am not sure why this version was deleted. A workaround is to download the deleted version manually, and install via dpkg.

I was able to download these debs at the following URLs:
```https://bugs.launchpad.net/ubuntu/jammy/amd64/libgbm1/22.2.5-0ubuntu0.1~22.04.2
https://bugs.launchpad.net/ubuntu/jammy/amd64/libgbm-dev/22.2.5-0ubuntu0.1~22.04.2```

To download & install, run:
```wget http://launchpadlibrarian.net/665862724/libgbm1_22.2.5-0ubuntu0.1~22.04.2_amd64.deb
wget http://launchpadlibrarian.net/665862697/libgbm-dev_22.2.5-0ubuntu0.1~22.04.2_amd64.deb
dpkg -i libgbm1_22.2.5-0ubuntu0.1~22.04.2_amd64.deb 
dpkg -i libgbm-dev_22.2.5-0ubuntu0.1~22.04.2_amd64.deb```

MirkoFerrati, 2023-06-07:
it turns out that the  22.2.5-0ubuntu0.1\~22.04.2 update has been deleted due to a regression https://bugs.launchpad.net/ubuntu/+source/chromium-browser/+bug/2020604 but the people affected had the libgbm1 installed already from that. Now libgbm-dev is only available in the previous version 22.2.5-0ubuntu0.1\~22.04.1 and hence failing to install.

This will only be resolved once the currently proposed SRU is released to -updates.
You can see that in https://launchpad.net/ubuntu/+source/mesa/22.2.5-0ubuntu0.1~22.04.2/+publishinghistory
The reason we could not reproduce the issue is that if you setup a clean instance you won't hit this bug, only if you try to install on an instance that already updated to that version in the 3 days  it was in -updates like the people affected here.

clalancette, 2023-11-10:
Given that this was a packaging bug in Ubuntu 22.04, I don't think there is any action to take here.  I'm going to close this out, but please report another issue if you run into this again.

closed 2023-11-10 by clalancette

Akshat220302, 2024-01-11:
Hi, I am facing a similar issue. I recently uninstalled ros2 foxy on 20.04 and upgraded to 22.04 and have been trying to get ros2 humble.
sudo apt install ros-humble-desktop
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libignition-math6-dev : Depends: libignition-math6 (= 6.15.0-1~jammy) but 6.15.1-1~focal is to be installed
E: Unable to correct problems, you have held broken packages.
This is what I am getting. I wanted to try @arsenyvasilyev 's method but I don't have the correct build name. Help would be appreciated.

Akshat220302, 2024-01-11:
Update: I tried installing the missing package:
sudo apt install -y libignition-math6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libignition-math6 is already the newest version (6.15.1-1~focal).
libignition-math6 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
sudo apt install ros-humble-desktop
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libignition-math6-dev : Depends: libignition-math6 (= 6.15.0-1~jammy) but 6.15.1-1~focal is to be installed
E: Unable to correct problems, you have held broken packages.
But I am still getting the same error