ros2/ros2 #1436: No way to pass a byte array with ros2 run

marc0777, 2023-06-05:
I've created a simple ROS node that requires a BYTE_ARRAY as a parameter using:

```python
self.declare_parameter("param", [])
``` 

There doesn't seem a way however to actually pass a byte array using `ros2 run`, because it always gets interpreted as an integer array, no matter the way.

```
ros2 run my_pkg my_node --ros-args -p byte_array:=[0x11,0x12,0x13]
Error while starting My_node node: Trying to set parameter 'byte_array' to '[17, 18, 19]' of type 'INTEGER_ARRAY', expecting type 'BYTE_ARRAY'

ros2 run my_pkg my_node --ros-args --params-file test.yaml
Error while starting My_node node: Trying to set parameter 'byte_array' to '[17, 18, 19]' of type 'INTEGER_ARRAY', expecting type 'BYTE_ARRAY'
```

Where the `test.yaml` file looks like:
```yaml
my_node:
    ros__parameters:
        byte_array: [0x11,0x12,0x13]
```

Am I doing something wrong or is it simply not implemented? 

I'm using ROS humble installed on Ubuntu 22.04 using the ROS apt repositories.

GBergatto, 2023-06-06:
I'll play around with it today. But do we really need to pass in  a byte array? Couldn't we just pass an `INTEGER_ARRAY` and then convert it to whatever format we need in Python?

clalancette, 2023-06-15:
I think this is something that is currently not implemented.

The problem is that when we parse the command-line, we interpret it as YAML.  And there is no way for the YAML parser to disambiguate between integers and bytes currently.

There might be something we could do using the `!!` syntax of YAML to force a type, but someone would have to look into it.