ros2/ros2 #1439: Heap-use-after-free on copy_from_fastrtps_guid_to_byte_array()

Rrooach, 2023-06-14:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
Source code
- Version or commit hash:
 Humble
- DDS implementation:
FastRTPS
- Client library (if applicable):
RCLCPP

#### Steps to reproduce issue
Basically, I installed ros-humble on a remote server, and I installed and deploy moveit2 with [ASAN](https://github.com/google/sanitizers/wiki/AddressSanitize).  When I ran a simple demo in moveit2 (moveit_init_demo_warehouse), I get an ASAN report on the function copy_from_fastrtps_guid_to_byte_array().

#### Expected behavior
The program run without any report.

#### Actual behavior
=================================================================
==2506185==ERROR: AddressSanitizer: heap-use-after-free on address 0x61e0000110d0 at pc 0x7f738b44991a bp 0x7ffd49ea7670 sp 0x7ffd49ea6e18
READ of size 12 at 0x61e0000110d0 thread T0
    #0 0x7f738b449919 in __interceptor_memcpy ../../../../src/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:827
    #1 0x7f737ef2ecc1 in void rmw_fastrtps_shared_cpp::copy_from_fastrtps_guid_to_byte_array<unsigned char>(eprosima::fastrtps::rtps::GUID_t const&, unsigned char*) /path/to/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/guid_utils.hpp:55
    #2 0x7f737ef2e9e7 in rmw_fastrtps_shared_cpp::create_rmw_gid(char const*, eprosima::fastrtps::rtps::GUID_t const&) /path/to/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/create_rmw_gid.cpp:32
    #3 0x7f737ef7e309 in rmw_fastrtps_shared_cpp::__rmw_destroy_service(char const*, rmw_node_s*, rmw_service_s*) /path/to/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_service.cpp:62
    #4 0x7f737fdb390d in rmw_destroy_service /path/to/ros2/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_service.cpp:501
    #5 0x7f738527dac2 in rmw_destroy_service /path/to/ros2/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:523
    #6 0x7f73852b0737 in rcl_service_fini /path/to/ros2/src/ros2/rcl/rcl/src/rcl/service.c:196
    #7 0x7f738a4fbe20 in rclcpp::Service<rcl_interfaces::srv::ListParameters>::Service(std::shared_ptr<rcl_node_s>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<rcl_interfaces::srv::ListParameters>, rcl_service_options_s&)::{lambda(rcl_service_s*)#1}::operator()(rcl_service_s*) const /path/to/ros2/src/ros2/rclcpp/rclcpp/include/rclcpp/service.hpp:319
    #8 0x7f738a515659 in std::_Sp_counted_deleter<rcl_service_s*, rclcpp::Service<rcl_interfaces::srv::ListParameters>::Service(std::shared_ptr<rcl_node_s>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::AnyServiceCallback<rcl_interfaces::srv::ListParameters>, rcl_service_options_s&)::{lambda(rcl_service_s*)#1}, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() /usr/include/c++/12/bits/shared_ptr_base.h:527
    #9 0x560c5ea7f895 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/data/asan-mv/crash/moveit_init_demo_warehouse+0x29895)
    #10 0x7f738a3811de in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/12/bits/shared_ptr_base.h:1071
    #11 0x7f738a3bcfbf in std::__shared_ptr<rcl_service_s, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/12/bits/shared_ptr_base.h:1524
    #12 0x7f738a3bcfdf in std::shared_ptr<rcl_service_s>::~shared_ptr() /usr/include/c++/12/bits/shared_ptr.h:175
    #13 0x7f738a52d8f5 in rclcpp::ServiceBase::~ServiceBase() /path/to/ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/service.cpp:38
    #14 0x7f738a5168ff in rclcpp::Service<rcl_interfaces::srv::ListParameters>::~Service() /path/to/ros2/src/ros2/rclcpp/rclcpp/include/rclcpp/service.hpp:435
    #15 0x7f738a51abea in void std::_Destroy<rclcpp::Service<rcl_interfaces::srv::ListParameters> >(rclcpp::Service<rcl_interfaces::srv::ListParameters>*) /usr/include/c++/12/bits/stl_construct.h:151
    #16 0x7f738a517bdb in void std::allocator_traits<std::allocator<void> >::destroy<rclcpp::Service<rcl_interfaces::srv::ListParameters> >(std::allocator<void>&, rclcpp::Service<rcl_interfaces::srv::ListParameters>*) /usr/include/c++/12/bits/alloc_traits.h:648
    #17 0x7f738a515fa2 in std::_Sp_counted_ptr_inplace<rclcpp::Service<rcl_interfaces::srv::ListParameters>, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() /usr/include/c++/12/bits/shared_ptr_base.h:613
    #18 0x560c5ea7f6bc in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release_last_use_cold() (/data/asan-mv/crash/moveit_init_demo_warehouse+0x296bc)
    #19 0x7f738a3811de in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/12/bits/shared_ptr_base.h:1071
    #20 0x7f738a44f475 in std::__shared_ptr<rclcpp::Service<rcl_interfaces::srv::ListParameters>, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/12/bits/shared_ptr_base.h:1524
    #21 0x7f738a44f495 in std::shared_ptr<rclcpp::Service<rcl_interfaces::srv::ListParameters> >::~shared_ptr() /usr/include/c++/12/bits/shared_ptr.h:175
    #22 0x7f738a44f4b9 in rclcpp::ParameterService::~ParameterService() (/path/to/ros2/install/rclcpp/lib/librclcpp.so+0xa4f4b9)
    #23 0x7f738a44f524 in void std::_Destroy<rclcpp::ParameterService>(rclcpp::ParameterService*) (/path/to/ros2/install/rclcpp/lib/librclcpp.so+0xa4f524)
    #24 0x7f738a44ece3 in void std::allocator_traits<std::allocator<void> >::destroy<rclcpp::ParameterService>(std::allocator<void>&, rclcpp::ParameterService*) (/path/to/ros2/install/rclcpp/lib/librclcpp.so+0xa4ece3)
    #25 0x7f738a44e084 in std::_Sp_counted_ptr_inplace<rclcpp::ParameterService, std::allocator<void>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (/path/to/ros2/install/rclcpp/lib/librclcpp.so+0xa4e084)
    #26 0x560c5ea7f895 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/data/asan-mv/crash/moveit_init_demo_warehouse+0x29895)
    #27 0x7f738a3811de in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/12/bits/shared_ptr_base.h:1071
    #28 0x7f738a42712d in std::__shared_ptr<rclcpp::ParameterService, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/12/bits/shared_ptr_base.h:1524
    #29 0x7f738a42714d in std::shared_ptr<rclcpp::ParameterService>::~shared_ptr() /usr/include/c++/12/bits/shared_ptr.h:175
    #30 0x7f738a41f9e5 in rclcpp::node_interfaces::NodeParameters::~NodeParameters() /path/to/ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp:119
    #31 0x7f738a41fa65 in rclcpp::node_interfaces::NodeParameters::~NodeParameters() /path/to/ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp:119
    #32 0x7f738a409aa7 in std::_Sp_counted_ptr<rclcpp::node_interfaces::NodeParameters*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() /usr/include/c++/12/bits/shared_ptr_base.h:428
    #33 0x560c5ea7f895 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (/data/asan-mv/crash/moveit_init_demo_warehouse+0x29895)
    #34 0x7f738a3811de in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/12/bits/shared_ptr_base.h:1071
    #35 0x7f738a40077b in std::__shared_ptr<rclcpp::node_interfaces::NodeParametersInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/12/bits/shared_ptr_base.h:1524
    #36 0x7f738a403cd1 in std::__shared_ptr<rclcpp::node_interfaces::NodeParametersInterface, (__gnu_cxx::_Lock_policy)2>::reset() /usr/include/c++/12/bits/shared_ptr_base.h:1642
    #37 0x7f738a3fea8d in rclcpp::Node::~Node() /path/to/ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:272
    #38 0x560c5ea7f6bc in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release_last_use_cold() (/data/asan-mv/crash/moveit_init_demo_warehouse+0x296bc)
    #39 0x7f738b2f386e in std::unique_ptr<warehouse_ros::DatabaseLoader, std::default_delete<warehouse_ros::DatabaseLoader> >::~unique_ptr() (/data/asan-mv/install/moveit_ros_warehouse/lib/libmoveit_warehouse.so.2.5.4+0xd086e)
    #40 0x7f7389245a55 in __cxa_finalize (/lib/x86_64-linux-gnu/libc.so.6+0x45a55)
    #41 0x7f738b26cc76  (/data/asan-mv/install/moveit_ros_warehouse/lib/libmoveit_warehouse.so.2.5.4+0x49c76)

0x61e0000110d0 is located 2128 bytes inside of 2632-byte region [0x61e000010880,0x61e0000112c8)
freed by thread T0 here:
[... 21 lines cut ...]
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
[... cut ...]
#### Additional information
To better debug this issue, I build ros-humble with "-g" flag. 

#### Implementation considerations
It seems in eProsima/Fast-DDS/src/cpp/fastdds/subscriber/SubscriberImpl.cpp:SubscriberImpl::~SubscriberImpl() freed the guid_byte_array, where  src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/guid_utils.hpp:void rmw_fastrtps_shared_cpp::copy_from_fastrtps_guid_to_byte_array uses ```memcpy()``` to read it without any checking causes this UAF read.

fujitatomoya, 2023-06-15:
@Rrooach thanks for posting the issue here.

```bash
0x61e0000110d0 is located 2128 bytes inside of 2632-byte region [0x61e000010880,0x61e0000112c8)
freed by thread T0 here:
#0 0x7f738b4c0fe8 in operator delete(void*) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:152
https://github.com/ros2/ros2/issues/1 0x7f737e7977cd in eprosima::fastdds::dds::SubscriberImpl::~SubscriberImpl() /path/to/ros2/src/eProsima/Fast-DDS/src/cpp/fastdds/subscriber/SubscriberImpl.cpp:160
```

do you happen to have this callstack? or can we reproduce this issue using sample or demos?

Rrooach, 2023-06-16:
@fujitatomoya Thanks for your kind reply.

> ```shell
> #0 0x7f738b4c0fe8 in operator delete(void*) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:152
> https://github.com/ros2/ros2/issues/1 0x7f737e7977cd in eprosima::fastdds::dds::SubscriberImpl::~SubscriberImpl() /path/to/ros2/src/eProsima/Fast-DDS/src/cpp/fastdds/subscriber/SubscriberImpl.cpp:160
> ```

This callback is emitted by ASAN itself, indicating how this issue is triggered (from moveit2 to ros-humble).

> can we reproduce this issue using sample or demos?

To reproduce this issue we need to install moveit2 and ros2-humble with the Clang compiler, and with ASAN enabled, then run moveit_init_demo_warehouse within the moveit.
```bash
colcon build --cmake-args  -DCMAKE_CXX_FLAGS="-g  -fsanitize=address"  -DCMAKE_C_FLAGS="-g -fsanitize=address "
```

After compiling, use ```objdump``` to check if the instrumentation is successful.
```bash
objdump -t ./install/moveit_ros_warehouse/lib/moveit_ros_warehouse/moveit_init_demo_warehouse  | grep asan
0000000000000000 l    df *ABS*	0000000000000000              asan_preinit.cpp
0000000000000000       F *UND*	0000000000000000              __asan_unpoison_stack_memory
0000000000000000       F *UND*	0000000000000000              __asan_stack_free_9
0000000000000000       F *UND*	0000000000000000              __asan_report_load4
0000000000000000       F *UND*	0000000000000000              __asan_report_store_n
```
then simply run  ./install/moveit_ros_warehouse/lib/moveit_ros_warehouse/moveit_init_demo_warehouse can trigger this issue.

BTW, I still have a few more bugs, should I keep posting them in this repository or I can email them to someone with detailed information? 

Best regards,

MiguelCompany, 2023-06-16:
Since the call stack stems from `__cxa_finalize` I think this might be related to https://github.com/ros2/rclcpp/issues/1766

`eprosima::fastdds::dds::SubscriberImpl::~SubscriberImpl()` can only be called from two places:
1. While shutting down the rmw context, it will be called [here](https://github.com/ros2/rmw_fastrtps/blob/rolling/rmw_fastrtps_shared_cpp/src/participant.cpp#L417)
2. While automatically destroying the `DomainParticipantFactory` singleton on process exit.

I think we are in case nb 2 here.
@Rrooach Is `std::unique_ptr<warehouse_ros::DatabaseLoader, std::default_delete<warehouse_ros::DatabaseLoader>>` in a global static variable?

Rrooach, 2023-06-16:
> Is `std::unique_ptr<warehouse_ros::DatabaseLoader, std::default_delete<warehouse_ros::DatabaseLoader>>` in a global static variable?

Yes, within the original codebase, this issue is triggered as a global static variable. But during this [commit](https://github.com/ros-planning/moveit2/commit/358bad65361a4119e0927155237f925c446baf58), it has become a static local variable within the loadDatabase function. 

As of now, I still try to recompile this new code, but with tons of compile error X(

clalancette, 2023-06-29:
@MiguelCompany So does this seem like a problem with the Fast-DDS destruction to you?  Or maybe something else?

MiguelCompany, 2023-06-30:
> @MiguelCompany So does this seem like a problem with the Fast-DDS destruction to you? Or maybe something else?

@clalancette Not exactly. I think it is the same thing as I commented [here](https://github.com/ros2/rclcpp/issues/1766#issuecomment-972631719)

For iron and rolling, we could overcome this if on [this struct](https://github.com/ros2/rmw_fastrtps/blob/rolling/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L92) we keep the result of calling [this method](https://github.com/eProsima/Fast-DDS/blob/802f224a9f4b2af6bb2500d7d689a4870e440581/include/fastdds/dds/domain/DomainParticipantFactory.hpp#L81)

This would delay the destruction of the `DomainParticipantFactory` until the rmw context is destroyed.