ros2/ros2 #1441: "Lazy" Services

emersonknapp, 2023-06-14:
## Feature request

"Lazy" services, like the "lazy publisher" pattern, should only incur costs if its service will actually be used.

Motivation: services create two topics each in current DDS-based RMW implementations, and discovery costs are N^2 with number of topics in the system. Each node by default creates a variety of parameter-related services already, and will create a log-level service and get-type-description service in new releases. This adds up to 10+ builtin topics per node, most of which are never used, and can bog down a network with discovery traffic on application startup.

The service client assets itself within the node are a minor consideration, but there is probably some small memory benefit to not creating service servers until they are needed.

#### Feature description

A type of Service Server in ROS 2 that only creates network resources (and therefore only incurs discovery costs) when a Service Client is created to request use of that service. 

#### Implementation considerations

* Feature should available for both C++ and Python nodes
* Can you list lazy services? For `ros2 service list` or `ros2 node info`, it would be nice to have access to these.
* How much lag would this delayed initialization introduce for Clients, compared to having the Service created immediately on Node startup?

emersonknapp, 2023-06-14:
@wjwwood FYI

fujitatomoya, 2023-06-14:
> This adds up to 10+ builtin topics per node, most of which are never used, and can bog down a network with discovery traffic on application startup.

totally agree on this, this has been concern for me.

> Can you list lazy services? For ros2 service list or ros2 node info, it would be nice to have access to these.

if we can list lazy services, that means the we have to create `DataWriter` and `DataReader` for the service? in that case, we could not reduce the discovery cost which is the main motivation here?

> The service client assets itself within the node are a minor consideration

if the both service server and client can be lazy, there is no way to detect the other endpoint is available? because either must be discovered in the ROS 2 network to be detected on the other side?

`matched_event` could be one of the solution, but currently it requires to create `DataWriter`/`DateReader` to receive the `StatusKind` by listener. (i may be missing something though.) that said, we need to create `DataWriter`/`DateReader` to receive those `matched_events`. this does not seem to work to reduce the discovery cost with `lazy` endpoint creation.
or we can rely on `graph` discovery data to receive the graph change event to check the lazy service server detects the matched client?
probably there would be more ideas how to implement this feature, that would be nice to discuss more.

related to https://github.com/ros-perception/image_common/issues/111

emersonknapp, 2023-06-14:
> if the both service server and client can be lazy

I think "lazy client" doesn't make sense - just like "lazy subscriber". The concept should probably limit to data producers/service providers. 

@wjwwood suggested an implementation that a `LazyService` would watch the ros graph for any `Client`s that are created, then it would create the underlying `Service` to match. We have `rmw_get_service_names_and_types` - which will return a service even if only an unmatched Client exists. However we don't have a way to list "number of clients" or "number of servers" for a service.

* We have `rmw_get_service_names_and_types` - which will list a service even if there only exists an unmatched Client
* There is `rcl_get_publishers_info_by_topic` - which when given a topic information , returns publishers for it
* There is not an `rcl_get_clients_info_by_service` or `rcl_get_servers_info_by_service` that would tell us how many Servers/Clients there are

Maybe we could make the assumption that there is only one service server, and so if the service server sees the service in the graph via `rmw_get_service_names_and_types`, then it can assume there is a client, then wake up and create the server.

Note: having multiple servers for the same client produces undefined behavior for clients, but it seems difficult or impossible to explicitly disable anywhere. Though maybe it would be possible, with some new APIs, to give warnings to service servers, if they discover another server.

If that assumption isn't good enough, then we probably need some new RMW APIs to list more information about participants in a service.

wjwwood, 2023-08-02:
Here's my prototype of this in Python:

```python
import threading

import rclpy
from rclpy.expand_topic_name import expand_topic_name
from rcl_interfaces.srv import DescribeParameters

hack_works = None

class LazyService:
    def __init__(self, node, service_type, service_name, *args, **kwargs):
        self.__node = node
        self.__service_type = service_type
        self.__service_name = service_name
        expanded_service_name = \
            expand_topic_name(service_name, node.get_name(), node.get_namespace())
        self.__expected_request_topic_name = f'rq{expanded_service_name}Request'
        self.__args = args
        self.__kwargs = kwargs
        self.__service = None
        self.__canceled = False
        self.__wait_for_client_thread = None
        if self.__check_that_hack_works():
            node.get_logger().info('hack works')
            self.__wait_for_client_thread = threading.Thread(target=self.wait_for_client)
            self.__wait_for_client_thread.start()
        else:
            node.get_logger().info('hack does not work')
            # Hack will not work, just create the service now...
            self.__create_service()

    def __check_that_hack_works(self):
        global hack_works
        if hack_works is not None:
            return hack_works
        # check if the service names are listed as topics with no_demangle,
        # which is required for this approach to work...
        # arbitrarily using DescribeParameters, the type doesn't actually matter
        def do_nothing(request, response):
            return response

        temp_service = \
            self.__node.create_service(DescribeParameters, '~/_/lazy_service_test', do_nothing)
        topic_names_and_types = self.__node.get_topic_names_and_types(no_demangle=True)
        temp_service.destroy()

        fq_node_name = self.__node.get_fully_qualified_name()
        expected_topic = f'rq{fq_node_name}/_/lazy_service_testRequest'
        hack_works = expected_topic in [x for x, y in topic_names_and_types]
        return hack_works

    def wait_for_client(self):
        while not self.__canceled:
            # check for at least one client (subscription on the request topic)
            # [print(x) for x, y in self.__node.get_topic_names_and_types(no_demangle=True)]
            print(f'looking for: {self.__expected_request_topic_name}')
            publishers_on_expected_topic = self.__node.get_publishers_info_by_topic(
                self.__expected_request_topic_name,
                no_mangle=True)
            # [print(x) for x in publishers_on_expected_topic]
            if len(publishers_on_expected_topic) > 0:
                self.__create_service()
                return
            self.__node.get_clock().sleep_for(rclpy.duration.Duration(seconds=1))

    def __create_service(self):
        self.__node.get_logger().info(
            f"Creating ROS Service on '{self.__service_name}' with "
            f"type '{self.__service_type.__name__}'")
        self.__service = self.__node.create_service(
            self.__service_type,
            self.__service_name,
            *self.__args,
            **self.__kwargs)

    def destroy(self):
        self.__canceled = True
        if self.__wait_for_client_thread is not None:
            self.__wait_for_client_thread.join()
            self.__wait_for_client_thread = None
        if self.__service is not None:
            self.__service.destroy()
        self.__service = None

def main():
    rclpy.init()
    n = rclpy.create_node('lazy_service_example_node')
    # again arbitrarily using DescribeParameters, but any type can be used
    def on_lazy_describe_parameters(request, response):
        print(f"In on_lazy_describe_parameters({request}, {response})")
        return response
    LazyService(n, DescribeParameters, '~/lazy_describe_parameters', on_lazy_describe_parameters)
    try:
        rclpy.spin(n)
    except KeyboardInterrupt:
        print("[ctrl-c], shutting down...")

if __name__ == '__main__':
    main()

```

I tested this by using this one liner to create a client:

```
ros2 service call /lazy_service_example_node/lazy_describe_parameters rcl_interfaces/srv/DescribeParameters {}
```

It does work, but since in Python we don't (seem) to have access to the graph guard condition and so it must be polled.

Even still this is pretty efficient and not too bad on latency. The C++ version would be truly non-blocking and should have even lower latency since it's event driven.

Also, I'm doing some funny things with the Topic introspection, which relies on the fact that the Services are implemented with Topics in our current implementations.

To do this properly we would want to implement:

- graph guard condition exposed in Python, and
- something like "get service clients info by service name" in rmw, rclcpp, and rclpy

But it at least shows that the idea works fine.

It would also be great to have a Services version of `wait_for_all_acked()`, which I noticed when testing this, e.g. shutting down in the Service server callback sometimes resulted in the response not being delivered to the `ros2 service call` cli in the other terminal.

wjwwood, 2023-08-02:
Obviously we'd need to address the other design points that you guys pointed out, like whether or not the Service Server is ever destroyed again or is left running after one access, and also how could we exchange the available lazy services, so that we could list them in tools like `ros2 service list --show-lazy`, for example.

fujitatomoya, 2023-08-02:
> graph guard condition exposed in Python, and
> something like "get service clients info by service name" in rmw, rclcpp, and rclpy

sounds good to me.

> Services version of wait_for_all_acked(),

not only for Lazy Service, but in general this would be useful for service server configuration option (probably with timeout).

> like whether or not the Service Server is ever destroyed again or is left running after one access

i think either way makes sense. the one is `once the service is created, never needs to check the graph change`, and the other is `once no clients are online, destroy the service. so that we can reduce discovery cost.`

traverseda, 2025-07-19:
>and discovery costs are N^2 with number of topics in the system.

Sorry for bumping this thread, but how do you know that? Is this documented somewhere? I'm trying to get more examples of real-world limitations

fujitatomoya, 2025-07-19:
DDS-based implementation creates the 2 individual topics with prefix Request/Replay for a single service endpoint such as service client. this leads to N^2 discovery costs in the system.

as implementation example, you can take a look around the following.

https://github.com/ros2/rmw_fastrtps/blob/09e5a5f1b9504431bbc8970bb098438041041dbe/rmw_fastrtps_cpp/src/rmw_client.cpp#L153-L156