ros2/ros2 #1446: Unable to connect ROS2 Humble on host to same in Docker

andrey-pr, 2023-06-27:
## Bug report

Hello. I'm trying to use my ros-based programs in docker. It works inside, but I cannot connect external ros programs on host. It just doesn't see anything inside docker. To make network available, I tried to use --net=host, but it didn't help. I need to connect ros on host to ros in docker to view visualizarions in rviz, because I use in docker ubuntu image without graphics.

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Humble
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Launch any docker image with ros with argument --net=host and try to connect to ros from host.

#### Expected behavior

It must connect ok, because I set --net=host

#### Actual behavior

ros on host can not see ros in container

Thank you

ajpether, 2023-11-29:
I just spent 24 hours banging my head against the wall trying to figure out why my ROS2 nodes inside docker couldn't communicate with the outside world. Finally figured out that it was because I set up rootless mode on my docker engine (as prompted to do so during docker installation) instead of just adding myself to the docker group. Putting this here in the hope it might help someone else avoid my pain.

fujitatomoya, 2023-11-29:
@andrey-pr besides @ajpether 's comment above, can you try `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`? it might be implementation to select the specific network interface? is there any warning or information message comes up when you start the application?