ros2/ros2 #1447: Launching more than two launch files from the same package gives "Package not found"

cyborgdennett, 2023-06-30:
## Bug report

**Required Info:**

- Operating System:
  - Windows 11 WSL Ubuntu 22.04
- Installation type:
  - sudo apt install
- Version or commit hash:
  - Humble 
- DDS implementation:
  - 
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
Make a package with 3 launch files.
Open 3 terminals
Launch a launch files in each of the terminals
ex:
>ros2 launch my_package launch_1.py
>ros2 launch my_package launch_2.py
>ros2 launch my_package launch_3.py
```

#### Expected behavior
I expect to have 3 launches

#### Actual behavior
Running launch_3.py gives error 'my_package' not found

#### Additional information
After closing launch_2.py terminal,
running launch_3.py will not give this error anymore

However, then running launch_2.py gives the 'my_package' not found error

**NOTE**
I just tested this on a clean package, and the problem did not occur when trying to make a reproducible problem.

Image of problem:
[image]
The third terminal cannot find the packages anymore, while the first two are running.

fujitatomoya, 2023-07-10:
I am not sure what went wrong, could you try the following commands?

1. build the package once.

```bash
> colcon build --packages-select my_package
```

2, start executables from each terminal

- terminal-1

```bash
> source install/setup.bash
> ros2 launch my_package AAA_launch.py
```

- terminal-2

```bash
> source install/setup.bash
> ros2 launch my_package BBB_launch.py
```

- terminal-3

```bash
> source install/setup.bash
> ros2 launch my_package CCC_launch.py
```

clalancette, 2026-10-26:
No response here in 3 years, so closing.

closed 2026-10-26 by clalancette