ros2/ros2 #1450: Will ROS2 support ubuntu 23.04

Kingkazara, 2023-07-01:
## Bug report

**Required Info:**

- Operating System:
  - <ubuntu 23.04>
- Installation type:
  - <from resource>
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

Kingkazara, 2023-07-01:
I do not see any installation instructions for Ubuntu 23.04; if there would be a new version to support Ubuntu 23.04 when it would come

clalancette, 2023-07-01:
In short, the current policy is to only support the Ubuntu LTS releases (on the even years; so 20.04, 22.04, 24.04, etc).  That said, it will likely work to build from source on 23.04, so if you want to give that a go, please follow the instructions at http://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html.  If you run into any problems, we are happy to review pull requests that fix things.

closed 2023-07-01 by clalancette

AndreCox, 2024-01-25:
Just replying to this to say that building ros2 from source on ubuntu 23.10 is totally possible. 
You will need to run this command.
```rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers ignition-math6 ignition-cmake2 python3-catkin-pkg-modules python3-vcstool python3-rosdistro-modules"```