ros2/ros2 #1451: how to check qos History(Depth)

jelly-FF, 2023-07-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Humble
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - ros2cli

#### Steps to reproduce issue

1. run minimal_publisher
```
ros2 run examples_rclcpp_minimal_publisher publisher_member_function
```

2. check qos profile
```
ros2 topic info -v /topic
```

#### Expected behavior

show all the qos profile correctly

#### Actual behavior

Most of the qos profile are shown correctly, except History (Depth). 
History (Depth) should not be "UNKNOWN".

[image]

clalancette, 2023-07-04:
Yes, you are correct.  This is unfortunately a limitation of Fast-RTPS at the moment; see the comment in https://github.com/ros2/rclpy/pull/849#issuecomment-988548722 for more details.

I'll leave this open to track this missing feature on the backlog.

jelly-FF, 2023-07-05:
@clalancette 
Is there any quick fix I can try in source code to [set QoS history nor depth during endpoint discovery]?

fujitatomoya, 2023-07-05:
@jelly-FF the behavior is dependent on rmw implementation (Fast-DDS), unfortunately i am not sure what we can try to patch this.

CC: @MiguelCompany @clalancette

jelly-FF, 2023-07-05:
@fujitatomoya okay! Thanks.