ros2/ros2 #1452: [Meta-Ticket] J-turtle changes for Ubuntu 23/24

Ryanf55, 2023-07-05:
# Scope of ticket

I've filed this meta-ticket to gain visibility into all the tasks that must be done when upgrading from Ubuntu 22 to the next version of Ubuntu. Although ROS 2 `rolling` currently has no planned support for compiling on Ubuntu Lunar 23.04, it may be used as a stepping stone to ease the migration pain to the unreleased Ubuntu 24. Furthermore, Ubuntu 23 supports Linux kernel 6 out of the box, which is useful for developers like me that have new hardware on their machines not supported on Ubuntu 22. I am NOT asking for any changes to REP2000 to officially support Ubuntu 23 as a tier 1 platform, nor am I asking the ROS CI infrastructure to support binary releases on Ubuntu 23.  

## Proposal

1. Allow developers to use Ubuntu 23 to compile and develop ROS 2 `rolling` without warnings/errors.
1. Allow/add documentation for Ubuntu 23 on the ROS documentation site on `rolling` that applies specifically to future versions. Prevent it from merging to `iron` docs. This has the advantage of providing a clear path for developers like me to contribute.
1. For CMake, attempt to remove all warnings when using CMake 3.27 to include future policy changes that will likely go into Ubuntu 24. This would involve bumping minimum CMake to above 3.5, among others.
1. For QT, remove dependencies on QT5 since it's now deprecated. Whether the packages are deprecated entirely or upgraded to the next applicable QT version has to be decided.
1. Allow rosdep to allow Ubuntu 23 specific package changes, since some have changed names from Ubuntu 22.
1. Release ros-dev-tools for Ubuntu 23. 

## Next steps

- [ ] Modify/agree on the above proposal
- [ ] Create follow-up tickets for each of the applicable repos and link them here
- [ ] Get help from the community
- [ ] Once rep-2000 is decided for versions of dependencies in J-turtle, this meta-ticket can be modified

clalancette, 2023-07-05:
First, thanks for bringing this up!  Starting on this now (rather than waiting for 24.04) should help ease the transition.  I have individual notes below:

>     1. Allow developers to use Ubuntu 23 to compile and develop ROS 2 `rolling` without warnings/errors.

Yes, that's a good goal.

>     2. Allow/add documentation for Ubuntu 23 on the ROS documentation site on `rolling` that applies specifically to future versions. Prevent it from merging to `iron` docs. This has the advantage of providing a clear path for developers like me to contribute.

This hasn't been our practice in the past, and I'm worried that this will be considered some sort of support.  I'd say that this is the very last thing we would do after all the other bullet points, and we can have another conversation about it.

>     3. For CMake, attempt to remove all warnings when using CMake 3.27 to include future policy changes that will likely go into Ubuntu 24. This would involve bumping minimum CMake to above 3.5, among others. CMake 3.24 adds improved CLI usage for developers, and 3.27 adds a CMake debugger, both useful for developers who want more tools to work faster. Although Ubuntu 23 uses CMake 3.25, I am requesting 3.27 here due to soon-coming policy changes that will likely affect Ubuntu 24's chosen CMake version.

We can't upgrade CMake past what is currently in Jammy, i.e. CMake 3.22.1.  That's because we are still going to be running nightly builds on Jammy until we switch over to Ubuntu 24.04.  Further, I would not do a mass upgrade even to CMake 3.22; we do have people who are building on older/different distributions.  What I'd say we do here is upgrade CMake where we need to, and only as far as we need to, with the upper ceiling being version 3.22.1.

>     4. For QT, remove dependencies on QT5 since it's now deprecated. Whether the packages are deprecated entirely or upgraded to the next applicable QT version has to be decided.

I don't think we should remove Qt5 support completely, at least not right away.  I think we should upgrade things so that we can build using either Qt5 or Qt6.  Once we have Qt6 working nicely, then we can consider removing the Qt5 support.

>     5. Allow rosdep to allow Ubuntu 23 specific package changes, since some have changed names from Ubuntu 22.

Yes, that sounds good.

>     6. Release ros-dev-tools for Ubuntu 23.

We haven't traditionally done this, and I don't really think we need it.  `ros-dev-tools` is merely a convenience and you can install all of the tools by hand without it.

Ryanf55, 2023-07-05:
Sweet, great info. 

> We can't upgrade CMake past what is currently in Jammy, i.e. CMake 3.22.1. That's because we are still going to be running nightly builds on Jammy until we switch over to Ubuntu 24.04. Further, I [...]

Totally, for CI and what REP2000 says that makes sense. That said, is there a reason you want to block Linux developers from upgrading their local machines? The CMake developers recommend using the latest release, and due to CMake's attention to not breaking ABI, I question what the cost of allowing ROS developers to use 3.27/latest is. 

As a note, I haven't seen any of the ROS core packages set maximum versions. If they did, then I simply couldn't build on 3.27. REP 2000 already calls out that some platforms have rolling distributions, and `macOS` is [already on 3.26.4](https://github.com/Homebrew/homebrew-core/blame/0d0aea5374bb2250b76cacdd0e4de69237d25508/Formula/cmake.rb#L4), which will get bumped to 3.27 anyways. 

It's not clear what the expectation is for Mac users and if it's the same as Linux developers who want their dev environments to operate in a more latest-rolling-packages like mac. 
 
> I don't think we should remove Qt5 support completely, at least not right away. I think we should upgrade things so that we can build using either Qt5 or Qt6. Once we have Qt6 working nicely, then [...]

What does it look like for a package author on rolling that uses qt5? Do they get to decide a day they update their [qt5 rosdep dependency](https://index.ros.org/d/qtbase5-dev) to qt6, fix any breaking changes on their `rolling` branch, and then it's updated at the next rolling sync? For headers exposed that have breaking changes and dependent libraries haven't yet updated, I don't understand what the recommended process is.

clalancette, 2023-07-05:
> Totally, for CI and what REP2000 says that makes sense. That said, is there a reason you want to block Linux developers from upgrading their local machines? The CMake developers recommend using the [...]

Developers can do anything they want locally; we aren't restricting the maximum.  But most developers are going to use what is provided by Ubuntu 22.04, which will be CMake 3.22.1.  So we can't mandate that the CMake version at the top of any `CMakeLists.txt` is greater than that.

> What does it look like for a package author on rolling that uses qt5? Do they get to decide a day they update their [qt5 rosdep dependency](https://index.ros.org/d/qtbase5-dev) to qt6, fix any [...]

Yeah, it depends on what the package is and what the downstream is willing to take.  The easiest way to start here is to start transitioning the core (anything https://github.com/ros2/ros2/blob/rolling/ros2.repos) to have the *option* to build with Qt6, but to have the default be Qt5 if it is available.  Once we have that stable, we can look into the wider ecosystem (anything in https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml), and see what would break.  And try to help out there where we can.  Then I would leave that solution in place more-or-less until we transition to Ubuntu 24.04, at which point we can drop the Qt5 backwards compatibility code.

Ryanf55, 2023-07-06:
> The way that Ubuntu 23 handles python dependencies in “externally managed environments” makes the default way of installing python deps through a mix of apt and pip not work out of the box: [Python pip error "externally managed environment" after upgrading to Ubuntu 23.04 - Stack Overflow 1](https://stackoverflow.com/questions/76082216/python-pip-error-externally-managed-environment-after-upgrading-to-ubuntu-23-0)
> Related, rosidl_adapter has warnings now about pytest not being found even I install it in my virtual environment

^ One thing I forgot to add to the list, but noted on Discourse, is how Ubuntu 23 and [pep-668](https://peps.python.org/pep-0668/) handle virtual environments and system python dependencies. 
More info here: https://pythonspeed.com/articles/externally-managed-environment-pep-668/

I'm not the right person to provide a recommendation for this, but as a user, I found it difficult or impossible to meet all dependencies during the ROS 2 rolling source build even when building within a virtual environment and using rosdep. I don't have a proposal for this aspect, perhaps the ROS 2 TSC can help.

clalancette, 2023-07-06:
> I'm not the right person to provide a recommendation for this, but as a user, I found it difficult or impossible to meet all dependencies during the ROS 2 rolling source build even when building [...]

This one is going to be really painful.  For the core, we can make sure that all of our dependencies are packaged in base Ubuntu 24.04, but that is not going to work for the larger world.  I guess we probably need to spend some time and figure out either how to workaround this limitation, or to make it so that ROS 2 works in virtualenvs (which isn't currently the case, at least not easily).  @cottsay might have some ideas here as well.

Ryanf55, 2023-07-06:
> Developers can do anything they want locally; we aren't restricting the maximum. But most developers are going to use what is provided by Ubuntu 22.04, which will be CMake 3.22.1. So we can't [...]

Would it be fair to say that Tier 1 Ubuntu should aim for or be required to have zero compile warnings at 3.22.1, zero CMake warnings, and zero CMake developer warnings, and MacOS's rolling release will be best-effort to fold in changes? I imagine even humble is somewhere along that spectrum.

clalancette, 2023-07-06:
> Would it be fair to say that Tier 1 Ubuntu should aim for or be required to have zero compile warnings at 3.22.1, zero CMake warnings, and zero CMake developer warnings, 

Yes, exactly.  In point of fact, that is what we have today on Ubuntu 22.04: https://ci.ros2.org/view/nightly/job/nightly_linux_release/2713/ .  So we should aim to keep that, and we'll want to get back to that for Ubuntu 24.04.

> and MacOS's rolling release will be best-effort to fold in changes?

macOS isn't as much as a concern; it is a Tier 3 platform, so we don't run nightly CI on it.

Ryanf55, 2023-07-06:
> Yes, exactly. In point of fact, that is what we have today on Ubuntu 22.04: https://ci.ros2.org/view/nightly/job/nightly_linux_release/2713/ . So we should aim to keep that, and we'll want to get [...]

Ah, but Windows 10 is Tier 1 and has is claimed to have rolling releases; what CMake version is that running today? 
https://ci.ros2.org/view/nightly/job/nightly_win_rel/lastBuild/parameters/

clalancette, 2023-07-06:
> Ah, but Windows 10 is Tier 1 and has is claimed to have rolling releases; what CMake version is that running today?

It looks like 3.24.3: https://github.com/ros-infrastructure/ros2-cookbooks/blob/6e23e56ddb058b29b41bc65b602a27f8b50d10de/cookbooks/ros2_windows/recipes/chocolatey_installs.rb#L10-L16

clalancette, 2023-07-19:
> Would it be helpful for me to create a Dockerfile that shows the current build process, and all workarounds to get ROS 2 rolling compiling on Ubuntu 23?

Sure, that would be useful to have.

For what it is worth, I was able to compile Rolling on Ubuntu 23.04 today without any changes to the source code.  I did have to install a bunch of dependencies by hand (to workaround the lack of rosdep), and there are a handful of warnings that should be looked at, but it does seem to work.

Ryanf55, 2023-07-19:
> Sure, that would be useful to have.
>
> For what it is worth, I was able to compile Rolling on Ubuntu 23.04 today without any changes to the source code. I did have to install a bunch of dependencies [...]

That aligns with my experience, aside from needing to skip anything that relies on QT5.

clalancette, 2023-07-19:
> That aligns with my experience, aside from needing to skip anything that relies on QT5.

I was able to build everything needing Qt5 as well, and I was even able to launch RViz.  All of the Qt5 libraries still seem to be shipped in Ubuntu 23.04.

clalancette, 2023-07-20:
By the way, this will eventually be relevant: https://doc.qt.io/qt-6/cmake-qt5-and-qt6-compatibility.html

Ryanf55, 2023-07-27:
> > That aligns with my experience, aside from needing to skip anything that relies on QT5.
> 
> I was able to build everything needing Qt5 as well, and I was even able to launch RViz. All of the Qt5 libraries still seem to be shipped in Ubuntu 23.04.

Here's the current docker, not yet complete. Already a few workarounds noted. 
https://github.com/Ryanf55/ros-on-lunar/tree/add-lunar-dockerfile

Ryanf55, 2023-09-03:
I've updated the dockerfile to now get through the colcon build. It has arguments for testing both `humble`,`lunar`, and `rolling` on ubuntu 23. All three of them fail because of this issue: 
https://github.com/ros2/rosidl/issues/491

@clalancette If you know of a modification to the dockerfile to get the `colcon build` to pass without skipping packages, please share. I also would need to figure out how to get around the failing `lark` import.

clalancette, 2024-06-20:
All of this got done, one way or another, with the transition to Ubuntu 24.04 and the release of Jazzy.  That's not to say there isn't anything left to do here, but I think those things are few enough that I'm going to close this, and we can open up individual tickets for anything left.

closed 2024-06-20 by clalancette