ros2/ros2 #1454: Implementing communication between ROS2 nodes on different gateways

junwu-d, 2023-07-07:
Two different IP addresses are set under a Layer 3 switch. After setting these two IP addresses on two different system devices, the devices can ping each other successfully, but ROS 2 cannot communicate.（Sending success by ros2 run demo_nodes_cpp talker but Receiving failure，no messages by ros2 run demo_nodes_py listener） 
my computer IP is 192.168.20.12，the other computer IP is192.168.10.11
[image]
[image]
[image]

junwu-d, 2023-07-07:
[image]
[image]

EduPonz, 2023-07-07:
Hi @junwu-d,

Do you have multicast communication between the two LANs?

junwu-d, 2023-07-07:
HI @EduPonz I have only completed the tasks mentioned in the previous chart and have not implemented multicast.

EduPonz, 2023-07-07:
The thing is that automatic ROS 2 Discovery relies on multicast, so without it there is not going to be communication unless you change the discovery approach. For this, I'd recommend taking a look at the Discovery Server (check [this tutorial](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html), the important bit would be [this](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#setup-discovery-server)), which does not relay on multicast but unicast instead.

junwu-d, 2023-07-07:
thanks @EduPonz

clalancette, 2023-07-07:
I'm going to close this one out.  If you are still having issues, I'll suggest opening a question on https://answers.ros.org, which is our central question and answer site.