ros2/ros2 #1455: Client successfully calls service, but hangs after receiving response

Schteve-Earl, 2023-07-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version:
  - Humble
- Client library (if applicable):
  - rclpy

#### Problem code
```
class ImageCaptureControlNode(Node):
    def __init__(self):
        super().__init__("image_capture_control_node")
        self.initialize_parameters()
        self.load_parameters()
        self.initialize_publishers()
        self.initialize_clients()
        self.initialize_services()

    def initialize_parameters(self):
        self.declare_parameter("camera_names", ["camera_1"])
        self.declare_parameter("trigger_topic", "trigger_cameras")
        self.declare_parameter("packaged_topic", "captured_images")

    def load_parameters(self):
        self.camera_names = (
            self.get_parameter("camera_names").get_parameter_value()
            .string_array_value
        )
        self.trigger_topic = (
            self.get_parameter("trigger_topic").get_parameter_value()
            .string_value
        )
        self.packaged_topic = (
            self.get_parameter("packaged_topic").get_parameter_value()
            .string_value
        )

    def initialize_publishers(self):
        """Initializes all the required publishers
        """
        self.image_pair_publisher = self.create_publisher(
            ImagePairs, self.packaged_topic, 10
        )

    def initialize_clients(self):
        """Initiailizes all the required service clients
        """
        self.camera_clients = dict()

        for name in self.camera_names:
            client = self.create_client(CameraTrigger, f'{name}/trigger')
            if not client.wait_for_service(timeout_sec=1.0):
                self.get_logger().error(f"Service {name}/trigger not available")
                rclpy.shutdown()
            self.camera_clients[name] = client

    def initialize_services(self):
        """Initializes all the required services
        """
        self.image_capture_service = self.create_service(
            Trigger, self.trigger_topic, self.capture_callback
        )

    def capture_callback(self, request, response):

        # Prepare to store all captured images
        image_pairs = ImagePairs()

        # Call all camera services
        for camera_name, client in self.camera_clients.items():
            # Construct a request
            camera_request = CameraTrigger.Request()
            self.get_logger().info(f"calling {camera_name}/trigger")
            future = client.call_async(camera_request)
            self.get_logger().info(f"spinning until {camera_name}/trigger returns")
            rclpy.spin_until_future_complete(self, future)
            self.get_logger().info(f"{camera_name}/trigger returned")
            image_pairs.image_pairs.append(future.result().image_pair)
                    
        # Package all images into a single message
        response.success = True
        response.message = "Successfully captured all images"

        self.image_pair_publisher.publish(image_pairs)

        return response

def main(args=None):
    rclpy.init(args=args)
    node = ImageCaptureControlNode()

    rclpy.spin(node)

    node.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()

```

#### Expected behavior
The above code is a file that creates a class that contains a service and multiple clients. The callback for the service should call all available clients, and then add all the responses to a single message which it then publishes over a topic

#### Actual behavior
The code works up until these lines:
```
camera_request = CameraTrigger.Request()
self.get_logger().info(f"calling {camera_name}/trigger")
future = client.call_async(camera_request)
self.get_logger().info(f"spinning until {camera_name}/trigger returns")
rclpy.spin_until_future_complete(self, future)
self.get_logger().info(f"{camera_name}/trigger returned")
image_pairs.image_pairs.append(future.result().image_pair)
```
Instead, the client does call the service, and the service successfully returns a result. The code then hangs, and is unable to retrieve the result. I have tried using a synchronous call, like so:
```
camera_result = client.call(camera_request)
```
but the same issue occurs. The service is called and returns correctly, but the client hangs and will not progress.
#### Additional information
I have verified that the service being called is not the issue, I can call it from the command line and it reliably returns the correct results.

clalancette, 2023-07-11:
Please don't open the same issue on multiple repositories.  I'm going to close this one in favor of https://github.com/ros2/rclpy/issues/1141

closed 2023-07-11 by clalancette

Schteve-Earl, 2023-07-11:
> Please don't open the same issue on multiple repositories. I'm going to close this one in favor of [ros2/rclpy#1141](https://github.com/ros2/rclpy/issues/1141)

Sorry about that, I wasn't sure if this could be a ros communication error and belonged here, or if it was more likely to be a rclpy library issue and belonged there.