ros2/ros2 #1458: Communication fails after starting and stopping nodes

martin-traxara, 2023-07-14:
## Bug report

I'm running a vehicle and sensor simulation and controlling it with an autonomous control stack, all on a single Windows 10 PC. The ros system consists of about 12 nodes, two for the simulation and about 10 for the various pieces (perception, planning, control) of the autonomy stack. There are a total of about 75 topics, mostly single values, but there are a couple of grids and a point cloud.

My normal workflow is to launch the simulation, then start and stop the stack as I test and make changes. 

I find that after I've been doing this for a while (maybe half a dozen starts), some nodes of the stack don't run correctly when started. It's not always the same nodes, and sometimes some parts of the code seem to work, while other parts of the same node don't. After digging further, it seems some nodes just stop receiving messages on the topics they're subscribed to, which makes it look like the nodes aren't executing. I can still see the topics publishing (using 'ros2 topic echo'), but the nodes don't. 

I've run more-complicated simulations with four vehicles in the same system, meaning four times more nodes in different namespaces. While the performance is still fast enough (no indication of slow execution or dropped messages while running), in this case I can only restart the stack once or twice before some random topic will stop being received. 

Most times, I can get things working again by stopping all the nodes (including the simulation, which I normally don't because it takes a while to spin up again) and restarting, until it fails again after restarting the autonomy stack a few times. Some times it does not start properly at all, until I delete a fast-rtps folder (in C:\ProgramData\eprosima\fastrtps_interprocess).

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Binaries
- Version or commit hash:
  - Foxy (but also tested in Humble)
- DDS implementation:
  - Fast-RTPS
- Client library:
  - rclcpp and rclpy

#### Steps to reproduce issue

The project is too complicated to share here, but the general steps are:
- Start two nodes when a simulation is launched
- Start an autonomy stack
- Stop and restart the autonomy stack a few times

#### Expected behavior

The system keeps simulating correctly.

#### Actual behavior

Certain nodes seem to no longer receive messages on their subscribed topics. Which message fails changes every time.

#### Additional information

clalancette, 2023-07-27:
My initial thought here is that during the restarts, discovery isn't completely happening.  But this:

```
Some times it does not start properly at all, until I delete a fast-rtps folder (in C:\ProgramData\eprosima\fastrtps_interprocess).
```

Is something we've seen elsewhere as well.  @MiguelCompany @eduponz can you take a look?

One other thing you can try is switching to CycloneDDS, to see if there are any differences there.  You can run with `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`.

martin-traxara, 2023-07-27:
Thanks for the reply.

I think the issue where the nodes don't start at all until I delete the fast-rtps folder could be a different issue. It seems to happen sometimes if nodes crash or freeze and are forced down, not just from starting and stopping nodes. In that case, nothing works anymore, even restarting the machine, until I delete that folder. I mentioned it in case it seems relevant. 

I found the solution to that problem in a ROS Answers thread here:
https://answers.ros.org/question/402008/ros2-windows-node-always-hang/

I tried switching to CycloneDDS, and that seems to work better. I'll keep working this way and see if the problem happens again, but early testing looks good.

I agree that this example is not great for debugging. I'll try building a simple demo and see if I can reproduce it, when I get some spare time. I had also hoped there might be a log or tool that could report the status of fastrtps, as another option rather than build a new project.

fujitatomoya, 2023-07-28:
https://github.com/ros2/rmw_fastrtps/issues/699 could be related.

fujitatomoya, 2023-07-28:
@martin-traxara 

I am not 100% sure but if you could, could you try the following patches to Fast-DDS on your environment to see if the problem can be solved?

- https://github.com/eProsima/Fast-DDS/pull/3759
- https://github.com/eProsima/Fast-DDS/pull/3753

thanks in advance.

martin-traxara, 2023-08-01:
That issue sure sounds similar, so it's definitely worth testing. It would take a while to get a fastrtps build set up, so I'll wait until the fixes are in Humble and try that.

fujitatomoya, 2023-09-01:
closing in favor of https://github.com/ros2/rmw_fastrtps/issues/699.

if we still see the problem, feel free to reopen.

fujitatomoya, 2023-09-01:
@clalancette ah, i do not have permission, can you do me a favor?

closed 2023-09-01 by clalancette