ros2/ros2 #1459: [Feature request] Event API for node internal/external event

aresyy, 2023-07-17:
## Feature request

Proposal for adding general Event abstraction like msg, service, action.

#### Feature description

Currently, we already serve many events:
for the internal event (happens inside current node only), ex: `Qos event`, `launch event`
for the external event, ex: `parameter event`. But those are implemented separately. 

Think of an async scenario: Module A handles a service req and response after 1 mins or fail in the middle. Module B is response to fire req and won't wait module A's response. Module C will collect all results that module A might give (success/fail), then do something else.

It's pretty common in an async system like bank payment. Event is suitable for this use case.

#### Implementation considerations

I know we can setup pub/sub topic with `qos reliable` to make it a event pub/sub. I think that's a good way to implement it. But it's still good to provide event abstraction.

Also, it might provide some event deduplication/filter so people could have these builtin features.

clalancette, 2023-07-27:
Could you give some more concrete examples of why you want this particular feature?  Some of this may be possible but it may also be very inefficient, so knowing exactly what you are looking for would help us determine whether this is feasible.

aresyy, 2023-08-17:
Sorry for the late, currently my team create an extra layer between ros-node, so we can add our own event abstraction on our own. I might come back when I have a deeper knowledge of ros2 and proposal a detail design.

Close it for now

closed 2023-08-17 by aresyy