ros2/ros2 #1462: Do not generate the exception when action service response failed.

fujitatomoya, 2023-07-23:
follow up for https://github.com/ros2/ros2/issues/1253 in action services.

see more details for https://github.com/ros2/rclcpp/pull/2215#issuecomment-1643763666.

fujitatomoya, 2024-03-26:
@alsora @mjcarroll ether of you, could you review the following PRs? minor patch, but avoid core crash on racy service condition in action.

CC: @clalancette 

- https://github.com/ros2/rclcpp/pull/2464
- https://github.com/ros2/rcl/pull/1138
- https://github.com/ros2/rclpy/pull/1258

i think that we should backport them to iron.

alsora, 2024-03-26:
They all look good to me.

I agree on backporting wherever possible.

closed 2024-03-27 by fujitatomoya

alireza-moayyedi, 2024-04-11:
Hello @fujitatomoya, can these be backported to Humble as well?

fujitatomoya, 2024-05-02:
@alireza-moayyedi all backport PRs are created as you can see above. i will make sure CI is green and then we can merge them.

alireza-moayyedi, 2024-05-02:
@fujitatomoya Thanks a million! Much appreciated.