ros2/ros2 #1467: Reconcile best practice for setting CXX_STANDARD

Ryanf55, 2023-08-01:
This ticket is to decide the best way to set the CXX standard as it is not consistent. The approach used in the core libraries does not handle the case that CMAKE_CXX_STANDARD is defined, but an empty value. 

Affected: 
* ROS2 core, 
* ROS2 pkg create CLI
* ROS docs

              I ended up not doing this, because this isn't what we do in the core.

Actually, this is one place where what `ros2 pkg create` does and what the core does differ.  In the core, we use:

```
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 17)
endif()
```

But in `ros2 pkg create`, we generate:

```
target_compile_features(mylib PUBLIC c_std_99 cxx_std_17)  # Require C99 and C++17
```

I'm not sure which is better, and we don't use `DEFINED` in either place.  I think I'm going to leave this one as-is for now, though we should probably rectify this situation one way or the other.

_Originally posted by @clalancette in https://github.com/ros2/ros2_documentation/pull/3812#discussion_r1280665709_

mjcarroll, 2023-08-03:
>  The approach used in the core libraries does not handle the case that CMAKE_CXX_STANDARD is defined, but an empty value.

While I understand that this case isn't handled, I'm curious is the case for having CMAKE_CXX_STANDARD defined but empty?  To me, that seems like something being misconfigured by a user passing in a bad value?

clalancette, 2023-08-10:
After discussion this, this is what we think should happen going forward:

1.  Remove c_std_99 from `ros2 pkg create`.  Most people using `ros2 pkg create` aren't going to be writing C code, so having it here is just confusing and unnecessary.  Those that are using C will have to add it, but they should know what they are doing.
2.  Change the core packages to use `target_compile_features` as well as leaving the `CMAKE_CXX_STANDARD` at the top.  This lets us use the modern features of CMake, and also export the flags to downstream projects properly.  But it also leaves the `CMAKE_CXX_STANDARD` in place, which we think we need for certain parts of the core (like message generation).
3.  Change `ament_cmake_ros` to add the `CMAKE_CXX_STANDARD` by default.  That way when we want to update in the future, we have one place to change it.  Note that this will only help some packages, since many packages use `ament_cmake` instead of `ament_cmake_ros` (even in the core).  However, this will at least reduce the number of places that need to change.

All of that said, we don't have concrete plans to do this anytime soon.  I'm going to put this on the backlog as future work.